jsk-ros-pkg/jsk_recognition #1675: [jsk_pcl_ros] I don't know how to use depth_error_calibration.py

kochigami, 2016-05-18:
I'm following [this tutorial](http://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_pcl_ros/calibration.html)  in order to learn how to calibrate kinect sensor. 
(In the future, I'd like to try it on pepper.)

However, when I executed ``` rosrun jsk_pcl_ros depth_error_calibration.py --model quadratic-uv-quadratic-abs```, I got stuck.

My environment is ubuntu 14.04 and ros indigo. I use jsk_recognition from source.
I changed ``` <include file="$(find openni2_launch)/launch/openni2.launch">``` to ``` <include file="$(find freenect_launch)/launch/freenect.launch">``` of ```openni2_remote(and _local) .launch ``` . (I don't know why, but these programs didn't run for my kinect.)

I experienced these while showing checker board to kinect, press Ctrl+c and y button.
- arrow, checker board appeared in rviz (while showing checker board)
[image]
- failed in plotting graph by matplotlib (while showing checker board)
- making calibration csv file, which content is null (after pressing Ctrl+c)
- failed in making calibration yaml file (after pressing y)

Error message is as follows:
```
rosrun jsk_pcl_ros depth_error_calibration.py --model quadratic-uv-quadratic-abs
/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py:354: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
^CSave calibration parameters to calibration-2016-05-18-14-44-09.csv
Dump result into yaml file? [y/n] Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 536, in callit
    func(*args)
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/backend_tkagg.py", line 363, in idle_draw
    self.draw()
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/backend_tkagg.py", line 349, in draw
    tkagg.blit(self._tkphoto, self.renderer._renderer, colormode=2)
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/tkagg.py", line 20, in blit
    tk.call("PyAggImagePhoto", photoimage, id(aggimage), colormode, id(bbox_array))
TclError: this isn't a Tk application
y
writing to calibration_parameter_2016_05_18_14_44_12.yaml
Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py", line 434, in <module>
    main()
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py", line 414, in main
    c = classifier.coef_
AttributeError: 'LinearRegression' object has no attribute 'coef_'
```
I think 
- Failed in showing graph by matplotlib relates to fail in using matplotlib on linux
- Attribute error relates to sklearn package version

but I'm not sure. 
Could anyone help me? Sorry for my trouble.

wkentaro, 2016-05-18:
> Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py", line 434, in <module>
    main()
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py", line 414, in main
    c = classifier.coef_
AttributeError: 'LinearRegression' object has no attribute 'coef_'

The error seems to be caused by not passed --csv option.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py#L388-L401

So, I think the above error is the actual cause.

> rosrun jsk_pcl_ros depth_error_calibration.py --model quadratic-uv-quadratic-abs
/home/kochigami/catkin_ws/src/jsk_recognition/jsk_pcl_ros/scripts/check_depth_error/depth_error_calibration.py:354: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
^CSave calibration parameters to calibration-2016-05-18-14-44-09.csv
Dump result into yaml file? [y/n] Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 536, in callit
    func(*args)
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/backend_tkagg.py", line 363, in idle_draw
    self.draw()
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/backend_tkagg.py", line 349, in draw
    tkagg.blit(self._tkphoto, self.renderer._renderer, colormode=2)
  File "/usr/lib/pymodules/python2.7/matplotlib/backends/tkagg.py", line 20, in blit
    tk.call("PyAggImagePhoto", photoimage, id(aggimage), colormode, id(bbox_array))
TclError: this isn't a Tk application
y

kochigami, 2016-05-19:
Thank you very much for your help.

It may relate to my kinect version. 
Once I can find kinect one, I will try it with ```openni2_local (remote).launch```. 
Thank you very much again.

------------------------------------
What I found

1. unpublished ```depth_image_error/output``` topic from depth_error.launch
I found in order to draw graph in this program, ```depth_image_error/output``` topic from ```jsk_pcl_ros/launch/depth_error.launch``` is required, 
but it is not published in my case.
I also found my kinect is not kinect one, which is recommended to use in this tutorial.
I'd like to find kinect one and try openni2_local(remote).launch first.

2. no problem in depth_error.launch
I also compared how to use depth_image_error program in ```depth_error.launch```, but it is no problem.

■ no change in ```depth_error.launch``` (use jsk_pcl_ros package) 
(I'm still uncertain why it works even though jsk_pcl_ros package doesn't seem to have depth_image_error.cpp or something..)
```
<node pkg="jsk_pcl_ros" type="depth_image_error"
        name="depth_image_error"
        output="screen">
```
■ use jsk_pcl_utils nodelet
```
 <node pkg="nodelet" type="nodelet"
        name="depth_image_error"
        args="standalone jsk_pcl_utils/DepthImageError"
        output="screen">
```

comparison of ```rosnode info /depth_image_error```:

```
rosnode info /depth_image_error (jsk_pcl_ros package version)
--------------------------------------------------------------------------------
Node [/depth_image_error]
Publications: 
 * /depth_image_error/output [jsk_recognition_msgs/DepthErrorResult]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: 
 * /depth_image_error/list
 * /depth_image_error/load_nodelet
 * /depth_image_error/set_logger_level
 * /depth_image_error/get_loggers
 * /depth_image_error/unload_nodelet

contacting node http://192.168.97.25:44630/ ...
Pid: 13242
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
```

```
depth_error_image をnodeletで書いた時
 rosnode info /depth_image_error 
--------------------------------------------------------------------------------
Node [/depth_image_error]
Publications: 
 * /depth_image_error/output [jsk_recognition_msgs/DepthErrorResult]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: 
 * /depth_image_error/set_logger_level
 * /depth_image_error/get_loggers

contacting node http://192.168.97.25:55356/ ...
Pid: 6159
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
```

closed 2016-05-19 by kochigami