jsk-ros-pkg/jsk_recognition #1682: Test fails on hydro if jsk_topic_tools / jsk_recognition_utils are installed via apt

wkentaro, 2016-05-20:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1681#issuecomment-220589893

@k-okada This is fixed in indigo via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1512
But maybe needs passing --ignore-src in rosdep install on travis.

wkentaro, 2016-05-20:
Reported https://github.com/jsk-ros-pkg/jsk_travis/issues/265

k-okada, 2016-05-21:
This is the detailed log
```
/usr/bin/c++   -DROS_PACKAGE_NAME=\"jsk_perception\" -DROSCONSOLE_BACKEND_LOG4CXX -DUSE_OLD_YAML -fopenmp -I/tmp/ws/devel/include -I/opt/ros/hydro/include/opencv -I/opt/ros/hydro/include -I/usr/include/eigen3 -I/tmp/ws/src/jsk_recognition/jsk_perception/include -I/tmp/ws/src/jsk_topic_tools/include -I/tmp/ws/src/jsk_recognition/jsk_recognition_utils/include -I/usr/include/pcl-1.7 -I/usr/include/ni -I/usr/include/vtk-5.8 -I/usr/include/qhull -I/tmp/ws/build/jsk_perception/build/patched-SLIC-Superpixels    -g -O2 -o CMakeFiles/linemod.dir/src/linemod.cpp.o -c /tmp/ws/src/jsk_recognition/jsk_perception/src/linemod.cpp
```
The problem was '/opt/ros/hydro/include' is listed before `${catkin_INCLUDE_DIR}`, the problem was `OpenCVConfig.cmake` has
```
# Provide the include directories to the caller
set(OpenCV_INCLUDE_DIRS "${OpenCV_INSTALL_PATH}/include/opencv;${OpenCV_INSTALL_PATH}/include")
include_directories(${OpenCV_INCLUDE_DIRS})
```
and on hydro `${OpenCV_INSTALL_PATH}` is `/opt/hydro` and on indigo it is `/usr/` (from kinetic it might be /opt/kinetic` again for OpenCV3, so we need care about it )

This problem is introduced in https://github.com/jsk-ros-pkg/jsk_recognition/pull/1598 and fixed in https://github.com/jsk-ros-pkg/jsk_recognition/pull/1684/commits/7602645a0e4e31471a937acf6339abeba3b420cf

wkentaro, 2016-05-21:
Thank you. I updated https://github.com/jsk-ros-pkg/jsk_recognition/pull/1685 .

wkentaro, 2016-05-25:
This is fixed on indigo via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1512.
And hydro users can avoid installing jsk ros packages via apt. Closing now.

closed 2016-05-25 by wkentaro