jsk-ros-pkg/jsk_recognition #1683: Proposals of introducing mention-bot

wkentaro, 2016-05-20:
https://github.com/facebook/mention-bot#mention-bot-

The bot automatically assign people for reviews.
Currently, there is few reviewers and GitHub flow is too slow for me. (after graduation of garaemon)
I hope this works well.
Any idea?

knorth55, 2016-05-29:
i support @wkentaro

k-okada, 2016-05-29:
if that bot analyze blame for mention, all mention goes to garaemon so I
think this will not change the situation, only solution for this is to
increase number of people who understand this package.

wkentaro, 2016-05-29:
> if that bot analyze blame for mention, all mention goes to garaemon 

That can be prevented by creating blacklist.

Anyway, could you please add this to [start-jsk/jsk_apc](https://github.com/start-jsk/jsk_apc) first? @k-okada