jsk-ros-pkg/jsk_recognition #1690: Travis CI test fails on Hydro

wkentaro, 2016-05-22:
Found at https://github.com/jsk-ros-pkg/jsk_recognition/pull/1646#issuecomment-220849844

It won't fix...

wkentaro, 2016-05-22:
Is this a Jenkins specific problem? Or it happens in all Hydro + Ubuntu 12.04 environments?

k-okada, 2016-05-23:
I can not reproduce this with docker used in jenkins, may be something goes wrong within jsk_travis?
```
lokada@jenkins:~$ docker run --rm -i -t ros-ubuntu:12.04 /bin/bash
root@e740c7b7ebd4:/# sudo -H apt-get install -y -qq ros-hydro-axis-camera
bash: sudo: command not found
root@e740c7b7ebd4:/# apt-get install sudo
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  sudo
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 288 kB of archives.
After this operation, 1020 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu/ precise-updates/main sudo amd64 1.8.3p1-1ubuntu3.7 [288 kB]
Fetched 288 kB in 1s (170 kB/s)
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 1.)
debconf: falling back to frontend: Readline
Selecting previously unselected package sudo.
(Reading database ... 152752 files and directories currently installed.)
Unpacking sudo (from .../sudo_1.8.3p1-1ubuntu3.7_amd64.deb) ...
Processing triggers for man-db ...
Setting up sudo (1.8.3p1-1ubuntu3.7) ...
root@e740c7b7ebd4:/# sudo -H apt-get install -y -qq ros-hydro-axis-camera
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 2.)
debconf: falling back to frontend: Readline
Selecting previously unselected package ros-hydro-camera-info-manager-py.
(Reading database ... 152792 files and directories currently installed.)
Unpacking ros-hydro-camera-info-manager-py (from .../ros-hydro-camera-info-manager-py_0.2.3-0precise-20150328-1043-+0000_amd64.deb) ...
Selecting previously unselected package ros-hydro-axis-camera.
Unpacking ros-hydro-axis-camera (from .../ros-hydro-axis-camera_0.2.0-0precise-20150506-2142-+0000_amd64.deb) ...
Setting up ros-hydro-camera-info-manager-py (0.2.3-0precise-20150328-1043-+0000) ...
Setting up ros-hydro-axis-camera (0.2.0-0precise-20150506-2142-+0000) ...
root@e740c7b7ebd4:/# 
```

wkentaro, 2016-05-23:
But no error at this time: https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/131734256
(at the commit I upgraded the jsk_travis)

```
Unpacking xfonts-100dpi (from .../xfonts-100dpi_1%3a1.0.3_all.deb) ...

Selecting previously unselected package xfonts-75dpi.

Unpacking xfonts-75dpi (from .../xfonts-75dpi_1%3a1.0.3_all.deb) ...

Selecting previously unselected package ros-hydro-dynamic-tf-publisher.

Unpacking ros-hydro-dynamic-tf-publisher (from .../ros-hydro-dynamic-tf-publisher_2.0.3-0precise-20150728-0056-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-euslisp.

Unpacking ros-hydro-euslisp (from .../ros-hydro-euslisp_9.15.1-0precise-20150831-1454-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-geneus.

Unpacking ros-hydro-geneus (from .../ros-hydro-geneus_2.2.4-0precise-20150708-0843-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-image-view2.

Unpacking ros-hydro-image-view2 (from .../ros-hydro-image-view2_2.0.3-0precise-20150728-0052-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-libsiftfast.

Unpacking ros-hydro-libsiftfast (from .../ros-hydro-libsiftfast_2.0.8-0precise-20150915-2056-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-jsk-recognition-msgs.

Unpacking ros-hydro-jsk-recognition-msgs (from .../ros-hydro-jsk-recognition-msgs_0.3.6-0precise-20150911-0859-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-jsk-topic-tools.

Unpacking ros-hydro-jsk-topic-tools (from .../ros-hydro-jsk-topic-tools_2.0.3-0precise-20150808-0742-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-jsk-recognition-utils.

Unpacking ros-hydro-jsk-recognition-utils (from .../ros-hydro-jsk-recognition-utils_0.3.6-0precise-20150911-1012-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-posedetection-msgs.

Unpacking ros-hydro-posedetection-msgs (from .../ros-hydro-posedetection-msgs_2.0.0-1precise-20150619-1847-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-jskeus.

Unpacking ros-hydro-jskeus (from .../ros-hydro-jskeus_1.0.10-0precise-20150831-1552-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-roseus.

Unpacking ros-hydro-roseus (from .../ros-hydro-roseus_1.3.9-0precise-20150915-0101-+0000_amd64.deb) ...

Selecting previously unselected package ros-hydro-jsk-perception.

Unpacking ros-hydro-jsk-perception (from .../ros-hydro-jsk-perception_0.3.6-0precise-20150915-0828-+0000_amd64.deb) ...
```

wkentaro, 2016-05-23:
I'm going to just restart it.

wkentaro, 2016-05-23:
Fixed now.

closed 2016-05-23 by wkentaro