jsk-ros-pkg/jsk_recognition #1717: [jsk_pcl_ros] OrganizedMultiPlaneSegmentationの使い方についてのご相談

kochigami, 2016-05-31:
Pepperで机の平面を認識したいのですが，認識することができません．
机からは60cm~1mくらい離れて斜め方向に机が見えます．

[image]

平面に対して直角に顔が向かないと，認識できないのでしょうか．
他のロボットで，覗きこむようにしなくても机が認識できるのか知りたいです．

床だと認識ができます．
[image]
[image]

このようなプログラムでOrganizedMultiPlaneSegmentation, PlaneExtractionを試しています．
パラメータのうち
min_size: 200
max_curvature: 0.3 など変えてみたけれど，認識はできませんでした．
```
<!-- -*- mode: xml -*- -->
<launch>
  <arg name="INPUT" default="/pepper_robot/camera/depth/points"/>
  <arg name="FILTER_NAME_SUFFIX" default=""/>
  <arg name="create_manager" default="true" />
  <arg name="manager" default="plane_extraction_manager" />

  <node if="$(arg create_manager)"
        pkg="nodelet" type="nodelet" name="$(arg manager)"
        args="manager" output="screen"/>
  
  <node name="input_relay" pkg="nodelet" type="nodelet"
        args="load jsk_topic_tools/Relay $(arg manager)">
      <remap from="~input" to="$(arg INPUT)" />
  </node>
  
  <node pkg="nodelet" type="nodelet" name="multi_plane_estimate"
    	args="load jsk_pcl/OrganizedMultiPlaneSegmentation $(arg manager)"
        output="screen">
    <remap from="~input" to="/input_relay/output" />
    <rosparam>
      use_async: true
      min_size: 2000
      max_curvature: 0.01 
      estimate_normal: true
      publish_normal: true  
    </rosparam>
  </node>
 
  <node pkg="nodelet" type="nodelet" name="plane_extraction" args="load jsk_pcl/MultiPlaneExtraction $(arg manager)" output="screen">
    <rosparam>
      use_async: true
      max_height: 1.0
      use_indices: false
    </rosparam>
    <remap from="~input" to="/input_relay/output" />
    <remap from="~indices" to="/multi_plane_estimate/output_refined" />
    <remap from="~input_polygons" to="/multi_plane_estimate/output_refined_polygon" />
    <remap from="~input_coefficients" to="/multi_plane_estimate/output_refined_coefficients" />
  </node> 

</launch>
```

kinectで一様同じプログラムが動きました．
[image]

k-okada, 2016-05-31:
これは，キネクトのカメラだと平面がでているけど，Pepper備え付けのカメラだと平面が出ないということを言っている？
となると，平面性が足りない，ということになりますね．おそらくカメラ−のデータの質があまり良くない．（僕は圧縮しているんじゃないかと穿っている部分）

min_size: 200 を可能な限り小さくして
max_curvature: 0.3 を可能な限り大して，

正解を含んでいるけど，不正解が大量にはいっている平面候補が出るか調べてみてください．

kochigami, 2016-06-02:
ペッパーが見えている範囲は一般のxtionの1/4なので，
（pepper: height*width = 240*320に対し，xtion: 480*640）
物にできるだけ近づくと平面性が保たれることが分かりました．
きちんと理解しておらずすみません．データシートを見直してみます．
パラメータはかなりminを小さく，curvatureを大きくしないと駄目でした．
他のロボットと同じようにはいかないみたいです．

近視用でなく，実は遠視用のカメラを新たにつける必要があるかもしれません．
ご教示ありがとうございました．

60cm 机から離れて首ピッチ角0度（まっすぐの姿勢）で見た時
min_size: 20 curvature: 1.0 
認識されません．
[image]

40cm 机から離れて首ピッチ角30度（机を覗きこむ姿勢）で見た時
min_size: 50 curvature: 2.5 
きれいに認識されました．
[image]

40cm 机から離れて首ピッチ角30度（机を覗きこむ姿勢）で見た時　ぬいぐるみを並べた状態で
min_size: 50 curvature: 2.5 
認識されませんでした．
[image]

closed 2016-06-02 by kochigami

k-okada, 2016-06-06:
2016-06-02 19:14 GMT+09:00 Kanae Kochigami <notifications@github.com>:

> ペッパーが見えている範囲は一般のxtionの1/4なので，
> （pepper: height*width = 240*320に対し，xtion: 480*640）
>

これは確認して欲しいんだけど，間引いたら画像サイズは小さくなるけど，画角は同じ，というのはでいるんだけど，ちがうかな．

--
◉ Kei Okada

kochigami, 2016-06-06:
> これは確認して欲しいんだけど，間引いたら画像サイズは小さくなるけど，画角は同じ，というのはでいるんだけど，ちがうかな．

アドバイスありがとうございます．xtion pro live （QVGA 320*240, 30fps）で見てみたら，
サイズは小さくなり，画角は変わりませんでした．

見える面積は1/4になったが，ペッパーみたいな点群の凸凹はないことから可能性は次の２つかなと思いました．
- 体内でxtionに何かしていると思われる（体内でのコードはまだ探せていない（そもそも見えないようになっている？））
- pepperが持っているのは，xtion pro live ではなくxtion なので，性能が違う？

xtion pro live　QVGA (320*240) 30fps の画像　（ペッパーはQVGA 20fps）： 
正面
[image]

横
[image]