jsk-ros-pkg/jsk_recognition #1724: Deprecated nodes/nodelets are not deprecated in actual

wkentaro, 2016-06-02:
@iory 
You changed the documentation for deprecated nodes/nodelets by this [PR](https://github.com/jsk-ros-pkg/jsk_recognition/pull/1696).
Maybe you should also update as below for `edge_detector`:

- `jsk_perception::EdgeDetector` to `edge_detector::EdgeDetector` at [here](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/jsk_perception_nodelets.xml#L189)
- `jsk_perception/EdgeDetector` to `vision_opencv/EdgeDetector` at [here](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/CMakeLists.txt#L142)

and same changes for other deprecated nodes/nodelets.

wkentaro, 2016-08-01:
@iory What status is this?

iory, 2016-08-01:
I'm waiting for this.
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1726

also https://github.com/ros-perception/opencv_apps/pull/17

wkentaro, 2016-09-17:
Once this issue is closed, but why is it reopened?

k-okada, 2016-09-21:
i thought this was wrongly closed by @iory, because there is comment on why you close this, 
but found this was solved in https://github.com/jsk-ros-pkg/jsk_recognition/pull/1859

closed 2016-09-21 by k-okada