jsk-ros-pkg/jsk_recognition #1751: [jsk_pcl_ros]cluster_decomposer in hsi_color_filter.launch doesn't publish mask image

Horisu, 2016-06-17:
Original point cloud has width and height.
But after processed color_filter, height is 1 and width seems to be the length of point cloud.
Color_filter uses cloud's height and width to make mask image , so it can't make correct mask image.
How can we fix it?

k-okada, 2016-06-17:
please write all information that we can reproduce your problem. One of the bast way is to put launch file here.

k-okada, 2016-06-17:
you can use https://gist.github.com/ to put launch file and add link to here

Horisu, 2016-06-17:
You need to do is only to make cloud topic and to launch hsi_color_filter.launch .
```bash
roscore
# launch depth camera
roslaunch openni2_launch openni2.launch
# launch hsi_color_filter
roslaunch jsk_pcl_ros hsi_color_filter.launch INPUT:=/camera/depth_registered/points
# configure and make some clusters
rosrun rqt_reconfigure rqt_reconfigure
#mask image never comes
rostopic echo /HSI_color_filter/cluster_decomposer/mask
```
[image]

iory, 2016-06-17:
@Horisu 
Please fetch and try https://github.com/jsk-ros-pkg/jsk_recognition/pull/1752

Horisu, 2016-06-17:
awesome!

wkentaro, 2016-07-11:
Fixed via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1752

closed 2016-07-11 by wkentaro