jsk-ros-pkg/jsk_recognition #1778: Build failed with opencv-3.x

mizmizo, 2016-07-13:
indigo,opencv-3.1.0でjsk_perceptionをbuildすると、
```
/home/hiroto/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/src/bounding_rect_mask_image.cpp: In member function ‘virtual void jsk_perception::BoundingRectMaskImage::convert(const ConstPtr&)’:
/home/hiroto/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/src/bounding_rect_mask_image.cpp:73:5: error: ‘vector’ is not a member of ‘cv’
     cv::vector<cv::vector<cv::Point> > contours;
     ^
```
といったようなエラーが出てbuildに失敗します。

http://stackoverflow.com/questions/33400823/opencv-3-0-0-cvvector-missing
を参考に
d592a1c
のようにするとbuildが通るのですが、この方法で解決法として正しいでしょうか？

k-okada, 2016-07-13:
```
/home/hiroto/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/src/bounding_rect_mask_image.cpp:73:5:
error: ‘vector’ is not a member of ‘cv’
```
bounding_rect_mask_image.cpp の 73行目のcv::vectorをstd::vectorにかえるのでいいんじゃないかな．

wkentaro, 2016-07-15:
Fixed via #1779

closed 2016-07-15 by wkentaro