jsk-ros-pkg/jsk_recognition #1816: build disables ssl verification globally

tfoote, 2016-08-04:
disabling ssl verification globally for anyone that builds this is a very bad side effect. 

https://github.com/jsk-ros-pkg/jsk_recognition/blob/e85db7790a312af4f45c029c08d255f28dc56e8a/jsk_perception/Makefile.slic#L16

This will cause any future git checkout by that user to not check the ssl certificate. 

You should only disable the verification for the urls that are known using `-c http.sslVerify=false` in the line [where you clone.](https://github.com/jsk-ros-pkg/jsk_recognition/blob/e85db7790a312af4f45c029c08d255f28dc56e8a/jsk_perception/Makefile.slic#L22)

A good reference is here: 
https://stackoverflow.com/questions/11621768/how-can-i-make-git-accept-a-self-signed-certificate

wkentaro, 2016-08-23:
Fixed?

tfoote, 2016-08-23:
#1818 looks to fix this.

closed 2016-08-23 by tfoote