jsk-ros-pkg/jsk_recognition #1817: jsk_preception build is unstable due to cloning during the build process

tfoote, 2016-08-04:
This job is failing frequently due to cloning SLIC-Superpixels during the build. http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__jsk_perception__ubuntu_trusty_armhf__binary/

It's strongly recommended not to fetch resources during the build process as that means that your build process is reliant on having network connectivity as well as the remote server being available at the time of build. 

Looking at that repo there appear to be only 2 files in the repo and they appear to be pretty static. As such I'd recommend simply copying them into the repo, and you can have a script to update them from upstream when and if there's a change of interest. 

Also I'll note that there's no license which should be requested of upstream before it should be used either by cloning or by copying.

k-okada, 2016-08-05:
> uThf__jsk_perception__ubuntu_trusty_armhf__binary/

thanks, I originally designed jsk_3rdparty repository to support this type
of external library...
Other solution is if we can use submodule (
https://github.com/ros-infrastructure/bloom/issues/217), we can apply
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1425 patch

wkentaro, 2016-08-23:
Fixed?

tfoote, 2016-08-23:
I don't think there's been any changes. The failure is intermittent, it looks to be ~ 75% of builds fail.

k-okada, 2016-08-25:
@tfoote do you have plan to release new version of vcstools https://github.com/vcstools/vcstools ? if so, we can fix this very soon.

tfoote, 2016-08-26:
@k-okada released

k-okada, 2016-08-30:
thanks, create pr to solve this problem, thanks https://github.com/jsk-ros-pkg/jsk_recognition/pull/1851

closed 2016-08-30 by k-okada