jsk-ros-pkg/jsk_recognition #1831: remove box_array_to_box.py

k-okada, 2016-08-07:
there are C++ version and python version for same purpose:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1799/commits/25060c3e517b2a433060762f4afa4e9d2baa7597
https://github.com/jsk-ros-pkg/jsk_recognition/commits/master/jsk_pcl_ros/scripts/box_array_to_box.py

we first find the use case of box_array_to_box.py and consider if it is possible to move to C++ version.

wkentaro, 2016-08-07:
I sent PR: https://github.com/jsk-ros-pkg/jsk_recognition/pull/1833
But this breaks current API, so it needs to be merged for a major release.
Do you have any idea about the rule about release and breaking API? @k-okada 
For example

|                                    | BugFix | New Node/Nodelets | Break API |
|------------------------------------|--------|-------------------|-----------|
| Major release (ex. 0.3.X -> 1.0.0) | Yes    | Yes               | Yes       |
| Minor release (ex. 0.3.X -> 0.4.0) | Yes    | Yes               | No        |
| Patch release (ex. 0.3.8 -> 0.3.9) | Yes    | No                | No        |

k-okada, 2016-08-08:
I think the best timing is when we decided to drop hydro support, currently
hrp2 and pr2 is blocker,

wkentaro, 2016-08-08:
> I think the best timing is when we decided to drop hydro support

I think breaking API and supporting hydro is unrelated.
For example, currently jsk_recognition is already released to hydro, indigo and jade, and
if we break the API the effect is not limited to hydro but indigo and jade.
For this kind of problem, I thought controlling tag version number is most popular solution.
What do you think?

k-okada, 2016-08-08:
api change and distro upgrade is not related, I just want to push major
 release with some evens, like dropping hydro, start new semester, finished
some event, so users thinks it is reasonable even if they can not
compile/run their program.

wkentaro, 2016-08-08:
> api change and distro upgrade is not related,

For usual softwares, api change and version number is related, isn't it?
But I understood that for this package they are unrelated.

k-okada, 2016-08-09:
Oh I mean api change and dropping hydro is not related as you said. and
since we do not create distro specific branch, our software do not have
normal "distro upgrade" process, we'll support newer distro while everyone
(or most of people) using old one. Dropping old distro is only event that
effects to users and that is "distro upgrade" process for us.

wkentaro, 2016-08-23:
I agree your idea that the best timing of breaking API is the time of dropping distro in the lab.
But it seems once in 4 year, and it is too few times I think.

Generally speaking, how do you plan to manage this jsk_recognition package for API changes?
Do you have idea like I commented in https://github.com/jsk-ros-pkg/jsk_recognition/issues/1831#issuecomment-238089223 .

k-okada, 2016-08-25:
On Tue, Aug 23, 2016 at 1:09 PM, Kentaro Wada <notifications@github.com>
wrote:

> I agree your idea that the best timing of breaking API is the time of
> dropping distro in the lab.
> But it seems once in 4 year, and it is too few times I think.
>

I agree that dropping distro is very rare situation, but we'll almost able
to drop hydro very soon ( @mmurooka, @iroy, @furushchev )

> Generally speaking, how do you plan to manage this jsk_recognition package
> for API changes?
> Do you have idea like I commented in #1831 (comment)
> <https://github.com/jsk-ros-pkg/jsk_recognition/issues/1831#issuecomment-238089223>
> .
>
this is very general and reasonable idea for version number rule

--
◉ Kei Okada

wkentaro, 2016-08-30:
Ok, so the process would be below, and current status is **1**? If so, could you please release jsk_recognition minor minor version, 0.4.0 as a final version of supporting hydro?

1. Release jsk_recognition as version which supports hydro
2. Remove testing hydro and remove suppot on hydro
3. hrp2 user and pr2 user migrate to indigo
4. Keep developing jsk_recognition without support of hydro

k-okada, 2016-08-30:
1. hrp2 user and pr2 user migrate to indigo
2. Release jsk_recognition as version which supports hydro
3. Remove testing hydro and remove suppot on hydro
4. Keep developing jsk_recognition without support of hydro

so now @Furushchev is the blocker
--
◉ Kei Okada

wkentaro, 2016-08-30:
>  hrp2 user and pr2 user migrate to indigo

I think it takes about 1 year, and we can't wait, and if you agree the releasing rule at https://github.com/jsk-ros-pkg/jsk_recognition/issues/1831#issuecomment-238089223, why not release final version of jsk_recognition for hydro first?
Because this kind of API breaking is not limited to box_array_to_box.py and we will have many similar cases, and I can't wait for the next distro at each time.

mmurooka, 2016-08-30:
> 1. hrp2 user and pr2 user migrate to indigo

Also, we need to care about all HRP2 in other buildings.

k-okada, 2016-08-30:
btw how about @yuki-asano

closed 2016-10-27 by k-okada