jsk-ros-pkg/jsk_recognition #1834: Sample of OctomapServerContact does not work on indigo: voxel is not updated with contact?

wkentaro, 2016-08-09:
@mmurooka I think we encountered this problem last month. Is this fixed?

mmurooka, 2016-08-10:
ちょっと調べたのですがすぐに分からなくて，少し時間がかかるかもしれないです．．

wkentaro, 2016-08-10:
了解です。

mmurooka, 2017-05-16:
One possible reason is https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issuecomment-301604262

mmurooka, 2017-05-16:
Now I could not reproduce the problem that voxel is not updated with contact.
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issue-228764609

Is this problem still occur in your environment?

wkentaro, 2017-05-16:
Fixed. Thanks .

```
workspace: /home/wkentaro/Projects/label_octomap/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 start-jsk/rtmros_tutorials    git master         64d2b3c25035 github.com/start-jsk/rtmros_tutorials.git
 jsk-ros-pkg/jsk_recognition   git master         6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition.git
 OctoMap/octomap_ros           git indigo-devel   29ab0016c267 github.com/OctoMap/octomap_ros.git
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel          4f0127598d50 github.com/OctoMap/octomap.git
```

closed 2017-05-16 by wkentaro

mmurooka, 2017-05-16:
And I also confirmed that octomap source is no longer necessary. https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issuecomment-301725696