jsk-ros-pkg/jsk_recognition #1838: SegmentationFault in OrganizedMultiPlaneSegmentation

mmurooka, 2016-08-10:
カメラの前で静かに地面を映しているときは起きないで，
カメラの前で箱を激しく動かしたりすると置きますが確実な再現条件は分かっていないです．

SegmentationFaultのときのgdbの出力を最後に貼ります．
```
#13 pcl::KdTreeFLANN<pcl::PointXYZRGBA, flann::L2_Simple<float> >::~KdTreeFLANN
```
とあるので，
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp#L256-L273
のあたりではないかと思います．
```
[New Thread 0x7fffa6ae1700 (LWP 19219)]
*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': free(): corrupted unsorted chunks: 0x00007fff9836d660 ***

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffdffff700 (LWP 19166)]
0x00007ffff60c2c37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) btrace
Undefined command: "btrace".  Try "help".
(gdb) bt
#0  0x00007ffff60c2c37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff60c6028 in __GI_abort () at abort.c:89
#2  0x00007ffff60ff2a4 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff620d6b0 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff610b55e in malloc_printerr (ptr=<optimized out>, 
    str=0x7ffff620d800 "free(): corrupted unsorted chunks", action=1)
    at malloc.c:4996
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0)
    at malloc.c:3840
#5  0x00007fffdb8f6f98 in freeIndex (this=0x7fff98368ea0)
    at /usr/include/flann/algorithms/kdtree_single_index.h:344
#6  ~KDTreeSingleIndex (this=0x7fff98368ea0, __in_chrg=<optimized out>)
    at /usr/include/flann/algorithms/kdtree_single_index.h:133
#7  flann::KDTreeSingleIndex<flann::L2_Simple<float> >::~KDTreeSingleIndex (
    this=0x7fff98368ea0, __in_chrg=<optimized out>)
    at /usr/include/flann/algorithms/kdtree_single_index.h:134
#8  0x00007fffdb8e8086 in ~Index (this=0x7fff98368d60, 
    __in_chrg=<optimized out>) at /usr/include/flann/flann.hpp:131
#9  flann::Index<flann::L2_Simple<float> >::~Index (this=0x7fff98368d60, 
    __in_chrg=<optimized out>) at /usr/include/flann/flann.hpp:132
#10 0x00000000004061ae in boost::detail::sp_counted_base::release() ()
#11 0x00007fffdb50a836 in ~shared_count (this=0x7fffdfffe220, 
    __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:371
#12 ~shared_ptr (this=0x7fffdfffe218, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_ptr.hpp:328
#13 pcl::KdTreeFLANN<pcl::PointXYZRGBA, flann::L2_Simple<float> >::~KdTreeFLANN
    (this=0x7fffdfffe1d0, __in_chrg=<optimized out>)
    at /usr/local/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:139
#14 0x00007fffdb4fe008 in jsk_pcl_ros::OrganizedMultiPlaneSegmentation::connectPlanesMap (this=0xc6eaf0, input=..., 
    model_coefficients=std::vector of length 2, capacity 2 = {...}, 
    boundary_indices=std::vector of length 2, capacity 2 = {...}, 
Python Exception <class 'IndexError'> list index out of range: 
    connection_map=std::map with 1 elements)
    at /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp:273
#15 0x00007fffdb5000b7 in jsk_pcl_ros::OrganizedMultiPlaneSegmentation::segmentFromNormals (this=0xc6eaf0, input=..., normal=..., header=...)
    at /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp:581
#16 0x00007fffdb5047d2 in jsk_pcl_ros::OrganizedMultiPlaneSegmentation::segment
    (this=0xc6eaf0, msg=...)
    at /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp:750
#17 0x00007fffdb2ee00c in operator() (a0=..., this=<optimized out>)
    at /usr/include/boost/function/function_template.hpp:767
#18 boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) (function_obj_ptr=..., 
    a0=...) at /usr/include/boost/function/function_template.hpp:153
#19 0x00007fffdb396c17 in operator() (a0=..., this=0x7fff98000cd8)
    at /usr/include/boost/function/function_template.hpp:767
#20 ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call (this=0x7fff98000cd0, 
    params=...)
    at /opt/ros/indigo/include/ros/subscription_callback_helper.h:144
#21 0x00007ffff74f96b5 in ros::SubscriptionQueue::call() ()
   from /opt/ros/indigo/lib/libroscpp.so
#22 0x00007ffff74b3107 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/indigo/lib/libroscpp.so
#23 0x00007ffff74b3527 in ros::CallbackQueue::callOne(ros::WallDuration) ()
   from /opt/ros/indigo/lib/libroscpp.so
#24 0x00007ffff7bc1b10 in nodelet::detail::CallbackQueue::callOne() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#25 0x00007ffff7bc2d06 in nodelet::detail::CallbackQueueManager::workerThread(no---Type <return> to continue, or q <return> to quit---
delet::detail::CallbackQueueManager::ThreadInfo*) ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#26 0x00007ffff5625a4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#27 0x00007ffff5404184 in start_thread (arg=0x7fffdffff700)
    at pthread_create.c:312
#28 0x00007ffff618637d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

mmurooka, 2016-08-10:
Ubuntu14.04, indigo pcl version1.8をソースから入れています．

mmurooka, 2016-08-10:
@iory の環境（Ubuntu 14.04, indigo, pclはaptから1.71）では再現しませんでした．
@oshiroy の環境（Ubuntu 14.04, indigo, pclはソースから1.8）では同じエラーが再現しました．

ちなみにソースのPCLはOrganizedEdgeDetectorを試したくて入れていました．
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/CMakeLists.txt#L241-L244

YuOhara, 2016-08-11:
ちなみに、もしまだ治ってなければ、時間あるときに私のpcでも起こっているか試してもらえると助かります。
もし起こってなければ、

* どこかの層からコンパイルしなおす(indigo or indigo_parentをcatkin clean?)
* pcl のバージョンを1.8.1 とかにする

とかで治ったりしないでしょうか？

mmurooka, 2016-08-11:
はい，試してみます．
書き忘れていましたが，以下のようにして再現するか確かめていました．
https://gist.github.com/mmurooka/0e66da9a6f830e0b0e30fcc2fc9bf084

YuOhara, 2016-08-18:
私のPCでは（今は）問題なく動いているように見えます。

YuOhara, 2016-08-18:
@wkentaro に聞いたら、pcl上流の該当箇所で、1.7.2のタイミングで変更があったそうです。
ひょっとしたら、私の環境の方が間違ってるのかもしれないですが、
とりあえず
develなど以下のjskpcl関連をきれいにして、もう一度コンパイルして、試してみていただけますでしょうか？

mmurooka, 2016-08-19:
devel, buildを消して再コンパイルしたら，セグフォらなくなりました．ありがとうございました．

mmurooka, 2016-08-19:
> @wkentaro に聞いたら、pcl上流の該当箇所で、1.7.2のタイミングで変更があったそうです。

というのは，たぶんhttps://github.com/PointCloudLibrary/pcl/pull/618 ですね，ありがとうございます．

YuOhara, 2016-10-09:
再発してしまいました・・・
buildしなおしても同じです．(stat上げてるPCで再発しました．)

sourceからコンパイルしたpclだけでなく，aptでpclが入ってるのがダメなのでしょうか？

```
[FATAL] [1476002610.796187743]: [jsk_pcl_ros::OrganizedMultiPlaneSegmentation::estimateNormal] unknown estimation method, force to use COVARIANCE_MATRIX: 14301408
[pcl::IntegralImagesNormalEstimation::setNormalSmoothingSize] Invalid normal smoothing size given! (0.000000). Allowed ranges are: 0 < N. Defaulting to 10.000000.
*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': free(): corrupted unsorted chunks: 0x00007fff982175c0 ***

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffdffff700 (LWP 25672)]
0x00007ffff60c2c37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) btrace
Undefined command: "btrace".  Try "help".
(gdb) bt
#0  0x00007ffff60c2c37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff60c6028 in __GI_abort () at abort.c:89
#2  0x00007ffff60ff2a4 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff620d6b0 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff610b55e in malloc_printerr (ptr=<optimized out>, 
    str=0x7ffff620d800 "free(): corrupted unsorted chunks", action=1)
    at malloc.c:4996
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0)
    at malloc.c:3840
#5  0x00007fffdb8d7328 in freeIndex (this=0x7fff981c24c0)
    at /usr/include/flann/algorithms/kdtree_single_index.h:344
#6  ~KDTreeSingleIndex (this=0x7fff981c24c0, __in_chrg=<optimized out>)
    at /usr/include/flann/algorithms/kdtree_single_index.h:133
#7  flann::KDTreeSingleIndex<flann::L2_Simple<float> >::~KDTreeSingleIndex (
    this=0x7fff981c24c0, __in_chrg=<optimized out>)
    at /usr/include/flann/algorithms/kdtree_single_index.h:134
#8  0x00007fffdb8c8416 in ~Index (this=0x7fff981c2380, 
    __in_chrg=<optimized out>) at /usr/include/flann/flann.hpp:131
#9  flann::Index<flann::L2_Simple<float> >::~Index (this=0x7fff981c2380, 
    __in_chrg=<optimized out>) at /usr/include/flann/flann.hpp:132
#10 0x00000000004061ae in boost::detail::sp_counted_base::release() ()
---Type <return> to continue, or q <return> to quit---Quit
```

mmurooka, 2016-10-09:
> sourceからコンパイルしたpclだけでなく，aptでpclが入ってるのがダメなのでしょうか？

というのは，このPC（leus@himのことかな）では，pclがソースでは入っていなくてaptのみということでしょうか．

YuOhara, 2016-10-09:
いえ、ソースからコンパイルしてmake installしていて(usr/local以下)、それとは別にaptのpclが入っているまま、特にaptitude removeとうはしていないということです。
なので、たぶんこれまでの議論と同じことだと思います。(leus@himです)

その場合にpclが二つあるため、includeしてコンパイルするときはpclのソースを見ていて、
.soを読み込んだりするときはpclのaptの方を読み込んじゃったりするのかなと想像していました。（だから依然のときは、pclを入れたあとコンパイルしなおしたら治るのかと思いました）

wkentaro, 2016-10-09:
lddすると、1.7の方になっているということですか？

wkentaro, 2016-10-09:
CMakeLists.txtの中でPCL_INCLUDE_DIRSを見るとどちらでしょうか。
リンクもincludeも1.8になっていれば問題なさそうですが、、

wkentaro, 2016-10-10:
apcデモでもPCL1.8を使っているので
@mmurooka スクリプトで試しましたが、baxter@sheetaでは起こらないですね。
ちなみにlddの結果は以下でした。

```
% ldd libjsk_pcl_ros.so | grep pcl
        libpcl_ros_tf.so => /home/baxter/ros/jsk_apc_ws/devel/lib/libpcl_ros_tf.so (0x00007f9841154000)
        libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007f9840ea4000)
        libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007f9840b75000)
        libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 (0x00007f984079a000)
        libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007f98403e4000)
        libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007f983f60a000)
        libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007f983f1ad000)
        libpcl_surface.so.1.8 => /usr/local/lib/libpcl_surface.so.1.8 (0x00007f983e9bd000)
        libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007f983e00b000)
        libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007f983bafe000)
        libpcl_recognition.so.1.8 => /usr/local/lib/libpcl_recognition.so.1.8 (0x00007f983b35d000)
        libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007f983a50e000)
        libpcl_keypoints.so.1.8 => /usr/local/lib/libpcl_keypoints.so.1.8 (0x00007f983a13f000)
        libpcl_tracking.so.1.8 => /usr/local/lib/libpcl_tracking.so.1.8 (0x00007f9839b54000)
        libpcl_people.so.1.8 => /usr/local/lib/libpcl_people.so.1.8 (0x00007f9839936000)
        libjsk_pcl_ros_base.so => /home/baxter/ros/jsk_apc_ws/devel/lib/libjsk_pcl_ros_base.so (0x00007f9836f12000)
        libpcl_visualization.so.1.8 => /usr/local/lib/libpcl_visualization.so.1.8 (0x00007f9834150000)
        libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007f9833274000)
        libpcl_ml.so.1.8 => /usr/local/lib/libpcl_ml.so.1.8 (0x00007f983029d000)
```

mmurooka, 2016-10-10:
leus@himでの結果です．1.7と1.8が混在しているからこれが原因みたいですね．

```
$ ldd lib/libjsk_pcl_ros.so  | grep pcl
	libpcl_gpu_kinfu_large_scale.so.1.8 => /usr/local/lib/libpcl_gpu_kinfu_large_scale.so.1.8 (0x00007fd6096eb000)
	libpcl_gpu_containers.so.1 => /usr/local/lib/libpcl_gpu_containers.so.1 (0x00007fd6094e5000)
	libjsk_pcl_ros_utils.so => /home/leus/ros/indigo/devel/.private/jsk_pcl_ros_utils/lib/libjsk_pcl_ros_utils.so (0x00007fd608c6d000)
	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 (0x00007fd6065b3000)
	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 (0x00007fd606295000)
	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 (0x00007fd605f12000)
	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 (0x00007fd605b73000)
	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 (0x00007fd604f29000)
	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 (0x00007fd604afa000)
	libpcl_features.so.1.7 => /usr/lib/libpcl_features.so.1.7 (0x00007fd6031fb000)
	libpcl_segmentation.so.1.7 => /usr/lib/libpcl_segmentation.so.1.7 (0x00007fd6026d5000)
	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 (0x00007fd601e98000)
	libpcl_ros_tf.so => /home/leus/ros/indigo/devel/.private/pcl_ros/lib/libpcl_ros_tf.so (0x00007fd601bf5000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007fd601946000)
	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007fd601596000)
	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007fd600bf2000)
	libpcl_tracking.so.1.8 => /usr/local/lib/libpcl_tracking.so.1.8 (0x00007fd600606000)
	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007fd5fe14a000)
	libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007fd5fd317000)
	libpcl_surface.so.1.8 => /usr/local/lib/libpcl_surface.so.1.8 (0x00007fd5fcbec000)
	libpcl_recognition.so.1.8 => /usr/local/lib/libpcl_recognition.so.1.8 (0x00007fd5fc46f000)
	libpcl_keypoints.so.1.8 => /usr/local/lib/libpcl_keypoints.so.1.8 (0x00007fd5fc0a3000)
	libpcl_people.so.1.8 => /usr/local/lib/libpcl_people.so.1.8 (0x00007fd5fbe85000)
	libjsk_pcl_ros_base.so => /home/leus/ros/indigo/devel/.private/jsk_pcl_ros/lib/libjsk_pcl_ros_base.so (0x00007fd5f8b5f000)
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007fd5f74fe000)
	libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 (0x00007fd5f6ecc000)
	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007fd5f6114000)
	libpcl_visualization.so.1.8 => /usr/local/lib/libpcl_visualization.so.1.8 (0x00007fd5f5dde000)
	libpcl_io_ply.so.1.7 => /usr/lib/libpcl_io_ply.so.1.7 (0x00007fd5f42b4000)
	libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007fd5f2063000)
	libpcl_ml.so.1.8 => /usr/local/lib/libpcl_ml.so.1.8 (0x00007fd5f1e3f000)
	libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007fd5ee17b000)
```

baxter@sheeta にはaptのpcl-1.7は入っているでしょうか．

@YuOhara 
> buildしなおしても同じです．

は，buildとdevelを両方消してcatkin buildしなおしたのかな．

YuOhara, 2016-10-10:
> buildとdevelを両方消してcatkin buildしなおしたのかな．

すいません。そういえばdevelのjsk_pcl_ros, jsk_pcl_ros_utils しか消してなかった気がしました。

ただ、私の机のpcではcatkin clean までしてみたのですが、そちらでもエラーが出ていたと記憶しています。

mmurooka, 2016-10-10:
> ただ、私の机のpcではcatkin clean までしてみたのですが、そちらでもエラーが出ていたと記憶しています。

そっちでもlddしてみるといいかもね．

YuOhara, 2016-10-10:
himと同じように、1.7と1.8が混ざってるようにみえます。
```
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel$ ldd lib/libjsk_pcl_ros.so  | grep pcl
	libpcl_gpu_kinfu_large_scale.so.1.8 => /usr/local/lib/libpcl_gpu_kinfu_large_scale.so.1.8 (0x00007f4239616000)
	libpcl_gpu_containers.so.1 => /usr/local/lib/libpcl_gpu_containers.so.1 (0x00007f423940f000)
	libjsk_pcl_ros_utils.so => /home/leus/ros/indigo/devel/lib/libjsk_pcl_ros_utils.so (0x00007f4237729000)
	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 (0x00007f4235fd3000)
	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 (0x00007f4235cb6000)
	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 (0x00007f4235933000)
	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 (0x00007f4235593000)
	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 (0x00007f423494a000)
	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 (0x00007f423451b000)
	libpcl_features.so.1.7 => /usr/lib/libpcl_features.so.1.7 (0x00007f4232c1b000)
	libpcl_segmentation.so.1.7 => /usr/lib/libpcl_segmentation.so.1.7 (0x00007f42320f6000)
	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 (0x00007f42318b9000)
	libpcl_ros_tf.so => /home/leus/ros/indigo/devel/lib/libpcl_ros_tf.so (0x00007f42311c7000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007f4230f18000)
	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007f4230b67000)
	libpcl_surface.so.1.8 => /usr/local/lib/libpcl_surface.so.1.8 (0x00007f4230439000)
	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007f422fa93000)
	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007f422d5d2000)
	libpcl_keypoints.so.1.8 => /usr/local/lib/libpcl_keypoints.so.1.8 (0x00007f422d206000)
	libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007f422c3d0000)
	libpcl_recognition.so.1.8 => /usr/local/lib/libpcl_recognition.so.1.8 (0x00007f422bc52000)
	libpcl_people.so.1.8 => /usr/local/lib/libpcl_people.so.1.8 (0x00007f422ba35000)
	libpcl_tracking.so.1.8 => /usr/local/lib/libpcl_tracking.so.1.8 (0x00007f422b449000)
	libjsk_pcl_ros_base.so => /home/leus/ros/indigo/devel/lib/libjsk_pcl_ros_base.so (0x00007f4228a1f000)
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007f42273be000)
	libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 (0x00007f4225511000)
	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007f4224755000)
	libpcl_visualization.so.1.8 => /usr/local/lib/libpcl_visualization.so.1.8 (0x00007f422441f000)
	libpcl_io_ply.so.1.7 => /usr/lib/libpcl_io_ply.so.1.7 (0x00007f422396c000)
	libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007f422171b000)
	libpcl_ml.so.1.8 => /usr/local/lib/libpcl_ml.so.1.8 (0x00007f42208e4000)
	libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007f421df96000)
```

YuOhara, 2016-10-10:
pcl_rosは上手くいってるように見えます。（直上と同じパソコンです）
```
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd libpcl_ros_io.so  | grep pcl
	libpcl_ros_tf.so => /home/leus/ros/indigo/devel/lib/libpcl_ros_tf.so (0x00007f363788d000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007f363569e000)
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007f363524f000)
	libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007f3631a19000)
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd libpcl_ros_features.so.so  | grep pcl
ldd: ./libpcl_ros_features.so.so: No such file or directory
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd libpcl_ros_features.so  | grep pcl
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007fc27e8f4000)
	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007fc27e542000)
	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007fc27c081000)
	libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007fc27b24c000)
	libpcl_recognition.so.1.8 => /usr/local/lib/libpcl_recognition.so.1.8 (0x00007fc27aacd000)
	libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 (0x00007fc27752d000)
	libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007fc277200000)
	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007fc27685b000)
	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007fc275a9e000)
	libpcl_ml.so.1.8 => /usr/local/lib/libpcl_ml.so.1.8 (0x00007fc27587a000)
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007fc27542c000)
	libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007fc273518000)
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd libpcl_ros_segmentation.so  | grep pcl
	libpcl_ros_tf.so => /home/leus/ros/indigo/devel/lib/libpcl_ros_tf.so (0x00007fe27a57b000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007fe2785d9000)
	libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007fe2777a4000)
	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007fe274443000)
	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007fe273686000)
	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007fe272ce1000)
	libpcl_ml.so.1.8 => /usr/local/lib/libpcl_ml.so.1.8 (0x00007fe272abd000)
	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007fe2705fb000)
	libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 (0x00007fe26e98f000)
	libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007fe26e663000)
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd pcl_ros.so  | grep pcl
ldd: ./pcl_ros.so: No such file or directory
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd pcl_ros/  | grep pcl
bag_to_pcd                   convert_pointcloud_to_image  pointcloud_to_pcd            
convert_pcd_to_image         pcd_to_pointcloud            
leus@leus-HP-Z620-Workstation:~/ros/indigo/devel/lib$ ldd pcl_ros/pcd_to_pointcloud  | grep pcl
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007f285492b000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007f2851ef5000)
	libpcl_io_ply.so.1.8 => /usr/local/lib/libpcl_io_ply.so.1.8 (0x00007f2851c9c000)

```

wkentaro, 2016-10-10:
```bash
for ~/ros/indigo/devel -iname -type f; do
  ldd $f | grep pcl | grep 1.7
done
```

YuOhara, 2016-10-10:
https://gist.github.com/YuOhara/dbb9d3680176ff3a3b460c4cb18de485

wkentaro, 2016-10-10:
1.7のやつがあるのでそれを直せばいいのではと思うのですが、
それともそれがどれかわからないという状況でしょうか。

YuOhara, 2016-10-10:
あまりわかってないので申し訳ないのですが，
直すって，CMakeListsをちゃんと書くってことでしょうか

wkentaro, 2016-10-10:
1. 多分、どれがpcl1.7とリンクされているのかを調べる。
2. 1.で見つけた link (L) または executable (E) がソースで入っている pkg (P) のコンパイルによって生成されたことを確認。
3. L, Eを削除する。or catkin clean -a をするのがいいかと。
3. P を再ビルドする。

k-okada, 2016-10-13:
>    4. P を再ビルドする。

```
catkin b -v -i --make-args VERBOSE=1 2>&1 | tee compile.log
```
して.1.7という文字列を探すのが、pclの1.8を使わない、というアイデアを１番とすると、２，３，をぬいて４番目ぐらいに良いと思います．

furushchev, 2016-10-13:
```cmake
# not tested at all
if (ENV{FORCE_LINK_PCL_1_8})
find_package(PCL 1.8 EXACT REQUIRED)
else()
find_package(PCL REQUIRED)
endif()
```

という感じにするのはどうでしょうか？

wkentaro, 2016-10-13:
あれ、これまだ解決してませんか？

YuOhara, 2016-10-13:
@wkentaro　くんが、新たな1番目の方法を探してくれて、解決しました。
jsk_pcl_ros_utills > OctoMap/octomap_mapping > pcl
の順に依存していて、octomapがaptで入っていてsourceになかったのが原因でした。
octomapをsourceからコンパイルするようにすれば、pcl1.7が消えて、エラーがでなくなりました。
ありがとうございます。

closed 2016-10-13 by YuOhara

mmurooka, 2016-10-13:
@YuOhara , leus@himでも直しておいてもらえると嬉しいです．

YuOhara, 2016-10-13:
はい、明日確認します