jsk-ros-pkg/jsk_recognition #1854: release 0.4.0

k-okada, 2016-08-30:
0.3.22

- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1849
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1835
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1829
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1828

0.4.0
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1833
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1827
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1820
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1791
- [ ] https://github.com/jsk-ros-pkg/jsk_recognition/pull/1726

k-okada, 2016-09-13:
0.3.22 done  https://github.com/ros/rosdistro/pull/12681

wkentaro, 2017-02-08:
It's done. https://github.com/jsk-ros-pkg/jsk_recognition/releases/tag/1.0.0

closed 2017-02-08 by wkentaro