jsk-ros-pkg/jsk_recognition #1870: jsk_pr2_startupがpackage.xmlに含まれていない

ban-masa, 2016-09-13:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/organized_multi_plane_segmentation.launch
のSELF_FILTER_PARAMにおいてjsk_pr2_startupが使用されているのですが、package.xmlに含まれていないためjsk_pr2_startupがない場合、エラーがでて終了します。
デフォルトではSELF_FILTER_PARAMはfalseであり、実行に必須ではないため
```
<arg name="SELF_FILTER_PARAM" default="" />
```
とすることで実行できたのですが、package.xmlにjsk_pr2_startupを含めなくてよいのでしょうか。

ban-masa, 2016-09-13:
和田さんに教えてもらい、
```
<arg name="SELF_FILTER_PARAM" default="$(find jsk_pr2_startup)/jsk_pr2_sensors/tilt_self_filter.yaml" />
```
を
```
<group if="$(arg RUN_SELF_FILTER)">
```
のあとに移動させることによってエラーなく実行できました。

wkentaro, 2016-09-13:
変更をPRで送っておいてください :pray:

k-okada, 2016-09-13:
reopened until PR is merged

ban-masa, 2016-09-15:
#1873 
遅くなりましたがPR送りました

wkentaro, 2016-10-14:
Closed via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1873

closed 2016-10-14 by wkentaro