jsk-ros-pkg/jsk_recognition #1874: need to remove apt cython and install pip cython

k-okada, 2016-09-15:
installing fcn requires cython>=0.21, where apt version of cython is 0.20 and `rosdep resolve cython` returns apt. removing  cython and install `pip install cyton` manually solved problem but is there cleaner way to do this?
```
Downloading/unpacking scikit-image==0.12.3 (from fcn)
  Downloading scikit-image-0.12.3.tar.gz (20.7MB): 20.7MB downloaded
  Running setup.py (path:/tmp/pip_build_root/scikit-image/setup.py) egg_info for package scikit-image
    Unable to find pgen, not compiling formal grammar.
    No eggs found in /tmp/easy_install-QdyPq0/Cython-0.24.1/egg-dist-tmp-u24Y3Z (setup script problem?)
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_root/scikit-image/setup.py", line 146, in <module>
        **extra
      File "/usr/lib/python2.7/dist-packages/numpy/distutils/core.py", line 126, in setup
        dist = setup(**new_attr)
      File "/usr/lib/python2.7/dist-packages/numpy/distutils/core.py", line 169, in setup
        return old_setup(**new_attr)
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/numpy/distutils/numpy_distribution.py", line 16, in __init__
        Distribution.__init__(self, attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 239, in __init__
        self.fetch_build_eggs(attrs.pop('setup_requires'))
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 264, in fetch_build_eggs
        replace_conflicting=True
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 628, in resolve
        raise DistributionNotFound(req)
    pkg_resources.DistributionNotFound: cython>=0.21
    Complete output from command python setup.py egg_info:
    Unable to find pgen, not compiling formal grammar.

No eggs found in /tmp/easy_install-QdyPq0/Cython-0.24.1/egg-dist-tmp-u24Y3Z (setup script problem?)

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/scikit-image/setup.py", line 146, in <module>

    **extra

  File "/usr/lib/python2.7/dist-packages/numpy/distutils/core.py", line 126, in setup

    dist = setup(**new_attr)

  File "/usr/lib/python2.7/dist-packages/numpy/distutils/core.py", line 169, in setup

    return old_setup(**new_attr)

  File "/usr/lib/python2.7/distutils/core.py", line 111, in setup

    _setup_distribution = dist = klass(attrs)

  File "/usr/lib/python2.7/dist-packages/numpy/distutils/numpy_distribution.py", line 16, in __init__

    Distribution.__init__(self, attrs)

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 239, in __init__

    self.fetch_build_eggs(attrs.pop('setup_requires'))

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 264, in fetch_build_eggs

    replace_conflicting=True

  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 628, in resolve

    raise DistributionNotFound(req)

pkg_resources.DistributionNotFound: cython>=0.21

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/scikit-image
Storing debug log for failure in /root/.pip/pip.log
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U fcn] failed
```

wkentaro, 2016-09-16:
On my pc (Ubuntu 14.04.5, indigo), it tries to install **apt** for `python-skimage` rosdep key in jsk_recognition_utils.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_utils/package.xml#L35
```
% rosdep resolve python-skimage
#apt
python-skimage

% rosdep resolve python-skimage-pip
#pip
scikit-image

% lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.5 LTS
Release:        14.04
Codename:       trusty
```

wkentaro, 2016-09-17:
@k-okada Possibly, you need to run `rosdep update`.

k-okada, 2016-09-21:
thanks, I already have python-skimage 
```
c$ aptitude show python-skimage
Package: python-skimage                  
State: installed
Automatically installed: no
Version: 0.9.3-4build1
Priority: optional
Section: universe/python
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Uncompressed Size: 6,460 k
Depends: libfreeimage3, python-numpy, python-scipy (>= 0.10), python-skimage-lib
         (>= 0.9.3-4build1), python (>= 2.7), python (< 2.8), python:any (>=
         2.7.1-0ubuntu2)
Recommends: python-pil, python-imaging, python-matplotlib (>= 1.0), python-nose,
            python-qt4
Suggests: python-opencv, python-skimage-doc
Description: Python modules for image processing
 scikit-image is a collection of image processing algorithms for Python.  It
 performs tasks such as image loading, filtering, morphology, segmentation,
 color conversions, and transformations. 
 
 This package provides the Python 2 module.
Homepage: http://scikit-image.org
```

but pip insatll fcn try to install skimage-pip
```
Downloading/unpacking scikit-image==0.12.3 (from fcn)
  Downloading scikit-image-0.12.3.tar.gz (20.7MB): 20.7MB downloaded
  Running setup.py (path:/tmp/pip_build_root/scikit-image/setup.py) egg_info for package scikit-image
    Partial import of skimage during the build process.
    build_src
```
and failed at
```
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]

 #warning "Using deprecated NumPy API, disable it by " \

  ^

In file included from /tmp/pip_build_root/h5py/h5py/defs.c:303:0:

/tmp/pip_build_root/h5py/h5py/api_compat.h:27:18: fatal error: hdf5.h: No such file or directory

 #include "hdf5.h"

                  ^

compilation terminated.

error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

----------------------------------------
  Can't roll back h5py; was not uninstalled
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/h5py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-DFmLgk-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/h5py
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 235, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 42: ordinal not in range(128)
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U fcn] failed
```

I

k-okada, 2016-09-21:
I also have cython in
```
$ pip list | grep -i cython
Cython (0.20.1post0)
```

I'm not sure how pip works, but if you can control fcn package, could you install python library with version compatible with ubuntu 14.04?

wkentaro, 2016-09-21:
Thank u for the sharing the error messages in detail.
I think `fcn` package depends on the latest `scikit-image` package and that's why the rosdep installs `scikit-image` with apt even if it is installed via `apt`.
I can check if the version of `scikit-image` is being downgraded.

wkentaro, 2016-09-21:
BTW, the `sudo pip install -U scikit-image` fails? I think it is not supposed thing..

k-okada, 2016-09-21:
`pip install -u scikit-image` is ok
https://gist.github.com/k-okada/295951890599c8fe7e4c6d5937b0b45b
but `pip -H install -U fcn` still failing
https://gist.github.com/k-okada/91cc1b2d98d8af915f66f000b356e390

wkentaro, 2016-09-21:
I released `fcn==0.4.0` to pypi, and I think below works.
Could u please try it?
```
# sudo apt-get install liblapack-dev   # for numpy
sudo apt-get install libhdf5-dev     # for h5py
sudo apt-get install libleveldb-dev  # for plyvel

sudo -H pip install -U fcn -vvv
```

k-okada, 2016-10-01:
```
customize UnixCCompiler using build_clib

building 'dfftpack' library

error: library dfftpack has Fortran sources but no Fortran compiler found

----------------------------------------
  Can't roll back scipy; was not uninstalled
Cleaning up...
  Removing temporary dir /tmp/pip_build_root...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/scipy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-rU1v5Z-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/scipy
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 283, in run
    requirement_set.install(install_options, global_options, root=options.root_path)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1436, in install
    requirement.install(install_options, global_options, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 707, in install
    cwd=self.source_dir, filter_stdout=self._filter_install, show_stdout=False)
  File "/usr/lib/python2.7/dist-packages/pip/util.py", line 715, in call_subprocess
    % (command_desc, proc.returncode, cwd))
InstallationError: Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/scipy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-rU1v5Z-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/scipy

Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 235, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 104: ordinal not in range(128)
```

k-okada, 2016-10-01:
http://stackoverflow.com/questions/26575587/cant-install-scipy-through-pip
it seems we need latest pip, but it produced another error. 
```
k-okada@p40-yoga:~/catkin_ws/ws_jsk_recognition/src/jsk_recognition$ sudo easy_install pip
Searching for pip
Best match: pip 1.5.4
Adding pip 1.5.4 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/lib/python2.7/dist-packages
Processing dependencies for pip
Finished processing dependencies for pip
k-okada@p40-yoga:~/catkin_ws/ws_jsk_recognition/src/jsk_recognition$ which pip
/usr/local/bin/pip
k-okada@p40-yoga:~/catkin_ws/ws_jsk_recognition/src/jsk_recognition$ sudo pip install -U fcn -vvv
Getting page https://pypi.python.org/simple/fcn/
URLs to search for versions for fcn in /usr/local/lib/python2.7/dist-packages:
* https://pypi.python.org/simple/fcn/
Analyzing links from page https://pypi.python.org/simple/fcn/
  Found link https://pypi.python.org/packages/25/e7/36afaa9a8491aaaee59cf442285cc39635194529bf189cb1421cc72cecab/fcn-1.2.5.tar.gz#md5=e5b7e0d1e01aaf1bcbf23eb57a1562a9 (from https://pypi.python.org/simple/fcn/), version: 1.2.5
  Found link https://pypi.python.org/packages/32/e1/27ab9d113ebaab5ade3a5c28b474b8516faf7e80c82463926a2f43ebaec5/fcn-1.2.8.tar.gz#md5=ef70f674df81e9476a1dddb1893e43ea (from https://pypi.python.org/simple/fcn/), version: 1.2.8
  Found link https://pypi.python.org/packages/37/dd/22b01ef45559fb3840187ed56bd3e555fbe51d0571506431b88a9c2fbb24/fcn-1.1.3.tar.gz#md5=0349899fadd7bf8bdb0fc19b09e47774 (from https://pypi.python.org/simple/fcn/), version: 1.1.3
  Found link https://pypi.python.org/packages/40/df/9c96f670145d5081e235c44bf6786e5712d150184352ad96ab4fd32582ac/fcn-1.3.4.tar.gz#md5=1fde0cb7508ce10ab9bd33638dea3cfd (from https://pypi.python.org/simple/fcn/), version: 1.3.4
  Found link https://pypi.python.org/packages/52/5b/f966a60e074cb215807ce53d4958a7a4d9c12463c1e96a0cb2426c06e489/fcn-1.2.1.tar.gz#md5=2c32ee075afc3fbd6e1ad4e786b0227f (from https://pypi.python.org/simple/fcn/), version: 1.2.1
  Found link https://pypi.python.org/packages/59/03/6b64cc71f1183c70e21b51c39ce70bf54b109e139738efd96730ea507375/fcn-1.2.6.tar.gz#md5=7d5227903b311348f4212368bc970c0b (from https://pypi.python.org/simple/fcn/), version: 1.2.6
  Found link https://pypi.python.org/packages/60/a8/7ec8693bf47b9e17dd01d85426ee082150a8a209033efd5f9422c0bc6c8f/fcn-2.0.0.tar.gz#md5=995a7f2aeb77089f0f7077854f0b9036 (from https://pypi.python.org/simple/fcn/), version: 2.0.0
  Found link https://pypi.python.org/packages/65/05/a9b4e510fadd88492c8d3a5f480b59a6165f58b1c24886ba87488e15e4ef/fcn-1.3.3.tar.gz#md5=a5289619d030affb14bdd3e0e18d9b7f (from https://pypi.python.org/simple/fcn/), version: 1.3.3
  Found link https://pypi.python.org/packages/71/b8/0bad111d8f4577fc3d66aff930e8a7d84c5f77bc7448b1b21d1606ed9977/fcn-4.0.1.tar.gz#md5=0bfa5ec3079bac45c51e6c76c770480c (from https://pypi.python.org/simple/fcn/), version: 4.0.1
  Found link https://pypi.python.org/packages/80/da/1fa25ed87e6ea7d1b50b3462eb14599248a1e5f5fe5b50bfd9cdd717d729/fcn-1.3.0.tar.gz#md5=e8aa074f8dfb147dbbca88104242f559 (from https://pypi.python.org/simple/fcn/), version: 1.3.0
  Found link https://pypi.python.org/packages/95/43/916a99b6e46daab95e29326bd712a503b60ce5977841fc43f52d64aaf803/fcn-1.2.9.tar.gz#md5=8cf2587fe671551bf39b6634eaf9832a (from https://pypi.python.org/simple/fcn/), version: 1.2.9
  Found link https://pypi.python.org/packages/95/9d/3c1ec1e4e7cb559e6af89fd28208516b0961ec88133798cdb414042c4c49/fcn-3.2.0.tar.gz#md5=f37e170cfe3e072e740ef9481d95b183 (from https://pypi.python.org/simple/fcn/), version: 3.2.0
  Found link https://pypi.python.org/packages/b1/f6/e7fbc68722feffac2ded72519f06400588cda5a9df416584ac9bee38dfbf/fcn-1.2.3.tar.gz#md5=631e71312d653b3fdd40684571004c5e (from https://pypi.python.org/simple/fcn/), version: 1.2.3
  Found link https://pypi.python.org/packages/b5/1d/7ede6cf7a6a2ab13b5dcaff9d0ecf4ce1f6d10169fdf0eaf6caa16bd1677/fcn-4.0.0.tar.gz#md5=ad27f23c5fd4d355c4f09a73710f575c (from https://pypi.python.org/simple/fcn/), version: 4.0.0
  Found link https://pypi.python.org/packages/b8/bc/aed346e2295dffcd85c6cd2cb65ccb3a939833817265c41faa41956e546a/fcn-1.2.2.tar.gz#md5=4ebaf7df017e9b7ab752e0b9b3de78b5 (from https://pypi.python.org/simple/fcn/), version: 1.2.2
  Found link https://pypi.python.org/packages/c6/3f/5e913f9988a1b043081fd36b9741004a2154dc8ce4b588fcb4e11eaca60b/fcn-1.3.5.tar.gz#md5=ef22bedb29dea74776eabf0754ff2d3e (from https://pypi.python.org/simple/fcn/), version: 1.3.5
  Found link https://pypi.python.org/packages/d1/52/54cb53ceca7d23ac3252878d0379b8049444395272835e3b6733d4657d8b/fcn-1.2.7.tar.gz#md5=cc99a051534dbdfe6f352ee8be0826a8 (from https://pypi.python.org/simple/fcn/), version: 1.2.7
  Found link https://pypi.python.org/packages/d6/2a/949133371b4308390e8daacc16aff0cb1a26daa2d1332082f3fe3c469c6d/fcn-3.0.0.tar.gz#md5=c4b27acbe4d74e7fbdcdef607e377471 (from https://pypi.python.org/simple/fcn/), version: 3.0.0
  Found link https://pypi.python.org/packages/d8/ab/0e1b0f5ad400d5e37894550dbd33f64eeab95137badf74b3c400e9aeaea6/fcn-1.3.2.tar.gz#md5=4c0f027e3b1c6aa7a55a9757cda32310 (from https://pypi.python.org/simple/fcn/), version: 1.3.2
  Found link https://pypi.python.org/packages/db/60/09e2d21b5a778b1bb59d12bcd1e418e79ecaecfae67822529217ada23521/fcn-3.1.0.tar.gz#md5=d987248c7070e4d7a3ed14e283ce1192 (from https://pypi.python.org/simple/fcn/), version: 3.1.0
[... cut ...]
  Found link https://pypi.python.org/packages/fc/54/87b79129ed1c2f5a6a5ee1b9d1b0492f422305ef75da16843175b91488fb/numpy-1.9.1.zip#md5=223532d8e1bdaff5d30936439701d6e1 (from https://pypi.python.org/simple/numpy/), version: 1.9.1
  Skipping https://pypi.python.org/packages/fd/0c/94d9583da9d5b141c078b7c69e0fcf367dea62d46040f724b5433825a59e/numpy-1.7.2-cp26-cp26m-manylinux1_x86_64.whl#md5=d6926263d135dfb3106c1f720ac2a6f5 (from https://pypi.python.org/simple/numpy/) because it is not compatible with this Python
  Skipping https://pypi.python.org/packages/fd/47/70fe002303661db1a87293244c15ac5d066

k-okada, 2016-10-01:
I see, 
```
jsk_recognition_utils/package.xml:  <build_depend>cython</build_depend>
```
so if you run rosdep install, it installs cython from apt and that is version 0.20, which did not work on installing scikit-image and that also happens on travis
```
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U -q fcn] failed
  pip: Failed to detect successful installation of [fcn]
+ EXIT_STATUS=1`
```
https://s3.amazonaws.com/archive.travis-ci.org/jobs/163738513/log.txt

The reason we run `rosdep` with `-r`  is because some packages has deprecated tag to keep one branch for different distros, so if you're confident with list of package.xml, we should  run rosdep without -r travis.

k-okada, 2016-10-01:
Summarize, 

1. pip install of latest scipy depends on latest pip (http://stackoverflow.com/questions/26575587/cant-install-scipy-through-pip), even 0.6.7 of jsk_travis did not work. btw can we write pip to build_depends or requirements.txt?
```
    customize UnixCCompiler using build_clib
    building 'dfftpack' library\
    error: library dfftpack has Fortran sources but no Fortran compiler found
    ----------------------------------------
    Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-VYEGgX/scipy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-F3Iufe-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-VYEGgX/scipy
```

2. pip install of scikit-image requires cython 0.23, but ubuntu 13.03 installs cython 0.20 ( http://packages.ubuntu.com/search?keywords=cython) and it seems if we already install cythin from apt, it does not install from pip.
```
  File "skimage/_build.py", line 30, in cython
    raise RuntimeError('Cython >= 0.23 needed to build scikit-image')
RuntimeError: Cython >= 0.23 needed to build scikit-image
```

k-okada, 2016-10-01:
Proposed solutions

1. do not write code depends on latest pip code
2. add depends to cython pip in jsk_perception, or ask scikit-image maintainer to  specify cython>=0.23 to requirement.txt
3. add pip to rosdep entrry if possible (this will run code like pip install -U pip)

wkentaro, 2016-10-01:
Thanks, I will look around them.

wkentaro, 2016-10-17:
> do not write code depends on latest pip code

Actually, I tried to remove the dependency on the latest scikit-image by fcn, but realized that it is hard.
And I prefer the third option, because the trouble in jsk-ros-pkg/jsk_travis#208 was caused by Travis not pip, I think.

k-okada, 2016-10-18:
is there any tools that display pip/python dependencies?

>
https://github.com/ros-visualization/rqt_robot_plugins/pull/112#issuecomment-254031001

its ok to use latest scikit-image, my concern is whether that module brings
other python module which conflicts with deb provided package.

k-okada, 2016-10-27:
@wkentaro  still have error

```
>   pip ['gdown', 'fcn', 'chainer']
executing command [sudo -H pip install -U chainer]
Collecting chainer
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Downloading chainer-1.17.0.tar.gz (1.1MB)
    100% |████████████████████████████████| 1.1MB 1.2MB/s 
    Complete output from command python setup.py egg_info:
    running egg_info
    creating pip-egg-info/chainer.egg-info
    writing requirements to pip-egg-info/chainer.egg-info/requires.txt
    writing pip-egg-info/chainer.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/chainer.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/chainer.egg-info/dependency_links.txt
    writing manifest file 'pip-egg-info/chainer.egg-info/SOURCES.txt'
    warning: manifest_maker: standard file '-c' not found
    
    Executing cythonize
    Options: {'profile': False, 'annotate': False, 'linetrace': False, 'no_cuda': False}
    **************************************************
    *** WARNING: nvcc not in path.
    *** WARNING: Please set path to nvcc.
    **************************************************
    Include directories: []
    Library directories: []
    creating /tmp/tmpVTOODo/tmp
    creating /tmp/tmpVTOODo/tmp/tmpVTOODo
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -c /tmp/tmpVTOODo/a.cpp -o /tmp/tmpVTOODo/tmp/tmpVTOODo/a.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    /tmp/tmpVTOODo/a.cpp:1:23: fatal error: cublas_v2.h: No such file or directory
     #include <cublas_v2.h>
                           ^
    compilation terminated.
    **************************************************
    *** WARNING: Include files not found: ['cublas_v2.h', 'cuda.h', 'cuda_profiler_api.h', 'cuda_runtime.h', 'curand.h', 'nvToolsExt.h']
    *** WARNING: Skip installing cuda support
    *** WARNING: Check your CPATH environment variable
    **************************************************
    Include directories: []
    Library directories: []
    creating /tmp/tmp4Ghd83/tmp
    creating /tmp/tmp4Ghd83/tmp/tmp4Ghd83
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -c /tmp/tmp4Ghd83/a.cpp -o /tmp/tmp4Ghd83/tmp/tmp4Ghd83/a.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    /tmp/tmp4Ghd83/a.cpp:1:19: fatal error: cudnn.h: No such file or directory
     #include <cudnn.h>
                       ^
    compilation terminated.
    **************************************************
    *** WARNING: Include files not found: ['cudnn.h']
    *** WARNING: Skip installing cudnn support
    *** WARNING: Check your CPATH environment variable
    **************************************************
    cython path:/usr/local/lib/python2.7/dist-packages
    error: Command '/usr/bin/python' failed:
    
      command: /usr/bin/python /usr/local/lib/python2.7/dist-packages/cython.py --version
      return code: 2
      output:
    
    /usr/bin/python: can't open file '/usr/local/lib/python2.7/dist-packages/cython.py': [Errno 2] No such file or directory
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-xJYfND/chainer/
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U chainer] failed
```

wkentaro, 2016-10-27:
> is there any tools that display pip/python dependencies?
ros-visualization/rqt_robot_plugins#112 (comment)

```
% sudo pip install pipdeptree
% pipdeptree -p fcn  
Warning!!! Possibly conflicting dependencies found:
* gshell==4.1.1
 - PyYAML [required: ==3.11, installed: 3.12]
* gotenshita==2.0.2
 - requests [required: ==2.8.1, installed: 2.11.1]
------------------------------------------------------------------------
Warning!! Cyclic dependencies found:
* Sphinx => sphinx-rtd-theme => Sphinx
* sphinx-rtd-theme => sphinx => sphinx-rtd-theme
------------------------------------------------------------------------
fcn==4.0.1
  - chainer [required: >=1.9.1, installed: 1.16.0]
    - filelock [required: None, installed: 2.0.6]
    - nose [required: None, installed: 1.3.7]
    - numpy [required: >=1.9.0, installed: 1.11.2]
    - protobuf [required: None, installed: 3.1.0.post1]
      - setuptools [required: None, installed: 28.6.1]
      - six [required: >=1.9, installed: 1.10.0]
    - six [required: >=1.9.0, installed: 1.10.0]
  - gdown [required: >=2.0.0, installed: 2.2.0]
  - h5py [required: >=2.2.1, installed: 2.6.0]
    - numpy [required: >=1.6.1, installed: 1.11.2]
    - six [required: None, installed: 1.10.0]
  - matplotlib [required: >=1.3.1, installed: 1.5.3]
    - cycler [required: None, installed: 0.10.0]
      - six [required: None, installed: 1.10.0]
    - numpy [required: >=1.6, installed: 1.11.2]
    - pyparsing [required: !=2.0.0,>=1.5.6,!=2.0.4,!=2.1.2, installed: 2.1.10]
    - python-dateutil [required: None, installed: 2.5.3]
      - six [required: >=1.5, installed: 1.10.0]
    - pytz [required: None, installed: 2016.7]
  - numpy [required: >=1.8.2, installed: 1.11.2]
  - plyvel [required: >=0.9, installed: 0.9]
  - scikit-image [required: >=0.12.3, installed: 0.12.3]
    - dask [required: >=0.5.0, installed: 0.11.1]
    - matplotlib [required: >=1.3.1, installed: 1.5.3]
      - cycler [required: None, installed: 0.10.0]
        - six [required: None, installed: 1.10.0]
      - numpy [required: >=1.6, installed: 1.11.2]
      - pyparsing [required: !=2.0.0,>=1.5.6,!=2.0.4,!=2.1.2, installed: 2.1.10]
      - python-dateutil [required: None, installed: 2.5.3]
        - six [required: >=1.5, installed: 1.10.0]
      - pytz [required: None, installed: 2016.7]
    - networkx [required: >=1.8, installed: 1.11]
      - decorator [required: >=3.4.0, installed: 4.0.10]
    - pillow [required: >=2.1.0, installed: 3.4.2]
    - scipy [required: >=0.9.0, installed: 0.18.1]
    - six [required: >=1.7.3, installed: 1.10.0]
  - scikit-learn [required: >=0.14.1, installed: 0.18]
  - scipy [required: >=0.13.3, installed: 0.18.1]
  - tqdm [required: >=4.5.0, installed: 4.8.4]
```