jsk-ros-pkg/jsk_recognition #1882: List of rostest that is unstable on Travis/Jenkins testing

wkentaro, 2016-09-18:


k-okada, 2016-09-29:
another example https://s3.amazonaws.com/archive.travis-ci.org/jobs/163112491/log.txt
```
jsk_pcl_ros/test_results/jsk_pcl_ros/rosunit-test_color_based_region_growing_segmentation.xml: 1 tests, 0 errors, 1 failures
jsk_perception/test_results/jsk_perception/rosunit-test_label_image_decomposer.xml: 1 tests, 0 errors, 1 failures
```

k-okada, 2016-10-03:
```
[jsk_perception:make] [WARN] [WallTime: 1475423926.305871] Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[jsk_perception:make] terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
[jsk_perception:make]   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator LabelEncoder from version pre-0.18 when using version 0.18. This might lead to breaking code or invalid results. Use at your own risk.
[jsk_perception:make]   UserWarning)
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator LogisticRegression from version pre-0.18 when using version 0.18. This might lead to breaking code or invalid results. Use at your own risk.
[jsk_perception:make]   UserWarning)
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator NearestNeighbors from version pre-0.18 when using version 0.18. This might lead to breaking code or invalid results. Use at your own risk.
[jsk_perception:make]   UserWarning)
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:111: UserWarning: Possible precision loss when converting from float64 to uint8
[jsk_perception:make]   "%s to %s" % (dtypeobj_in, dtypeobj))
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:111: UserWarning: Possible precision loss when converting from float64 to uint8
[jsk_perception:make]   "%s to %s" % (dtypeobj_in, dtypeobj))
[jsk_perception:make] [ERROR] [WallTime: 1475423963.494207] bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x2b66f0608650>>
[jsk_perception:make] Traceback (most recent call last):
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
[jsk_perception:make]     cb(msg)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 74, in callback
[jsk_perception:make]     self.signalMessage(msg)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
[jsk_perception:make]     cb(*(msg + args))
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 199, in add
[jsk_perception:make]     self.signalMessage(*msgs)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
[jsk_perception:make]     cb(*(msg + args))
[jsk_perception:make]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 129, in _apply
[jsk_perception:make]     result_img_pil = PIL.Image.open(f)
[jsk_perception:make]   File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2028, in open
[jsk_perception:make]     raise IOError("cannot identify image file")
[jsk_perception:make] IOError: cannot identify image file
[jsk_perception:make] 
[jsk_perception:make] terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
[jsk_perception:make]   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[jsk_perception:make] testtest_sklearn_classifier ... ok
[jsk_perception:make] 
[jsk_perception:make] [ROSTEST]-----------------------------------------------------------------------
[jsk_perception:make] 
[jsk_perception:make] [jsk_perception.rosunit-test_sklearn_classifier/test_published][FAILURE]-------
```
any idea?
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/164358073

wkentaro, 2016-10-05:
```
jsk_perception/test_results/jsk_perception/rosunit-test_bof_histogram_extractor.xml: 1 tests, 0 errors, 1 failures

<?xml version="1.0" encoding="utf-8"?>

<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="30.069">

  <testcase classname="__main__.TestTopicPublished" name="test_published" time="30.0663">

    <failure type="AssertionError">/bof_histogram_extractor/output is not published

  File "/usr/lib/python2.7/unittest/case.py", line 331, in run

    testMethod()

  File "/workspace/ros/ws_jsk_recognition/src/jsk-ros-pkg/jsk_common/jsk_tools/src/test_topic_published.py", line 56, in test_published

    assert_true(checker.check(), msg)

  File "/usr/lib/python2.7/unittest/case.py", line 424, in assertTrue

    raise self.failureException(msg)

    </failure>

  </testcase>

  <system-out><![CDATA[ Checking /bof_histogram_extractor/output for 30 seconds

]]></system-out>

  <system-err><![CDATA[WARNING: topic [/bof_histogram_extractor/output] does not appear to be published yet

]]></system-err>

</testsuite>
```

mmurooka, 2016-10-06:
https://s3.amazonaws.com/archive.travis-ci.org/jobs/165166786/log.txt
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/165166786
```
[jsk_perception:make] -- run_tests.py: verify result "/workspace/ros/ws_jsk_recognition/build/jsk_perception/test_results/jsk_perception/rostest-test_split_fore_background.xml"
[jsk_perception:make] make[3]: Leaving directory `/workspace/ros/ws_jsk_recognition/build/jsk_perception'
[jsk_perception:make] /usr/bin/cmake -E cmake_progress_report /workspace/ros/ws_jsk_recognition/build/jsk_perception/CMakeFiles
[jsk_perception:make] Built target _run_tests_jsk_perception_rostest_test_split_fore_background.test
[jsk_perception:make] [WARN] [WallTime: 1475723932.226069] Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:111: UserWarning: Possible precision loss when converting from float64 to uint8
[jsk_perception:make]   "%s to %s" % (dtypeobj_in, dtypeobj))
[jsk_perception:make] /usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:111: UserWarning: Possible precision loss when converting from float64 to uint8
[jsk_perception:make]   "%s to %s" % (dtypeobj_in, dtypeobj))
[jsk_perception:make] [ERROR] [WallTime: 1475723938.097561] bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x2afb18ad2f90>>
[jsk_perception:make] Traceback (most recent call last):
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
[jsk_perception:make]     cb(msg)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 74, in callback
[jsk_perception:make]     self.signalMessage(msg)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
[jsk_perception:make]     cb(*(msg + args))
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 199, in add
[jsk_perception:make]     self.signalMessage(*msgs)
[jsk_perception:make]   File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
[jsk_perception:make]     cb(*(msg + args))
[jsk_perception:make]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 129, in _apply
[jsk_perception:make]     result_img_pil = PIL.Image.open(f)
[jsk_perception:make]   File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2028, in open
[jsk_perception:make]     raise IOError("cannot identify image file")
[jsk_perception:make] IOError: cannot identify image file
[jsk_perception:make]
[jsk_perception:make] testtest_label_image_decomposer ... ok
[jsk_perception:make]
[jsk_perception:make] [ROSTEST]-----------------------------------------------------------------------
[jsk_perception:make]
[jsk_perception:make] [jsk_perception.rosunit-test_label_image_decomposer/test_published][FAILURE]----
[jsk_perception:make] /label_image_decomposer_with_names/output/label_viz is not published
[jsk_perception:make]   File "/usr/lib/python2.7/unittest/case.py", line 331, in run
[jsk_perception:make]     testMethod()
[jsk_perception:make]   File "/workspace/ros/ws_jsk_recognition/src/jsk-ros-pkg/jsk_common/jsk_tools/src/test_topic_published.py", line 56, in test_published
[jsk_perception:make]     assert_true(checker.check(), msg)
[jsk_perception:make]   File "/usr/lib/python2.7/unittest/case.py", line 424, in assertTrue
[jsk_perception:make]     raise self.failureException(msg)
[jsk_perception:make] --------------------------------------------------------------------------------
[jsk_perception:make]
[jsk_perception:make]
[jsk_perception:make] SUMMARY
[jsk_perception:make]  * RESULT: FAIL
[jsk_perception:make]  * TESTS: 1
[jsk_perception:make]  * ERRORS: 0
[jsk_perception:make]  * FAILURES: 1
```

wkentaro, 2016-10-06:
Please update the top comment, also.

mmurooka, 2016-10-06:
added.

wkentaro, 2016-10-17:
Seems fixed.

closed 2016-10-17 by wkentaro