jsk-ros-pkg/jsk_recognition #1889: could not run rostest-test_label_image_decomposer.xml

k-okada, 2016-10-01:
```
$ rostest label_image_decomposer.test 
... logging to /home/k-okada/.ros/log/rostest-p40-yoga-3689.log
[ROSUNIT] Outputting test results to /home/k-okada/.ros/test_results/jsk_perception/rostest-test_label_image_decomposer.xml
INFO: cannot create a symlink to latest log directory: [Errno 2] No such file or directory: '/home/k-okada/.ros/log/latest'
/usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:110: UserWarning: Possible precision loss when converting from float64 to uint8
  "%s to %s" % (dtypeobj_in, dtypeobj))
/usr/local/lib/python2.7/dist-packages/skimage/util/dtype.py:110: UserWarning: Possible precision loss when converting from float64 to uint8
  "%s to %s" % (dtypeobj_in, dtypeobj))
[ERROR] [WallTime: 1475308531.550940] bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f22c9c79290>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 74, in callback
    self.signalMessage(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 199, in add
    self.signalMessage(*msgs)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
    cb(*(msg + args))
  File "/home/k-okada/catkin_ws/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 130, in _apply
    result_img = fromimage(result_img_pil, mode='RGB')
TypeError: fromimage() got an unexpected keyword argument 'mode'

^Ctesttest_label_image_decomposer ... ERROR!
```

k-okada, 2016-10-01:
https://docs.scipy.org/doc/scipy-0.13.0/reference/generated/scipy.misc.fromimage.html#scipy.misc.fromimage
http://docs.scipy.org/doc/scipy/reference/generated/scipy.misc.fromimage.html#scipy.misc.fromimage
seems `fromimage(mode='RGB') is not wok on deb version of scipy.

k-okada, 2016-10-01:
travis ないではpipがつかわれているんだろうか＿
```
Warnings   << jsk_recognition_utils:make /workspace/ros/ws_jsk_recognition/logs/jsk_recognition_utils/build.make.000.log

In file included from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1777:0,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,

                 from /workspace/ros/ws_jsk_recognition/build/jsk_recognition_utils/python/jsk_recognition_utils/color.c:352:

/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]

 #warning "Using deprecated NumPy API, disable it by " \

  ^

In file included from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1777:0,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,

                 from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,

                 from /workspace/ros/ws_jsk_recognition/build/jsk_recognition_utils/python/jsk_recognition_utils/nms.c:352:

/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]

 #warning "Using deprecated NumPy API, disable it by " \

  ^
```
rosdistro/rosdep$ grep scipy python.yaml  | grep trusty
    trusty: [python-scipy]
    trusty_python3: [python3-scipy]
```
なので、scipyをpipからインストールするrosdepはない、けど、どこか別のpipプログラムを最初にインストールすると依存関係で、pipのscipyになっているのかな．

wkentaro, 2016-10-01:
多分travis内ではscikit-imageがscipyをhttps://github.com/scikit-image/scikit-image/blob/master/requirements.txt#L3
入れていそうです。
fcn -> scikit-image -> scipy
という依存関係で。

PR #1890 (2016-10-13): Stop using scipy fromimage that is not supported by apt version
Close #1889
closed 2016-10-13 by k-okada