jsk-ros-pkg/jsk_recognition #1919: Compile error on checkerboard_detector

wkentaro, 2016-10-24:
Related to https://github.com/jsk-ros-pkg/jsk_recognition/pull/1893

@mmurooka Could you please fix this? I think we should declare dependency on `XX_gencfg` before building the executable.

```
Errors     << checkerboard_detector:make /home/wkentaro/ros/indigo/logs/checkerboard_detector/build.make.000.log              
/home/wkentaro/ros/indigo/src/jsk-ros-pkg/jsk_recognition/checkerboard_detector/src/checkerboard_detector.cpp:54:62: fatal error: checkerboard_detector/CheckerboardDetectorConfig.h: No such file or directory
 #include <checkerboard_detector/CheckerboardDetectorConfig.h>
                                                              ^
compilation terminated.
The bug is not reproducible, so it is likely a hardware or OS problem.
make[2]: *** [CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o] Error 1
make[1]: *** [CMakeFiles/checkerboard_detector.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

mmurooka, 2016-10-24:
Sorry for trouble. Fixed in https://github.com/jsk-ros-pkg/jsk_recognition/pull/1920.

wkentaro, 2016-10-30:
Closed via #1920

closed 2016-10-30 by wkentaro