jsk-ros-pkg/jsk_recognition #1932: Rectified fisheye image

tongtybj, 2016-10-26:
@wkentaro 
The goal is to achieve the virtual gimbal as the [parrot bebop](http://gigazine.net/news/20140603-parrot-bebop-drone/) does.

Aright Now, [jsk_perception/fisheye_to_panorama.cpp](/https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/fisheye_to_panorama.cpp) has functions of: 1) panorama, 2) rectification which is orthogonal with center(optical) axis. 

I want to extend the second function which can get the rectifed image which is  orthogonal with any incident angle.

tongtybj, 2016-10-26:
Right now I send a pull request #1931 which is related to this issue.

I tested with a fisheye camera(pointgrey chamelon3, 1280x0124, 220deg)
[2016-10-21-fisheye-rectified-opencv3-fisheyemodel](https://drive.google.com/open?id=0BwphddqBjVJ_MnRzUFNobWdidlE)

```
$ roslaunch jsk_perception rectified_fisheye_image.launch
$ rosbag play 2016-10-21-fisheye-rectified-opencv3-fisheyemodel.bag
```

wkentaro, 2016-10-26:
After googling for a while, it seems that we need to calibrate the fisheye camera to get more rectified region of the frame, but camera_calibration ROS package does not support it currently.

- OpenCV3 supports fisheye camera model: 
  - http://docs.opencv.org/master/db/d58/group__calib3d__fisheye.html
  - https://github.com/opencv/opencv/blob/05b15943d6a42c99e5f921b7dbaa8323f3c042c6/samples/cpp/tutorial_code/calib3d/camera_calibration/camera_calibration.cpp
- GitHub Issue on image_pipeline to introduce fisheye camera calibration:
  - https://github.com/ros-perception/image_pipeline/issues/146

DavidTorresOcana, 2018-08-14:
Hello
Perhaps too late, but I added this functionality in this fork: image_pipeline

You can calibrate (with a tool) and rectify fisheye cameras.

The results for high FOV (+180deg) are better that those achieved with the current implementation