jsk-ros-pkg/jsk_recognition #1958: Mis-spelled node name: x background_substraction o background_subtraction in jsk_perception

wkentaro, 2016-11-11:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/CMakeLists.txt#L186

https://www.google.co.jp/search?espv=2&q=substraction+%E6%84%8F%E5%91%B3&oq=substraction+%E6%84%8F%E5%91%B3&gs_l=serp.3..0i10k1j0i8i7i30k1.5517.5517.0.5739.1.1.0.0.0.0.80.80.1.1.0....0...1.1.64.serp..0.1.79.CGhLlK_Z9yw

Ridhwanluthra, 2018-02-27:
I don't think this has been fixed yet.
Is there any particular reason?
It seems to be a fairly straightforward task, if its because no one got to it. i can do it

knorth55, 2018-02-28:
This is because of backcompatibility.
We are worried about the effect of changing the node name.
AND also, 'substraction' is not wrong but old term.