jsk-ros-pkg/jsk_recognition #1966: boxのみでpointcloudをclipするnodeが欲しい

h-kamada, 2016-11-21:
attention clipperは、inputにcamera_infoを必要とするが、laser scanなどを起源とするような点群に対してもboxを指定して切り取るようなことが可能であると良い

h-kamada, 2016-11-21:
rosbag file: https://drive.google.com/open?id=0B8bpI9nrQ7y3UjQyRUpDY2gxb2s

h-kamada, 2016-11-21:
https://drive.google.com/open?id=0B8bpI9nrQ7y3VkcwTHZJWElOYkk

k-okada, 2016-11-21:
box を clip して何がしたいのでしょうか．

wkentaro, 2016-11-21:
There is no boxes, and the frame_id is base_link.. Why?

[image]

wkentaro, 2016-11-21:
Done.
[image]
https://github.com/wkentaro/jsk_recognition/commits/h-kamada

h-kamada, 2016-11-25:
base_scanの点群を使ってテーブルの位置認識を行うeusのコードを用いようとしていますが、テーブル周辺に人の足や椅子が多数あるとテーブルの位置に関する誤認識が起こります。
それを防ぐために、/mapにおいてテーブル周辺のみの点群を切り取り、誤認識を防ぐことを考えています。

wkentaro, 2016-11-25:
@h-kamada ドキュメントでCameraInfoが必要と書いてあるのは間違っていて、PointCloudを使う場合はPointCloudが来ていればOKということだったと思うので、そのように更新お願いします。
http://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_pcl_ros/nodes/attention_clipper.html#subscribing-topic

h-kamada, 2016-11-25:
PR -> #1971

closed 2016-11-25 by h-kamada