jsk-ros-pkg/jsk_recognition #1980: Migrate srv files from jsk_pcl_ros to jsk_recognition_msgs

wkentaro, 2016-12-05:
Moved from https://github.com/jsk-ros-pkg/jsk_recognition/pull/1827#issuecomment-264796837

## TODOs for migration

- [x] (by https://github.com/jsk-ros-pkg/jsk_recognition/pull/1914) Copy .srv files to jsk_recognition_msgs 
- [x] (by https://github.com/jsk-ros-pkg/jsk_recognition/pull/1827#issuecomment-256150923) Migrate all jsk_pcl_ros/*.srv to jsk_reocgnition_msgs/*.srv for codes under jsk-ros-pkg or start-jsk.
- [ ] Remove .srv files in jsk_pcl_ros

## Background

Some packages in jsk_visualization depends on jsk_pcl_ros because it uses the srv files.
But jsk_pcl_ros is a very large package and the dependency should be avoided ideally.
The solution is placing msg/srv files into jsk_recognition_msgs.

## Plans

Merge this after jsk_recognition_msgs is released: http://repositories.ros.org/status_page/ros_indigo_default.html?q=jsk_recognition_msgs

## Related PRs

jsk-ros-pkg/jsk_recognition#1827
jsk-ros-pkg/jsk_recognition#1914

k-okada, 2016-12-06:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1983 I think this is the goal of this work, currently we have same file in different location and this is not a good situation.
 and  jsk-ros-pkg/jsk_visualization#623 is side effect

k-okada, 2016-12-13:
Proposed process was not correct -> http://build.ros.org/job/Ibin_uT32__jsk_pcl_ros_utils__ubuntu_trusty_i386__binary/74/console

```
10:42:29 make[4]: Entering directory `/tmp/binarydeb/ros-indigo-jsk-pcl-ros-utils-1.0.0/obj-i686-linux-gnu'
10:42:29 ../scripts/install_sample_data.py
10:42:29 Traceback (most recent call last):
10:42:29   File "../scripts/install_sample_data.py", line 3, in <module>
10:42:29     from jsk_data import download_data
10:42:29 ImportError: No module named jsk_data
10:42:29 make[4]: *** [CMakeFiles/install_sample_data] Error 1
10:42:29 make[4]: Leaving directory `/tmp/binarydeb/ros-indigo-jsk-pcl-ros-utils-1.0.0/obj-i686-linux-gnu'
10:42:29 make[3]: Leaving directory `/tmp/binarydeb/ros-indigo-jsk-pcl-ros-utils-1.0.0/obj-i686-linux-gnu'
10:42:29 make[3]: *** [CMakeFiles/install_sample_data.dir/all] Error 2
10:42:29 make[2]: *** [all] Error 2
10:42:29 make[2]: Leaving directory `/tmp/binarydeb/ros-indigo-jsk-pcl-ros-utils-1.0.0/obj-i686-linux-gnu'
10:42:29 dh_auto_build: make -j1 returned exit code 2
10:42:29 make[1]: *** [override_dh_auto_build] Error 2
10:42:29 make[1]: Leaving directory `/tmp/binarydeb/ros-indigo-jsk-pcl-ros-utils-1.0.0'
10:42:29 make: *** [build] Error 2
10:42:29 dpkg-buildpackage: error: debian/rules build gave error exit status 2
10:42:29 E: Building failed
10:42:29 Traceback (most recent call last):
10:42:29   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
10:42:29     subprocess.check_call(cmd, cwd=source_dir)
10:42:29   File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
10:42:29     raise CalledProcessError(retcode, cmd)
10:42:29 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-indigo-jsk-pcl-ros-utils']' returned non-zero exit status 1
```

k-okada, 2016-12-15:
something wrong on jsk_demos
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/trusty-travis-jsk-ros-pkg-jsk_demos-indigo-deb-true-ros-indigo-convex-decomposition-ros-indigo-ivcon-true/82/console
```
[drc_task_common:cmake] -- [roseus.cmake] drc_task_common depends on audio_common_msgs;std_msgs;actionlib_msgs;bond;diagnostic_msgs;geometry_msgs;jsk_footstep_msgs;jsk_hark_msgs;move_base_msgs;nav_msgs;octomap_msgs;people_msgs;pr2_msgs;rosgraph_msgs;roscpp;actionlib;actionlib_tutorials;dynamic_reconfigure;ml_classifiers;nodelet;roscpp_tutorials;sensor_msgs;jsk_gui_msgs;jsk_network_tools;map_msgs;openni2_camera;pcl_msgs;jsk_recognition_msgs;posedetection_msgs;shape_msgs;object_recognition_msgs;smach_msgs;sound_play;std_srvs;stereo_msgs;tf;dynamic_tf_publisher;image_view2;jsk_rqt_plugins;laser_assembler;tf2_msgs;topic_tools;jsk_topic_tools;opencv_apps;trajectory_msgs;control_msgs;moveit_msgs;pr2_controllers_msgs;view_controller_msgs;jsk_rviz_plugins;visualization_msgs;jsk_footstep_planner;jsk_footstep_controller;jsk_perception;roseus_remote;drc_com_common;roseus_smach
[drc_task_common:cmake] -- [roseus.cmake] drc_task_common will compile bond;jsk_footstep_msgs;jsk_hark_msgs;octomap_msgs;people_msgs;ml_classifiers;nodelet;roscpp_tutorials;jsk_gui_msgs;jsk_network_tools;map_msgs;openni2_camera;pcl_msgs;jsk_recognition_msgs;posedetection_msgs;object_recognition_msgs;smach_msgs;stereo_msgs;dynamic_tf_publisher;image_view2;jsk_rqt_plugins;laser_assembler;jsk_topic_tools;opencv_apps;moveit_msgs;view_controller_msgs;jsk_rviz_plugins;drc_com_common
[jsk_demo_common:cmake] -- set EUSDIR to /opt/ros/indigo/share/euslisp/jskeus/eus
[jsk_demo_common:cmake] -- set ARCHDIR to Linux64
[jsk_demo_common:cmake] -- Generating .msg files for action jsk_demo_common/DemoCommon /workspace/ros/ws_jsk_demos/src/jsk_demos/jsk_demo_common/action/DemoCommon.action
[jsk_demo_common:cmake] Generating for action DemoCommon
[jsk_demo_common:cmake] -- jsk_demo_common: 9 messages, 0 services
[jsk_demo_common:cmake] -- Configuring done
[elevator_move_base_pr2:make] [100%] Built target elevator_move_base_pr2_generate_messages_eus
[elevator_move_base_pr2:make] Scanning dependencies of target elevator_move_base_pr2_generate_messages
[elevator_move_base_pr2:make] [100%] Built target elevator_move_base_pr2_generate_messages
[jsk_demo_common:cmake] -- Generating done
[jsk_demo_common:cmake] -- Build files have been written to: /workspace/ros/ws_jsk_demos/build/jsk_demo_common
[elevator_move_base_pr2:make] Linking CXX executable /workspace/ros/ws_jsk_demos/devel/.private/elevator_move_base_pr2/lib/elevator_move_base_pr2/color_point_detector
[jsk_demo_common:make] Scanning dependencies of target bond_generate_messages_eus
[jsk_demo_common:make] Scanning dependencies of target _jsk_demo_common_generate_messages_check_deps_EventHeader
[jsk_demo_common:make] Scanning dependencies of target std_msgs_generate_messages_lisp
[jsk_demo_common:make] Scanning dependencies of target actionlib_msgs_generate_messages_lisp
[jsk_demo_common:make] Scanning dependencies of target image_view2_generate_messages_eus
[jsk_demo_common:make] Scanning dependencies of target dynamic_tf_publisher_generate_messages_eus
[jsk_demo_common:make] [  0%] [  0%] Built target std_msgs_generate_messages_lisp
[jsk_demo_common:make] [  0%] Built target actionlib_msgs_generate_messages_lisp
[jsk_demo_common:make] Generating EusLisp code from bond/Constants.msg
[jsk_demo_common:make] [  0%] [  0%] Scanning dependencies of target geometry_msgs_generate_messages_lisp
[jsk_demo_common:make] Scanning dependencies of target _jsk_demo_common_generate_messages_check_deps_DemoCommonFeedback
[jsk_demo_common:make] Generating EusLisp code from bond/Constants.msg
[jsk_demo_common:make] Generating EusLisp code from bond/Constants.msg
[jsk_demo_common:make] [  0%] Built target geometry_msgs_generate_messages_lisp
[jsk_demo_common:make] [  0%] Generating EusLisp code from bond/Status.msg
[jsk_demo_common:make] [  0%] Built target _jsk_demo_common_generate_messages_check_deps_EventHeader
[jsk_demo_common:make] [  0%] Generating EusLisp manifest code for bond
[jsk_demo_common:make] [  0%] Built target _jsk_demo_common_generate_messages_check_deps_DemoCommonFeedback
[jsk_demo_common:make] Scanning dependencies of target _jsk_demo_common_generate_messages_check_deps_DemoCommonActionGoal
[jsk_demo_common:make] [  0%] Generating EusLisp code from bond/Status.msg
[jsk_demo_common:make] [  0%] [  0%] Generating EusLisp code from bond/Status.msg
[jsk_demo_common:make] Generating EusLisp manifest code for bond
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/Footstep.msg
[jsk_demo_common:make] [  0%] Generating EusLisp manifest code for bond
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/FootstepArray.msg
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/PlanFootstepsAction.msg
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/PlanFootstepsActionGoal.msg
[elevator_move_base_pr2:make] [100%] Built target color_point_detector
Finished <<< elevator_move_base_pr2                      [ 13.2 seconds ]
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/PlanFootstepsActionResult.msg
[jsk_demo_common:make] [  0%] Generating EusLisp code from jsk_footstep_msgs/PlanFootstepsActionFeedback.msg
[... cut ...]
[drc_task_common:cmake] Call Stack (most recent call first):
[drc_task_common:cmake]   CMakeLists.txt:46 (generate_messages)
[drc_task_common:cmake] 
[drc_task_common:cmake] 
[drc_task_common:cmake] -- Configuring incomplete, errors occurred!
[drc_task_common:cmake] See also "/workspace/ros/ws_jsk_demos/build/drc_task_common/CMakeFiles/CMakeOutput.log".
[drc_task_common:cmake] See also "/workspace/ros/ws_jsk_demos/build/drc_task_common/CMakeFiles/CMakeError.log".
Failed <<< drc_task_common                             [ 16.0 seconds ]
```

k-okada, 2016-12-15:
need https://github.com/jsk-ros-pkg/jsk_demos/pull/1210 ??

wkentaro, 2017-02-08:
This is done: https://github.com/jsk-ros-pkg/jsk_recognition/issues/1980#issue-193446350