jsk-ros-pkg/jsk_recognition #2022: jsk_pcl_ros is failing to build in Kinetic

tfoote, 2017-03-04:
It's blocking a bugfix sync. [Here's the thread](https://discourse.ros.org/t/need-to-sync-new-release-of-rqt-topic-indigo-jade-kinetic/1410/5?u=tfoote)

Details: 

http://build.ros.org/view/Kbin_uW64/job/Kbin_uW64__jsk_pcl_ros__ubuntu_wily_amd64__binary/10/consoleFull
```
CMakeFiles/jsk_pcl_ros.dir/src/geometric_consistency_grouping_nodelet.cpp.o -c /tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/src/geometric_consistency_grouping_nodelet.cpp
In file included from /usr/include/flann/util/matrix.h:35:0,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.7/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.7/pcl/kdtree/kdtree_flann.h:45,
                 from /tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/src/geometric_consistency_grouping_nodelet.cpp:39:
/usr/include/flann/util/serialization.h: In instantiation of ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/usr/include/flann/util/serialization.h:26:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:35:33:   required from ‘static void flann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:276:28:   required from ‘Archive& flann::serialization::InputArchive<Archive>::operator&(T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = flann::serialization::LoadArchive]’
/usr/include/flann/util/lsh_table.h:319:10:   required from ‘void flann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = flann::serialization::LoadArchive; ElementType = float]’
/usr/include/flann/util/serialization.h:18:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::lsh::LshTable<float>]’
/usr/include/flann/util/serialization.h:26:22:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/flann/util/serialization.h:18:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:26:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:35:33:   required from ‘static void flann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:276:28:   required from ‘Archive& flann::serialization::InputArchive<Archive>::operator&(T&) [with T = flann::LshIndex<flann::L2_Simple<float> >; Archive = flann::serialization::LoadArchive]’
/usr/include/flann/algorithms/lsh_index.h:204:9:   required from ‘void flann::LshIndex<Distance>::loadIndex(FILE*) [with Distance = flann::L2_Simple<float>; FILE = _IO_FILE]’
/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/src/geometric_consistency_grouping_nodelet.cpp:172:182:   required from here
/usr/include/flann/util/serialization.h:18:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
         type.serialize(ar);
         ^
/usr/include/flann/util/serialization.h: In instantiation of ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/usr/include/flann/util/serialization.h:26:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:40:33:   required from ‘static void flann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:294:28:   required from ‘Archive& flann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = flann::serialization::SaveArchive]’
/usr/include/flann/util/lsh_table.h:319:10:   required from ‘void flann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = flann::serialization::SaveArchive; ElementType = float]’
/usr/include/flann/util/serialization.h:18:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::lsh::LshTable<float>]’
/usr/include/flann/util/serialization.h:26:22:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/flann/util/serialization.h:18:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:26:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:40:33:   required from ‘static void flann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:294:28:   required from ‘Archive& flann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = flann::LshIndex<flann::L2_Simple<float> >; Archive = flann::serialization::SaveArchive]’
/usr/include/flann/algorithms/lsh_index.h:198:9:   required from ‘void flann::LshIndex<Distance>::saveIndex(FILE*) [with Distance = flann::L2_Simple<float>; FILE = _IO_FILE]’
/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/src/geometric_consistency_grouping_nodelet.cpp:172:182:   required from here
/usr/include/flann/util/serialization.h:18:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
CMakeFiles/jsk_pcl_ros.dir/build.make:1621: recipe for target 'CMakeFiles/jsk_pcl_ros.dir/src/geometric_consistency_grouping_nodelet.cpp.o' failed
make[4]: *** [CMakeFiles/jsk_pcl_ros.dir/src/geometric_consistency_grouping_nodelet.cpp.o] Error 1
make[4]: Leaving directory '/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/obj-x86_64-linux-gnu'
CMakeFiles/Makefile2:3555: recipe for target 'CMakeFiles/jsk_pcl_ros.dir/all' failed
make[3]: *** [CMakeFiles/jsk_pcl_ros.dir/all] Error 2
make[3]: Leaving directory '/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/obj-x86_64-linux-gnu'
Makefile:119: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/obj-x86_64-linux-gnu'
dh_auto_build: make -j1 returned exit code 2
debian/rules:36: recipe for target 'override_dh_auto_build' failed
make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0'
make[1]: *** [override_dh_auto_build] Error 2
debian/rules:23: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
E: Building failed
Traceback (most recent call last):
  File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
    subprocess.check_call(cmd, cwd=source_dir)
  File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-kinetic-jsk-pcl-ros']' returned non-zero exit status 1
# END SUBSECTION
```

k-okada, 2017-03-04:
Sorry for trouble, we're aware of this and working on this with @wkentaro
https://github.com/jsk-ros-pkg/jsk_common/pull/1506#issuecomment-283543973

k-okada, 2017-03-04:
@wkentaro we can reproduce this by
```
 hub clone http://github.com/tork-a/jsk_recognition-release
 cd jsk_recognition-release/
 git checkout debian/kinetic/xenial/jsk_pcl_ros
 dpkg-buildpackage -rfakeroot -b -uc
```
and workaround described in  https://github.com/mariusmuja/flann/issues/214 will fix this, see #2023

PR #2023 (2017-03-04): incldue flann before any opencv includes, fix #2022
Error reported at #2022
```
/tmp/binarydeb/ros-kinetic-jsk-pcl-ros-1.1.0/src/geometric_consistency_grouping_nodelet.cpp:172:182:   required from here
/usr/include/flann/util/serialization.h:18:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
         type.serialize(ar);
```
could be fixe by this issue, see https://github.com/mariusmuja/flann/issues/214 for detail

close #2022
commit 44b284d0 (2017-03-04): incldue flann before any opencv includes, fix #2022
commit 5b415067 (2017-03-04): Merge pull request #2023 from k-okada/fix_2022

incldue flann before any opencv includes, fix #2022
closed 2017-03-04 by k-okada