jsk-ros-pkg/jsk_recognition #2033: Travis test on Kinetic won't pass

wkentaro, 2017-03-16:
@k-okada do you have any idea?

https://s3.amazonaws.com/archive.travis-ci.org/jobs/211512793/log.txt
```
Starting  >>> jsk_perception                        
Errors     << jsk_perception:cmake /workspace/ros/ws_jsk_recognition/logs/jsk_perception/build.cmake.000.log
CMake Error at /opt/ros/kinetic/share/image_view2/cmake/image_view2Config.cmake:106 (message):
  Project 'image_view2' specifies '/opt/ros/kinetic/include/opencv-3.2.0-dev'
  as an include dir, which is not found.  It does neither exist as an
  absolute directory nor in
  '/opt/ros/kinetic//opt/ros/kinetic/include/opencv-3.2.0-dev'.  Ask the
  maintainer 'Kei Okada <k-okada@@jsk.t.u-tokyo.ac.jp>' to fix it.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:17 (find_package)
```

[image]

k-okada, 2017-03-17:
Can we reproduce this on local environment?

wkentaro, 2017-03-17:
I have not yet tried that.

k-okada, 2017-03-17:
Related?

https://github.com/ros-perception/vision_opencv/issues/162

vrabaud, 2017-03-17:
Why is your build installing OpenCV2 and not OpenCV3 ? To get a package unchanged between ROS distros and choose the right OpenCV, have it depend on cv_bridge  as explained at http://wiki.ros.org/opencv3#package.xml

k-okada, 2017-03-18:
@wkentaro Timed out -> https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/211353102, but should be solved http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/xenial-travis-jsk-ros-pkg-jsk-recognition-kinetic-deb-false-true/191/
@vrabaud we're not using opencv2 on kinetic, we basically depends on cv_bridge to avoid opencv version issue.

wkentaro, 2017-03-24:
Now it passes.

closed 2017-03-24 by wkentaro