jsk-ros-pkg/jsk_recognition #2034: Convert local image topics to remote image topics

mmurooka, 2017-03-16:
@wkentaro 

In my and @iory 's understanding, the following two programs try to publish remote image topics from local image topics. (Is that correct? In fetch program, topic name is local, but I think it should mean remote.)
What is difference and which is better?

HRP2 use relay to publish remote topoics.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/openni2_remote.launch#L119-L134

fetch use image_transport republish to publish remote topics.
https://github.com/start-jsk/jsk_apc/blob/master/jsk_apc2016_common/samples/fetch_openni_local.launch#L6-L21

wkentaro, 2017-03-16:
I'm not sure what do you mean with **remote**.
I mean **local** is the PC I use. **remote** is the PC on fetch.

wkentaro, 2017-03-16:
Maybe
For you
- robot = **local**
- controller (pc) = **remote**

For me
- robot = **remote**
- controller (pc) = **local**

mmurooka, 2017-03-16:
> For you
    robot = local
    controller (pc) = remote

[the computer connected with camera] = [robot] = [local]
I thought that this is common, but I don't know source of that.
In other robot, baxter, PR2, etc.., which is remote and which is local? cc: @furushchev

another question (and this is main) is that,
does anyone know the difference between relay and image_transport republish? and which is better?
HRP2 has communication speed problem https://github.com/start-jsk/rtmros_hrp2/pull/383#issue-145142863 and thought that it might be solved by replacing relay with image_transport republish.

wkentaro, 2017-03-16:
> does anyone know the difference between relay and image_transport republish? and which is better?

republish and relay is totally different.
relay does just copy the topic, but republish does decompress/compress images.

mmurooka, 2017-03-19:
I talked @wkentaro locally, and understand that compressedDepth (which is used in republish) is lossless compression, so https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/openni2_remote.launch#L119-L134 should be replaced with republish.

mmurooka, 2017-03-21:
> and understand that compressedDepth (which is used in republish) is lossless 

was not correct. https://github.com/jsk-ros-pkg/jsk_recognition/pull/2036#issuecomment-288012892

We add option to use republish. https://github.com/jsk-ros-pkg/jsk_recognition/pull/2036

closed 2017-03-21 by mmurooka