jsk-ros-pkg/jsk_recognition #2046: Large size of height map may fail

YoheiKakiuchi, 2017-03-30:
height map of 4000x4000 may stop nodelets.
(default size of height map is 800x800. 2000x2000 can be used)

YutoUchimi, 2019-09-20:
In `jsk_pcl_ros/sample/sample_heightmap_converter.launch` (will be added in #2440 ), I tried `resolution_x: 100000` and `resolution_y: 100000`, and it worked on my environment.
Also, `sample_heightmap_morphological_filtering.launch` and `sample_heightmap_to_pointcloud.launch` worked as well.

How can we reproduce your problem?

YutoUchimi, 2019-09-20:
Result of `roslaunch jsk_pcl_ros sample_heightmap_morphological_filtering.launch` (#2440 is required) with resolution 100000 x 100000 is here:

[image]

YutoUchimi, 2019-09-20:
Well, `100000` was ignored and automatically set to maximum resolution `4096`, which is defined in `cfg/HeightmapConverter.cfg`.
```
gen.add("resolution_x", int_t, 0, "", 400, 1, 4096)
gen.add("resolution_y", int_t, 0, "", 400, 1, 4096)
```

Anyway it seems working with 4096 x 4096.