jsk-ros-pkg/jsk_recognition #2048: Clusterpointindicesdecomposer crashes?

furushchev, 2017-04-10:
```bash
(gdb) bt
#0  0x00007ffff60bfc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff60c3028 in __GI_abort () at abort.c:89
#2  0x00007ffff66c7535 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff66c56d6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff66c5703 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff66c5922 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7bb0601 in void boost::throw_exception<boost::lock_error>(boost::lock_error const&) ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#7  0x00007ffff7bb0705 in boost::unique_lock<boost::mutex>::lock() () from /opt/ros/indigo/lib/libnodeletlib.so
#8  0x00007fffe44e6d31 in boost::unique_lock<boost::mutex>::unique_lock(boost::mutex&) ()
   from /home/furushchev/ros/indigo/devel/lib//libjsk_topic_tools.so
#9  0x00007fffe456abbc in jsk_topic_tools::VitalChecker::poke() ()
   from /home/furushchev/ros/indigo/devel/lib//libjsk_topic_tools.so
#10 0x00007fffd1ddc46e in jsk_pcl_ros::ClusterPointIndicesDecomposer::extract (this=0x12b0560, input=..., 
    indices_input=..., planes=..., coefficients=...)
    at /home/furushchev/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/cluster_point_indices_decomposer_nodelet.
cpp:443
#11 0x00007fffe457142c in boost::_mfi::mf1<void, jsk_topic_tools::DiagnosticNodelet, diagnostic_updater::DiagnosticStatusWra
pper&>::operator()(jsk_topic_tools::DiagnosticNodelet*, diagnostic_updater::DiagnosticStatusWrapper&) const ()
   from /home/furushchev/ros/indigo/devel/lib//libjsk_topic_tools.so
```

looks like `vital_checker_` is null pointer at this point, but this is initialized `DiagnosticNodelet::onInit` which is called on `ClusterPointIndicesDecomposer::onInit`.
Strange....

wkentaro, 2017-04-19:
If you provide the set of roslaunch and rosbag, someone may help.

furushchev, 2017-05-21:
@wkentaro Thanks for response. Unfortunately, now I could not reproduce this error any more.
I'll re-open this issue when it happens once again.

closed 2017-05-21 by furushchev