jsk-ros-pkg/jsk_recognition #2070: libsiftfast build error (1回目のみ)

mukuact, 2017-04-25:
libsiftfastを`catkin build`するとき，一度目には必ず失敗し，再び`catkin b`をするとうまく行きます．
```
....
mv: cannot stat ‘/home/muku/catkin_ws2/devel/.private/libsiftfast/lib/siftfastpy.so’: No such file or directory
make[3]: *** [libsiftfast] Error 1
make[2]: *** [/home/muku/catkin_ws2/devel/.private/libsiftfast/lib/libsiftfast.so] Error 2
make[1]: *** [CMakeFiles/libsiftfast.dir/all] Error 2
make: *** [all] Error 2
```

furushchev, 2017-04-25:
うーん、おかしいですね。これはjsk-ros-pkgなので、travisでチェックされている気がするんだけどなあ。
jsk-ros-pkg/jsk_3rdpartyにissueを出してもらえるとうれしいです。

furushchev, 2017-04-25:
失礼しました。これすでにjsk-ros-pkgでした。

furushchev, 2017-04-25:
失敗するときに`catkin build -v`をして詳細を教えてくれると嬉しいです。

wkentaro, 2017-04-25:
これはjsk_3rdpartyの話？

mukuact, 2017-04-25:
失礼しました，jsk_3rdpartyですね．

closed 2017-04-25 by mukuact

mukuact, 2017-04-25:
https://github.com/jsk-ros-pkg/jsk_3rdparty/issues/93
に開き直しました