jsk-ros-pkg/jsk_recognition #2073: rosrun can't find image_publisher.py

Horisu, 2017-04-28:
on ubuntu 14.04
ros-indigo-jsk-perception 1.1.0-0trusty-20170415-232441-0800

 I found that image_publisher is not found.
``` bash
 $ rosrun jsk_perception image_publisher.py
[rosrun] Couldn't find executable named image_publisher.py below /opt/ros/indigo/share/jsk_perception
```

but image_publisher.py exists like below
```bash
 $ dpkg -L ros-indigo-jsk-perception | grep image_publisher.py
/opt/ros/indigo/lib/jsk_perception/image_publisher.py
```
Why rosrun searches into /opt/ros/indigo/share/jsk_perception although it's actually in /opt/ros/indigo/lib/jsk_perception?

wkentaro, 2017-04-28:
On my env, it works without any problems:

```
wkentaro at sheeta in ~
% aptitude show ros-indigo-jsk-perception
Package: ros-indigo-jsk-perception
New: yes
State: installed
Automatically installed: no
Version: 1.1.0-0trusty-20170425-032131-0800
Priority: extra
Section: misc
Maintainer: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Architecture: amd64
Uncompressed Size: 28.2 M
Depends: libboost-filesystem1.54.0, libboost-program-options1.54.0, libboost-system1.54.0, libc6 (>=
         2.14), libconsole-bridge0.2, libgcc1 (>= 1:4.1.1), libgomp1 (>= 4.2.1), libopencv-calib3d2.4,
         libopencv-core2.4, libopencv-features2d2.4, libopencv-flann2.4, libopencv-highgui2.4,
         libopencv-imgproc2.4, libopencv-legacy2.4, libopencv-ml2.4, libopencv-objdetect2.4,
         libopencv-video2.4, libstdc++6 (>= 4.6), liburdfdom-world0.2, libyaml-cpp0.5, libeigen3-dev,
         libleveldb-dev, libyaml-cpp-dev, python-h5py, python-sklearn, ros-indigo-angles,
         ros-indigo-cv-bridge, ros-indigo-dynamic-reconfigure, ros-indigo-geometry-msgs,
         ros-indigo-image-geometry, ros-indigo-image-transport, ros-indigo-image-view,
         ros-indigo-image-view2, ros-indigo-imagesift, ros-indigo-jsk-data, ros-indigo-jsk-gui-msgs,
         ros-indigo-jsk-recognition-msgs, ros-indigo-jsk-recognition-utils, ros-indigo-jsk-rqt-plugins,
         ros-indigo-jsk-topic-tools, ros-indigo-libcmt, ros-indigo-message-runtime, ros-indigo-mk,
         ros-indigo-ml-classifiers, ros-indigo-nodelet, ros-indigo-opencv-apps,
         ros-indigo-openni2-launch, ros-indigo-pcl-ros, ros-indigo-posedetection-msgs,
         ros-indigo-robot-self-filter, ros-indigo-rosbag, ros-indigo-roscpp, ros-indigo-roseus,
         ros-indigo-rospack, ros-indigo-rospy, ros-indigo-rostopic, ros-indigo-rqt-gui,
         ros-indigo-sensor-msgs, ros-indigo-sound-play, ros-indigo-std-msgs, ros-indigo-tf
Conflicts: ros-indigo-jsk-perception
Description: jsk_perception

Homepage: http://ros.org/wiki/jsk_perception

wkentaro at sheeta in ~
% catkin_find --libexec jsk_perception
/opt/ros/indigo/lib/jsk_perception
wkentaro at sheeta in ~
% rosrun jsk_perception image_publisher.py
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
^CTraceback (most recent call last):
  File "/opt/ros/indigo/lib/jsk_perception/image_publisher.py", line 101, in <module>
    rospy.init_node('image_publisher')
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 333, in init_node
    raise rospy.exceptions.ROSInitException("Failed to initialize time. Please check logs for additional details")
rospy.exceptions.ROSInitException: Failed to initialize time. Please check logs for additional details
wkentaro at sheeta in ~
% cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros-shadow-fixed/ubuntu trusty main
```

wkentaro, 2017-04-28:
The important thing is

```
% catkin_find --libexec jsk_perception
/opt/ros/indigo/lib/jsk_perception
```

which is the way of rosrun to find executables.

Horisu, 2017-04-28:
I did't notice some files remained in catkin_workspace.
``` bash
 $ catkin_find --libexec jsk_perception
/home/leus/ros/indigo/devel/lib/jsk_perception
/opt/ros/indigo/lib/jsk_perception
```
I removed jsk_~~ from catkin workspace  and everything goes well. Thanks!

closed 2017-04-28 by wkentaro