jsk-ros-pkg/jsk_recognition #2087: Error on octomap_server_contact_nodelet with source installed octomap/octomap_mapping

wkentaro, 2017-05-15:
@mmurooka do you have any idea?

```bash
Errors     << jsk_pcl_ros:make /home/wkentaro/Projects/label_octomap/logs/jsk_pcl_ros/build.make.001.log                                                                                                                                                                                  
CMakeFiles/jsk_pcl_ros.dir/src/octomap_server_contact_nodelet.cpp.o: In function `OcTreeContact':
/home/wkentaro/Projects/label_octomap/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/OcTreeContact.h:80: undefined reference to `octomap::OcTree::OcTree(double)'
CMakeFiles/jsk_pcl_ros.dir/src/octomap_server_contact_nodelet.cpp.o: In function `octomap::OcTree::create() const':
/home/wkentaro/Projects/label_octomap/devel/include/octomap/OcTree.h:66: undefined reference to `octomap::OcTree::OcTree(double)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wkentaro/Projects/label_octomap/devel/lib/libjsk_pcl_ros.so] Error 1
make[1]: *** [CMakeFiles/jsk_pcl_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
WARNING: '/home/wkentaro/Projects/label_octomap/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/sample/data/sample_add_color_from_image_20170319.bag' exists
make: *** [all] Error 2
cd /home/wkentaro/Projects/label_octomap/build/jsk_pcl_ros; catkin build --get-env jsk_pcl_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..........................................................................................................................................................................................................................................................................................
Failed     << jsk_pcl_ros:make                       [ Exited with code 2 ]                                                                                                                                                                                                               
Failed    <<< jsk_pcl_ros                            [ 32.6 seconds ]                                                                                                                                                                                                                     
[build] Summary: 6 of 7 packages succeeded.                                                                                                                                                                                                                                               
[build]   Ignored:   7 packages were skipped or are blacklisted.                                                                                                                                                                                                                          
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                                                                                                                  
[build]   Abandoned: None.                                                                                                                                                                                                                                                                
[build]   Failed:    1 packages failed.                                                                                                                                                                                                                                                   
[build] Runtime: 34.7 seconds total.          
```

My workspace is:

```
% wstool info
workspace: /home/wkentaro/Projects/label_octomap/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 jsk-ros-pkg/jsk_recognition   git master         6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition.git
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel          4f0127598d50 github.com/OctoMap/octomap.git

% echo $CMAKE_PREFIX_PATH
/home/wkentaro/Projects/label_octomap/devel:/opt/ros/indigo
```

wkentaro, 2017-05-15:
I found the way to fix.
I overlooked the dependencies, and now it is fixed.

```
workspace: /home/wkentaro/Projects/label_octomap/src

 Localname                   S Version (Spec)
 ---------                   - --------------
 jsk-ros-pkg/jsk_recognition   master
 OctoMap/octomap_ros           indigo-devel
 OctoMap/octomap_msgs          indigo-devel
 OctoMap/octomap_mapping       indigo-devel
 OctoMap/octomap               devel
```

closed 2017-05-15 by wkentaro

mmurooka, 2017-05-15:
I tested in clean Ubuntu14.04/indigo environment in my VirtualBox and that error was not reproduced and octomap_server_contact was generated.

I think linking with octomap library fails in your environment and I have no idea about the reason.
Could you check following?

1. `OcTree::OcTree(double resolution)` is defined in `rosls octomap/src/OcTree.cpp`

2. there is `liboctomap.so` under `catkin_ws/devel/lib/`

3. `liboctomap.so` is contained in  `jsk_pcl_ros_LIB_DEPENDS` in `catkin_ws/build/jsk_pcl_ros/CMakeCache.txt`

mmurooka, 2017-05-15:
入れ違いでした．
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issue-228764609
だけでも（octomap_rosやoctomap_msgsはaptでいれていても），僕の今試した環境では大丈夫だったんだけれど，どうしてだろう．

wkentaro, 2017-05-15:
ありがとうございます。

octomap_msgsは多分大丈夫ですが、octomap_rosはoctomapに依存しているので本来はソースで入れないといけないのだと思います。
微妙な環境の差で変わるのかもしれないので、dockerで試してみます。

wkentaro, 2017-05-15:
**Update**

Below also works.

```
workspace: /home/wkentaro/Projects/label_octomap/src

 Localname                   S Version (Spec)
 ---------                   - --------------
 jsk-ros-pkg/jsk_recognition   master
 OctoMap/octomap_ros           indigo-devel
 OctoMap/octomap_mapping       indigo-devel
 OctoMap/octomap               devel
```

wkentaro, 2017-05-15:
ビルドは通りましたが、下のようになっているので、実行時に問題になると思われます。

```bash
root@72bac15ba79e:~/catkin_ws# ldd devel/lib/libjsk_pcl_ros.so | grep octomap
        liboctomap_server.so => /root/catkin_ws/devel/.private/octomap_server/lib/liboctomap_server.so (0x00007f8286bac000)
        liboctomap.so.1.6 => /opt/ros/indigo/lib/liboctomap.so.1.6 (0x00007f8286963000)
        liboctomap.so.1.9 => /root/catkin_ws/devel/lib/liboctomap.so.1.9 (0x00007f827a0f2000)
```

mmurooka, 2017-05-16:
> ビルドは通りましたが

は，https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issuecomment-301573088 の環境での話かな． https://github.com/jsk-ros-pkg/jsk_recognition/pull/2089, https://github.com/jsk-ros-pkg/jsk_recognition/pull/2090 は反映してないというので合ってるよね．

実行時に問題になるのは，sourceからできたoctomap.soよりもaptのoctomap.soが上にあるからということかな？

mmurooka, 2017-05-16:
ちなみに、最初のエラー https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issue-228764609 ですが、 https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issuecomment-301549745 の通り、以下の環境でもbuildは通ったから（実行時の問題はあるかもしれないけれど）、最初の指摘のエラーは起きなかったんだけれど、それについて思い当たる理由はあるかな。

```
leus@leus-VirtualBox:~/octomap_ws/src$ wstool info
workspace: /home/leus/octomap_ws/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 jsk-ros-pkg/jsk_recognition   git master  (-)    6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel  (-)     4f0127598d50 github.com/OctoMap/octomap.git
```

wkentaro, 2017-05-16:
> ビルドは通りましたが
は，#2087 (comment) の環境での話かな． #2089, #2090 は反映してないというので合ってるよね．

いえ、以下で、docker ros:indigoからでビルドした場合です。

```
% wstool info
workspace: /home/wkentaro/Projects/label_octomap/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 jsk-ros-pkg/jsk_recognition   git master         6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition.git
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel          4f0127598d50 github.com/OctoMap/octomap.git
```

> 実行時に問題になるのは，sourceからできたoctomap.soよりもaptのoctomap.soが上にあるからということかな？

そうです。

wkentaro, 2017-05-16:
> ちなみに、最初のエラー #2087 (comment) ですが、 #2087 (comment) の通り、以下の環境でもbuildは通ったから（実行時の問題はあるかもしれないけれど）、最初の指摘のエラーは起きなかったんだけれど、それについて思い当たる理由はあるかな。

それに関してはわかりません。aptのパッケージのbuild versionの日付などで微妙に違うのかもしれません。ちがいはros-indigo-octomap-rosにあるはずですが。

mmurooka, 2017-05-16:
octomap_rosをソースからいれないと、 https://github.com/jsk-ros-pkg/jsk_recognition/issues/2087#issuecomment-301604262 の問題が起きて、いれればその問題は起きない、ということを確認しました。
ちなみにどちらの環境でも（lddでaptのoctomap.soが上に来ていても）、https://github.com/jsk-ros-pkg/jsk_recognition/issues/1834 の問題は再現しませんでした。（cubeのマーカの入った領域がunknownからemptyになりました。）

```
leus@leus-VirtualBox:~$ wstool info -t octomap_ws/src/
workspace: /home/leus/octomap_ws/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 start-jsk/rtmros_tutorials  M git master  (-)    64d2b3c25035 github.com/start-jsk/rtmros_tutorials
 jsk-ros-pkg/jsk_recognition   git master  (-)    6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel  (-)     4f0127598d50 github.com/OctoMap/octomap.git
leus@leus-VirtualBox:~$ ldd octomap_ws/devel/lib/libjsk_pcl_ros.so | grep octo
	libjsk_recognition_utils.so => /home/leus/octomap_ws/devel/.private/jsk_recognition_utils/lib/libjsk_recognition_utils.so (0x00007f5b002bd000)
	liboctomap_server.so => /home/leus/octomap_ws/devel/.private/octomap_server/lib/liboctomap_server.so (0x00007f5aff485000)
	liboctomap.so.1.6 => /opt/ros/indigo/lib/liboctomap.so.1.6 (0x00007f5aff23c000)
	liboctomath.so.1.6 => /opt/ros/indigo/lib/liboctomath.so.1.6 (0x00007f5aff036000)
	libjsk_pcl_ros_base.so => /home/leus/octomap_ws/devel/.private/jsk_pcl_ros/lib/libjsk_pcl_ros_base.so (0x00007f5af61a8000)
	liboctomap.so.1.9 => /home/leus/octomap_ws/devel/lib/liboctomap.so.1.9 (0x00007f5af29cd000)
	liboctomath.so.1.9 => /home/leus/octomap_ws/devel/lib/liboctomath.so.1.9 (0x00007f5af27c6000)
```
```
leus@leus-VirtualBox:~$ wstool info -t octomap2_ws/src/
workspace: /home/leus/octomap2_ws/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 start-jsk/rtmros_tutorials  M git master  (-)    64d2b3c25035 github.com/start-jsk/rtmros_tutorials
 jsk-ros-pkg/jsk_recognition   git master  (-)    6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition
 OctoMap/octomap_ros           git indigo-devel   29ab0016c267 github.com/OctoMap/octomap_ros.git
 OctoMap/octomap_mapping       git indigo-devel   122bc6023092 github.com/OctoMap/octomap_mapping.git
 OctoMap/octomap               git devel  (-)     4f0127598d50 github.com/OctoMap/octomap.git
leus@leus-VirtualBox:~$ ldd octomap2_ws/devel/lib/libjsk_pcl_ros.so | grep octo
	libjsk_recognition_utils.so => /home/leus/octomap2_ws/devel/.private/jsk_recognition_utils/lib/libjsk_recognition_utils.so (0x00007fc650c89000)
	liboctomap_server.so => /home/leus/octomap2_ws/devel/.private/octomap_server/lib/liboctomap_server.so (0x00007fc64fe70000)
	liboctomap.so.1.9 => /home/leus/octomap2_ws/devel/lib/liboctomap.so.1.9 (0x00007fc64fc0b000)
	liboctomath.so.1.9 => /home/leus/octomap2_ws/devel/lib/liboctomath.so.1.9 (0x00007fc64fa04000)
	libjsk_pcl_ros_base.so => /home/leus/octomap2_ws/devel/lib/libjsk_pcl_ros_base.so (0x00007fc646b75000)
```

mmurooka, 2017-05-16:
octomap_server_contact was also built and worked correctly in the following environment.
So any package related to octomap is not ncessary to be installed from source.
```
leus@leus-VirtualBox:~$ wstool info -t octomap3_ws/src/
workspace: /home/leus/octomap3_ws/src

 Localname                   S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                   - --- -------------- -----------  ---------------------------
 start-jsk/rtmros_tutorials  M git master  (-)    64d2b3c25035 github.com/start-jsk/rtmros_tutorials
 jsk-ros-pkg/jsk_recognition   git master  (-)    6ea8dfd31c6e github.com/jsk-ros-pkg/jsk_recognition
leus@leus-VirtualBox:~$ ldd octomap3_ws/devel/lib/libjsk_pcl_ros.so | grep octo
	libjsk_recognition_utils.so => /home/leus/octomap3_ws/devel/.private/jsk_recognition_utils/lib/libjsk_recognition_utils.so (0x00007fa8e6d6a000)
	liboctomap_server.so => /opt/ros/indigo/lib/liboctomap_server.so (0x00007fa8e613d000)
	liboctomap.so.1.6 => /opt/ros/indigo/lib/liboctomap.so.1.6 (0x00007fa8e5ef4000)
	liboctomath.so.1.6 => /opt/ros/indigo/lib/liboctomath.so.1.6 (0x00007fa8e5cee000)
	libjsk_pcl_ros_base.so => /home/leus/octomap3_ws/devel/.private/jsk_pcl_ros/lib/libjsk_pcl_ros_base.so (0x00007fa8dce60000)
```