jsk-ros-pkg/jsk_recognition #2091: Proposal of introducing docker image ros:indigo + pcl1.8 to make test pass at one time

wkentaro, 2017-05-15:
The CI test fails because of build time of PCL1.8 at the first build, and it passes if we restarted the build thanks to the ccache.

In order to fix this, I propose the use of docker image with ros:indigo + pcl1.8.
We already have the feature of using custom docker image on jsk_travis thanks to @furushchev (https://github.com/jsk-ros-pkg/jsk_travis/pull/307),
so this should be easy to accomplish, and I believe this saves our time at testing PRs on CI.

Any idea?

[image]

- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/228236810
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/227975358
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/222761493
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/224566715

wkentaro, 2017-05-18:
@k-okada Do you have deb package of PCL1.8?

k-okada, 2017-05-18:
https://github.com/k-okada/pcl_1_8-release/releases ?

closed 2017-07-05 by k-okada