jsk-ros-pkg/jsk_recognition #2092: extend mmurooka's detection and  tracking method to round shaped objects

k-okada, 2017-05-18:
1. @mmurooka how can we use current detection and tracking method ?  is that already merged ? https://github.com/jsk-ros-pkg/jsk_recognition/pull/2027 ?
2. do we have doc for that ? if not, assign @Affonso-Gui 
3. extend to round shaped objects, i.e. dishes.

mmurooka, 2017-05-18:
1 It is not merged yet. I'd like to merge but current progress is 20% because many discussions are necessary for merging.
2 I have doc under euslib doc.
I updated install instruction of README yesterday and confirmed that rosbag tracking sample was worked correctly with this instruction in @ban-masa and @itohdak 's desktop PC. HRP2 model is not necessary for this sample, so it should work on @Affonso-Gui environment too.
https://github.com/mmurooka/euslib/tree/murooka-branch/demo/murooka/wholebody_manipulation_planner/em_motion_estimation#install
I noticed this instruction to @Affonso-Gui with e-mail so could you check please?
3 I think applying this method to round object is not easy and taking other method is easier..

mmurooka, 2017-05-18:
Also, I talked with @Affonso-Gui directly after Monday meeting, and in my understanding, his research topic is task transfer between single-arm robot and dual-arm robot. So I advised him that using simple method for visual recognition for the time being (color filter or table top), and try to make first simple task for single, dual-arm task transfer.

k-okada, 2017-05-18:
OK, @Affonso-Gui please share what you're doing.

Affonso-Gui, 2017-05-18:
This week I was focused on May Festival preparation, so wasn't able to really do anything remarkable.
For now I loaded the eus model for the plate being used and displayed it on irt according to subscribed topic position.