jsk-ros-pkg/jsk_recognition #2096: [jsk_pcl] Typo in ColorBasedRegionGrowingSegmentation

Affonso-Gui, 2017-05-29:
Noticed a typo in ColorBasedRegionGrowingSegmentation's parameters.

Wrong: thresho**u**ld
Correct: threshold

[image]
[image]

Documentation on read_the_docs states correct spelling.
[image]

wkentaro, 2017-05-29:
@Affonso-Gui Could you please fix this by a PR?

Affonso-Gui, 2017-05-30:
@wkentaro Pull Request created at #2098

k-okada, 2017-06-05:
thank you, closed via #2098

closed 2017-06-05 by k-okada