jsk-ros-pkg/jsk_recognition #2119: [jsk_perception] Question about making mask image from RectArray

kochigami, 2017-06-18:
I'd like to know the way of making mask image from RectArray.
What I want to do is like below.

Situation: many faces are obtained by RectArray, and I'd like to check each of them whether it is moving or not.

[image]

I'm going to make ```rect_array_to_mask_image.cpp``` like https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/rect_to_mask_image.cpp, but could anyone help me if it is right direction?  Sorry for my trouble.

furushchev, 2017-06-19:
First of all, you are not in trouble, but just a question.

Generally, It's fine to extend `rect_to_mask_image` to support `PolygonArray`, but IMO, it's better to try to find the way to "bypass" without changing nodelet if possible.
I cannot propose any pointer since it's still not clear for me what you want to do.
Please briefly describe expected output and full pipeline of your current recognition?

kochigami, 2017-06-19:
> First of all, you are not in trouble, but just a question.

I'm very sorry for my misleading expression.
In addition, I apologize for lack of information.

After thinking about one day, I found there is no need for ```rect_array_to_mask_image``` in my case.

I'm very sorry again and thank you very much for your comment.
Next time, I'll care about clarifying the question when writing an issue.

closed 2017-06-19 by kochigami