jsk-ros-pkg/jsk_recognition #2127: Color-histgram-classifier fails when object lable_name has "-"(hyphen)

mukuact, 2017-06-22:
In robocup objects, "おーいお茶" is.
I write it in English, "o-i_ocha" in yaml file.
Then, when the nodelet reads the o-i_ocha's histgram from rosparam, 
the nodelet manager gets unable to this nodelet.

Is this rosparam's feature or this nodelet' bug?
After i removed "o-i_ocha" from  yaml file, the nodelet runs fine.

I use color_histogram_classifier for recognize objects for the robocup.
I took color histogram reference and save them to a yaml file.

YAML file:
```
label_names:
    - boss
    - cafe_au_lait
    - chipstar
    - cream_soup
    - grapejoice
    - jagarico
    - matcha
    - o-i_ocha
    - oreo
    - soup_harusame
    - strawberrymilk
    - wand
histograms:
    boss: [0.18684209883213043, 0.22631579637527466, 0.015789473429322243, 0.0026315790601074696, 0.0, 0.0, 0.0026315790601074696, 0.005263158120214939, 0.10526315867900848, 0.05000000074505806, 0.028947368264198303, 0.028947368264198303, 0.02631578966975212, 0.06052631512284279, 0.0, 0.2605263292789459]
    cafe_au_lait: [0.20721925795078278, 0.616310179233551, 0.024064170196652412, 0.0, 0.0, 0.001336898421868682, 0.0, 0.001336898421868682, 0.0, 0.004010695032775402, 0.002673796843737364, 0.006684491876512766, 0.03208556026220322, 0.03609625622630119, 0.0, 0.06818182021379471]
    chipstar: [0.525801956653595, 0.2622036337852478, 0.013947001658380032, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.005578800570219755, 0.013947001658380032, 0.008368200622498989, 0.01813110150396824, 0.14086471498012543, 0.0, 0.01115760114043951]
    cream_soup: [0.21837088465690613, 0.6741767525672913, 0.003466204507276416, 0.0, 0.001733102253638208, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.003466204507276416, 0.005199306644499302, 0.0329289436340332, 0.0, 0.06065857782959938]
    grapejoice: [0.15059445798397064, 0.4663144052028656, 0.12813737988471985, 0.011889035813510418, 0.0026420080102980137, 0.0, 0.0026420080102980137, 0.015852047130465508, 0.07397621870040894, 0.0277410838752985, 0.023778071627020836, 0.01981505937874317, 0.02113606408238411, 0.031704094260931015, 0.018494054675102234, 0.005284016020596027]
    jagarico: [0.07459677755832672, 0.38508063554763794, 0.21169355511665344, 0.08064515888690948, 0.07459677755832672, 0.07258064299821854, 0.03629032149910927, 0.004032257944345474, 0.0, 0.0, 0.002016128972172737, 0.0, 0.002016128972172737, 0.0, 0.0, 0.05645161122083664]
    matcha: [0.03030303120613098, 0.12937062978744507, 0.12354312092065811, 0.08158507943153381, 0.011655011214315891, 0.0139860138297081, 0.023310022428631783, 0.20512820780277252, 0.29487180709838867, 0.03030303120613098, 0.016317015513777733, 0.005827505607157946, 0.00699300691485405, 0.0139860138297081, 0.0, 0.012820512987673283]
    o-i_ocha: [0.015568862669169903, 0.14850299060344696, 0.4598802328109741, 0.31736525893211365, 0.013173652812838554, 0.0, 0.0, 0.0011976048117503524, 0.0011976048117503524, 0.0, 0.0011976048117503524, 0.0011976048117503524, 0.0, 0.0011976048117503524, 0.0, 0.039520956575870514]
    oreo: [0.03790087625384331, 0.05636540427803993, 0.016520893201231956, 0.008746355772018433, 0.004859086591750383, 0.0038872691802680492, 0.0029154520016163588, 0.03790087625384331, 0.2876579165458679, 0.294460654258728, 0.03595723956823349, 0.008746355772018433, 0.016520893201231956, 0.020408162847161293, 0.0, 0.16715256869792938]
    soup_harusame: [0.04878048598766327, 0.9024389982223511, 0.04268292710185051, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0015243901871144772, 0.0, 0.0, 0.0, 0.0, 0.0045731705613434315]
    strawberrymilk: [0.5539083480834961, 0.1603773534297943, 0.020215634256601334, 0.001347708865068853, 0.0, 0.001347708865068853, 0.0, 0.004043126478791237, 0.001347708865068853, 0.001347708865068853, 0.0, 0.006738544441759586, 0.032345011830329895, 0.21159029006958008, 0.0, 0.005390835460275412]
    wand: [0.5948718190193176, 0.13076923787593842, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0025641026441007853, 0.0025641026441007853, 0.0025641026441007853, 0.020512821152806282, 0.09487179666757584, 0.0, 0.15128205716609955]
```

This is log.
```
started roslaunch server http://192.168.101.134:41137/

SUMMARY
========

PARAMETERS
 * /object_property_analysis/color_histogram/bin_size: 14
 * /object_property_analysis/color_histogram/histogram_policy: 0
 * /object_property_analysis/color_histogram_classifier/compare_policy: 0
 * /object_property_analysis/color_histogram_classifier/detection_threshold: 0.2
 * /object_property_analysis/color_histogram_classifier/histograms/boss: [0.18684209883213...
 * /object_property_analysis/color_histogram_classifier/histograms/cafe_au_lait: [0.20721925795078...
 * /object_property_analysis/color_histogram_classifier/histograms/chipstar: [0.52580195665359...
 * /object_property_analysis/color_histogram_classifier/histograms/cream_soup: [0.21837088465690...
 * /object_property_analysis/color_histogram_classifier/histograms/grapejoice: [0.15059445798397...
 * /object_property_analysis/color_histogram_classifier/histograms/jagarico: [0.07459677755832...
 * /object_property_analysis/color_histogram_classifier/histograms/matcha: [0.03030303120613...
 * /object_property_analysis/color_histogram_classifier/histograms/o-i_ocha: [0.01556886266916...
 * /object_property_analysis/color_histogram_classifier/histograms/oreo: [0.03790087625384...
 * /object_property_analysis/color_histogram_classifier/histograms/soup_harusame: [0.04878048598766...
 * /object_property_analysis/color_histogram_classifier/histograms/strawberrymilk: [0.55390834808349...
 * /object_property_analysis/color_histogram_classifier/histograms/wand: [0.59487181901931...
 * /object_property_analysis/color_histogram_classifier/label_names: ['boss', 'cafe_au...
 * /object_property_analysis/color_histogram_classifier/vital_rate: 0.2
 * /rosdistro: indigo
 * /rosversion: 1.11.21

[... 18 lines cut ...]
[ INFO] [1498099065.556320344]: Loading histograms/chipstar
[ INFO] [1498099065.560758570]: Loading histograms/cream_soup
[ INFO] [1498099065.573743005]: Loading histograms/grapejoice
[ INFO] [1498099065.579128448]: Loading histograms/jagarico
[ INFO] [1498099065.582312059]: Loading histograms/matcha
[ INFO] [1498099065.586084154]: Loading histograms/o-i_ocha
[FATAL] [1498099065.609134454]: Failed to load nodelet '/object_property_analysis/color_histogram_classifier` of type `jsk_pcl_utils/ColorHistogramClassifier` to manager `/property_analysis_nodelet_manager'
[property_analysis_nodelet_manager-1] process has died [pid 29020, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=property_analysis_nodelet_manager __log:=/home/m-takeda/.ros/log/20170622-085754_713495b6-56dd-11e7-a6bf-00306444d5ec/property_analysis_nodelet_manager-1.log].
log file: /home/m-takeda/.ros/log/20170622-085754_713495b6-56dd-11e7-a6bf-00306444d5ec/property_analysis_nodelet_manager-1*.log
[property_analysis_nodelet_manager-1] restarting process
process[property_analysis_nodelet_manager-1]: started with pid [29194]
[object_property_analysis/color_histogram_classifier-4] process has died [pid 29035, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load jsk_pcl_utils/ColorHistogramClassifier /property_analysis_nodelet_manager ~input/array:=color_histogram/output __name:=color_histogram_classifier __log:=/home/m-takeda/.ros/log/20170622-085754_713495b6-56dd-11e7-a6bf-00306444d5ec/object_property_analysis-color_histogram_classifier-4.log].
log file: /home/m-takeda/.ros/log/20170622-085754_713495b6-56dd-11e7-a6bf-00306444d5ec/object_property_analysis-color_histogram_classifier-4*.log
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[ INFO] [1498099066.299238444]: Initializing nodelet with 8 worker threads.
[WARN] [WallTime: 1498099072.010703] [/object_property_analysis/color_histogram_visualizer] subscribes topics only with child subscribers. Set '~always_subscribe' as True to have it subscribe always.
[object_property_analysis/color_histogram-2] process has finished cleanly
log file: /home/m-takeda/.ros/log/20170622-085754_713495b6-56dd-11e7-a6bf-00306444d5ec/object_property_analysis-color_histogram-2*.log
^C[object_property_analysis/color_histogram_visualizer-3] killing on exit
[property_analysis_nodelet_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

furushchev, 2017-07-04:
Please don't include hyphen. this is usually not supported in ROS world.

mukuact, 2017-07-04:
OK

closed 2017-07-04 by mukuact

k-okada, 2017-07-05:
How about add error message when we found invalid object name?