jsk-ros-pkg/jsk_recognition #2128: [jsk_perception] question about the warning message appeard when using nodelet of FlowVelocityThresholding

kochigami, 2017-06-22:
Hi, 

I'm wondering whether the warning appeared when I use nodelet of ```FlowVelocityThresholding (jsk_perception)```:

```[ WARN] [1498101017.010182008, 1476540909.631869327]: [/nodelet_manager] '~input/flows' has not been remapped.```

is ok to ignore or not.

-  When I executed ``` roslaunch jsk_perception sample_flow_velocity_thresholding.launch ```, 
it doesn't appear.

- In this launch, node "flow_velocity_thresholding" is written as follows.
```
  <node name="flow_velocity_thresholding"
        pkg="jsk_perception" type="flow_velocity_thresholding">
    <remap from="~input/flows" to="fback_flow/flows" />
    <rosparam>
      approximate_sync: false
      use_camera_info: false
      image_height: 480
      image_width: 640
      window_size: 16
      threshold: 5.0
    </rosparam>
  </node>
```

- However, when I change FlowVelocityThresholding to use nodelet, this message appears.
It seems that launch file works fine.

- I modified it like this
```
  <node name="flow_velocity_thresholding"
        pkg="nodelet" type="nodelet" respawn="true"
        args="load jsk_perception/FlowVelocityThresholding /$(arg manager)">
    <remap from="~input/flows" to="fback_flow/flows" />
    <rosparam>                                                                                                                                                           
      approximate_sync: false                                                                                                                                            
      use_camera_info: false                                                                                                                                             
      image_height: 480                                                                                                                                                  
      image_width: 640                                                                                                                                                   
      window_size: 16                                                                                                                                                    
      threshold: 5.0                                                                                                                                                     
    </rosparam>
  </node>
```

used whole launch file: https://gist.github.com/kochigami/b73131f1e04fa148ad4620340a5c8915
actual scene of execution: 
[image]

wkentaro, 2017-06-22:
> <https://github.com/jsk-ros-pkg/jsk_recognition/issues/2128> to @wkentaro

It seems ok, and the warning is a kind of bug.
I will try to fix it.

kochigami, 2017-06-25:
Sorry for my late reply.
Thank you very much for your comment!

closed 2017-06-25 by kochigami