jsk-ros-pkg/jsk_recognition #2136: Request of releasing 1.1.3

wkentaro, 2017-06-27:
We need https://github.com/jsk-ros-pkg/jsk_recognition/issues/2117 in other project.

k-okada, 2017-06-27:
ok, let's wait for 
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/247644449
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/247571214
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/247667512
- https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/241978259

wkentaro, 2017-07-04:
so it's time to release?

knorth55, 2017-07-07:
now this issue moves to release of `1.1.4`?

knorth55, 2017-07-07:
Oops, `1.1.3` is not released yet.
@k-okada which PR are you waiting for release of `1.1.3`?

k-okada, 2017-07-07:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2150

wkentaro, 2017-07-07:
@k-okada Could you please release 1.1.3?

closed 2017-07-07 by wkentaro

k-okada, 2017-07-07:
https://github.com/ros/rosdistro/pull/15455