jsk-ros-pkg/jsk_recognition #2145: centroid_publisherで入力ポイントクラウドが空のときにtfのエラーが出ます

itohdak, 2017-07-05:
@ioryと@mmurookaと見てみて，
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros_utils/src/centroid_publisher_nodelet.cpp#L56
の中で`pcl::compute3DCentroid`を読んで
`Eigen::Vector4f center;`に重心の位置をセットしていますが，
入力ポイントクラウドが空の時には，
https://github.com/PointCloudLibrary/pcl/blob/master/common/include/pcl/common/impl/centroid.hpp#L82-L83
で何もセットせずに抜けてくるので，初期化されていない値のまま，
tfがbroadcastされたり，poseがpublishされたりしてしまいます．

初期化されていない値は，とても大きい数だったり小さい数だったり，nanだったりして，
この値がそのままcentroid_publisherからtfやposeとして出力されてしまいます．
nanのときはtfをlistenしているeusのターミナルに以下の出力が出てしまいます．
```
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "centroid_publisher_left" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)
         at line 240 in /tmp/binarydeb/ros-indigo-tf2-0.5.15/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "centroid_publisher_left" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)
         at line 240 in /tmp/binarydeb/ros-indigo-tf2-0.5.15/src/buffer_core.cpp
[ERROR] [1499243387.467806989, 1499243387.393869000]: Ignoring transform for child_frame_id "centroid_publisher_left" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)
```

どうするのがいいでしょうか．
・入力ポイントクラウドが空のときには，tfやposeを出さない．
・入力ポイントクラウドが空のときには，tfやposeは原点を出すようにする．（この場合は重心が実際に原点にあるのと見分けが付かなくなりますが，現実的にそれはなさそうです．）
・これが仕様だとしてそのままにする．
などでしょうか．
詳しい方（@wkentaro , @furushchev ．．．），アドバイスいただけると嬉しいです．

wkentaro, 2017-07-05:
TFは出さない、PoseはQuartanionに全部0をいれたものを出すというのはどうでしょうか。

k-okada, 2017-07-05:
Tf出さなかったらposeもなくていい気がするけど、出すといい事あるかな

wkentaro, 2017-07-05:
> Tf出さなかったらposeもなくていい気がするけど、出すといい事あるかな

上流のトピックが止まっている（Syncできていない or カメラが壊れた)のか、ポイントクラウドが空なのかというのが区別できるとデバッグやシステムを作るときに便利だと考えています。

k-okada, 2017-07-06:
なるほど．そのままeusで読み続けると以下のようにエラーなるので、気がつくとは思うけど、もしかしたらどこかで変になるかもしれないですね．まぁやってみましょう．
```
1.irteusgl$ quaternion2matrix #f(0 0 0 0)
;; quaternion2matrix : invalid input #f(0.0 0.0 0.0 0.0), the norm is not 1
#2f((0.0 0.0 0.0) (0.0 0.0 0.0) (0.0 0.0 0.0))
```

https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/eustf.l#L88-L99

furushchev, 2017-07-06:
これ悩ましいところですね。
とくにこの結果を`message_filters`で他の周期的に出てくるトピックとApproximateSyncしたときに影響が出そうです。
確かに何かしらパブリッシュされるとデバッグには良いというのは一理あります。
IMO, これくらい単純な機能のnodeletなら、もしempty cloudだったら`ROS_WARN_THROTTLE`でナイヨと警告する（そしてトピックは出さない）くらいでもいいのかなと思いました。

wkentaro, 2017-07-06:
poseにnanがはいっているときはeusはなにか言うんでしょうか。

k-okada, 2017-07-06:
7.E1-irteusgl$ quaternion2matrix #f(0 0 0 0)
;; quaternion2matrix : invalid input #f(0.0 0.0 0.0 0.0), the norm is not 1
#2f((0.0 0.0 0.0) (0.0 0.0 0.0) (0.0 0.0 0.0))
8.E1-irteusgl$ quaternion2matrix (float-vector *nan* *nan* *nan* *nan*)
#2f((-nan -nan -nan) (-nan -nan -nan) (-nan -nan -nan))

何も言わないですねぇ．

wkentaro, 2017-07-06:
ros::tf-point->pos は#f(-nan -nan -nan)を返してくるので、どこかIKを解く当たりでエラーになるかもしれませんね。

closed 2017-07-07 by k-okada