jsk-ros-pkg/jsk_recognition #2152: jsk_recognition_utils/pcl/color_histogram cannot find

knorth55, 2017-07-07:
@furushchev any idea for this build error?
```
In file included from /home/shingo/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/color_histogram_nodelet.cpp:40:0:
/home/shingo/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/color_histogram.h:48:55: fatal error: jsk_recognition_utils/pcl/color_histogram.h: そのようなファイルやディレクトリはありません
 #include <jsk_recognition_utils/pcl/color_histogram.h>
                                                       ^
compilation terminated.
In file included from /home/shingo/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/color_histogram_classifier_nodelet.cpp:41:0:
/home/shingo/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/color_histogram_classifier.h:48:55: fatal error: jsk_recognition_utils/pcl/color_histogram.h: そのようなファイルやディレクトリはありません
 #include <jsk_recognition_utils/pcl/color_histogram.h>
                                                       ^
```

furushchev, 2017-07-07:
In our env it's ok. Maybe your environment is broken. Let's wait someone
complains.

wkentaro, 2017-07-07:
@knorth55 Please try `catkin clean -a`.

knorth55, 2017-07-07:
I clean up all workspace, and now it succeeded to build.

closed 2017-07-07 by wkentaro