jsk-ros-pkg/jsk_recognition #2163: [jsk_pcl_ros/pointcloud_screenpoint] service call might fail when there is no pointcloud around a point

kochigami, 2017-07-12:
I encounterred ```responded with an error```several times 
when I execute ```screen_to_point``` service of ```pointcloud_screenpoint``` as follows:
[document](http://jsk-recognition.readthedocs.io/en/latest/jsk_pcl_ros/nodes/pointcloud_screenpoint.html)

I think it is caused by there is no pointcloud around a point (coords of u, v),
and I'd like to return False or something instead of just an error.

code 
```
SCREEN_TO_POINT_SERVICE = "/people_detection/pointcloud_screenpoint/screen_to_point"
SCREEN_TO_POINT_SERVICE_MSG = TransformScreenpoint

# wait for service
rospy.wait_for_service(SCREEN_TO_POINT_SERVICE)
# create service proxy
screen_to_point = rospy.ServiceProxy(SCREEN_TO_POINT_SERVICE, SCREEN_TO_POINT_SERVICE_MSG)
# call service
screen_point = screen_to_point(u, v)
# ex) screen_point = screen_to_point((self.person.faces)[0].face.x, (self.person.faces)[0].face.y)
```

The error is shown below:
[image]

At the same time the log of the launch terminal is below:
This log says no transform ((u, v) to (x, y, z) ) is achieved.
[image]

k-okada, 2017-07-17:
Did you try using try-catch ?

```
    try:
        screen_point = screen_to_point((self.person.faces)[0].face.x, (self.person.faces)[0].face.y)
    except rospy.ServiceException, e:
        print "Service call failed: %s"%e
```

kochigami, 2017-07-17:
> Did you try using try-catch ?

Thank you very much for your comment.
No, and my problem will be solved by this. Sorry for that...

kochigami, 2017-07-17:
If I check this works, I'll close this

kochigami, 2017-07-18:
I

closed 2017-07-18 by kochigami