jsk-ros-pkg/jsk_recognition #2165: People Pose Estimation 2D uses large memory

furushchev, 2017-07-14:
I'm running `people_pose_estimation_2d.py` with CUDA 6.5.
It works well on start (publish processed results), but after a few seconds it does not publish any message.

[image]

I found that after callback is called sometimes, memory (not GPU memory) is run out.

- Why this program uses large memory (up to 2GB)
- Is it possible to reduce memory usage?

What I did are:

- ensure `del` for all chainer variables which are allocated by `.to_cpu` method
- call `gc.collect()` on every callback function 

But still no luck.
Thank you very much in advance.

furushchev, 2017-07-15:
During discussion with @iory, I found a hint: https://github.com/ros/ros_comm/issues/536#issuecomment-67061303,
Even if subscriber is initialized with `queue_size=1`, incomming messages are also queued in OS stack instead of ros node.
In this case, we have fast publisher (openni2 node that publishes image at ~30fps) and very slow subscriber (~0.5fps), so huge messages should be queued in OS stack which may occurs swapping and  OOM killer.
(Blue line grows, but still red line keeps constant)

I looked into `rospy` implementation a bit more and it seems good to throttle input topics and to set large `buff_size` enough not to store many messages in OS queue.

k-okada, 2017-07-15:
> During discussion with @iory <https://github.com/iory>, I found a hint: ros/ros_comm#536
> (comment)

If that is true, why we have not had this problem so far . I think this is
not a only node that subscribe openNi and takes a time for processing.

furushchev, 2017-07-15:
@k-okada Hmm, I also think about it and that's why I wrote this as a 'hint' not a solution...
One possible reason that I have see that case is that:
- if `queue_size=1` but `buff_size` is not specified (so default value is used), it is unclear (at least for me now) when data in OS queue is remained until the process is shutdown.
- if `queue_size=1` and `buff_size` is set a large value, ros node receives data up to `buff_size` on every callback spin and explicitly dispose the data except the latest message (https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/rospy/src/rospy/msg.py#L225)

I think it may affects a lot in env with poor memory.

furushchev, 2017-07-15:
FYI, I compared a little:

- queue_size == 1 and buff_size == 2**24

```bash
[INFO] [WallTime: 1500109868.204048] image_cb start: incomming msg is 193.034179 msec behind
[INFO] [WallTime: 1500109871.616009] image_cb end: elapsed 3409.711123 msec
[INFO] [WallTime: 1500109871.628772] image_cb start: incomming msg is 3115.109451 msec behind
[INFO] [WallTime: 1500109874.025788] image_cb end: elapsed 2396.065950 msec
[INFO] [WallTime: 1500109874.034504] image_cb start: incomming msg is 2906.559236 msec behind
[INFO] [WallTime: 1500109876.405360] image_cb end: elapsed 2369.572163 msec
[INFO] [WallTime: 1500109876.421334] image_cb start: incomming msg is 4254.416472 msec behind
[INFO] [WallTime: 1500109878.738572] image_cb end: elapsed 2316.210985 msec
[INFO] [WallTime: 1500109878.756336] image_cb start: incomming msg is 1427.985675 msec behind
[INFO] [WallTime: 1500109881.189101] image_cb end: elapsed 2431.800127 msec
[INFO] [WallTime: 1500109881.206497] image_cb start: incomming msg is 2470.606095 msec behind
[INFO] [WallTime: 1500109883.535645] image_cb end: elapsed 2328.184843 msec
[INFO] [WallTime: 1500109883.555345] image_cb start: incomming msg is 1601.945885 msec behind
[INFO] [WallTime: 1500109885.927037] image_cb end: elapsed 2370.676041 msec
[INFO] [WallTime: 1500109885.949410] image_cb start: incomming msg is 812.378652 msec behind
[INFO] [WallTime: 1500109888.256727] image_cb end: elapsed 2306.326866 msec
[INFO] [WallTime: 1500109888.276162] image_cb start: incomming msg is 524.908549 msec behind
```

- queue_size == 1 and buff_size is default

```bash
[INFO] [WallTime: 1500109904.011857] image_cb start: incomming msg is 139.244802 msec behind
[INFO] [WallTime: 1500109907.685491] image_cb end: elapsed 3669.833899 msec
[INFO] [WallTime: 1500109907.728083] image_cb start: incomming msg is 3487.276085 msec behind
[INFO] [WallTime: 1500109909.272262] image_cb end: elapsed 1538.429022 msec
[INFO] [WallTime: 1500109909.278068] image_cb start: incomming msg is 1719.632156 msec behind
[INFO] [WallTime: 1500109911.627080] image_cb end: elapsed 2348.061085 msec
[INFO] [WallTime: 1500109911.632871] image_cb start: incomming msg is 3571.409948 msec behind
[INFO] [WallTime: 1500109913.969041] image_cb end: elapsed 2335.243940 msec
[INFO] [WallTime: 1500109913.975906] image_cb start: incomming msg is 5377.596147 msec behind
[INFO] [WallTime: 1500109916.349598] image_cb end: elapsed 2372.147083 msec
[INFO] [WallTime: 1500109916.356228] image_cb start: incomming msg is 7289.487131 msec behind
[INFO] [WallTime: 1500109918.698271] image_cb end: elapsed 2339.069843 msec
[INFO] [WallTime: 1500109918.704896] image_cb start: incomming msg is 6923.114307 msec behind
[INFO] [WallTime: 1500109921.027561] image_cb end: elapsed 2321.686029 msec
[INFO] [WallTime: 1500109921.034075] image_cb start: incomming msg is 8749.837167 msec behind
[INFO] [WallTime: 1500109923.331831] image_cb end: elapsed 2296.417952 msec
[INFO] [WallTime: 1500109923.337116] image_cb start: incomming msg is 9410.709389 msec behind
[INFO] [WallTime: 1500109925.703760] image_cb end: elapsed 2365.485907 msec
[INFO] [WallTime: 1500109925.708828] image_cb start: incomming msg is 9671.107538 msec behind
[INFO] [WallTime: 1500109928.065879] image_cb end: elapsed 2355.963945 msec
[INFO] [WallTime: 1500109928.073907] image_cb start: incomming msg is 9421.962745 msec behind
[INFO] [WallTime: 1500109930.445769] image_cb end: elapsed 2370.448112 msec
[INFO] [WallTime: 1500109930.450919] image_cb start: incomming msg is 9687.563427 msec behind
[INFO] [WallTime: 1500109936.812584] image_cb end: elapsed 6360.581160 msec
[INFO] [WallTime: 1500109936.821156] image_cb start: incomming msg is 13544.175393 msec behind
[INFO] [WallTime: 1500109939.156145] image_cb end: elapsed 2334.013939 msec
[INFO] [WallTime: 1500109939.161439] image_cb start: incomming msg is 13672.584779 msec behind
[INFO] [WallTime: 1500109941.506940] image_cb end: elapsed 2344.081879 msec
[INFO] [WallTime: 1500109941.514605] image_cb start: incomming msg is 13411.368615 msec behind
```

In both cases, rospy reads from socket and use the latest msg, but in the latter case, the latest msg is already ~13 sec behind from `rospy.Time.now()` in the beginning of the callback.
It seems that if a node doesn't read enough data from socket on every spin, OS queue expands up to OS limit.

furushchev, 2017-07-15:
One more thing: I'm running `openni2` node and the recognition node in different machines. It should be difference from usual settings.

furushchev, 2017-07-16:
I changed not to use `message_filters` and just estimate 2d poses of people in a rgb image on recognition node, and then register the result with depth image on the machine that runs `openni2` node.
It now works without problem, so it is no more urgent.
I'll investigate more and more when my work load goes down.
Thanks for debugging anyway! @k-okada @iory

furushchev, 2017-10-04:
Sorry for very late response. It seems that the original problem was that this node synchronizes some topics which have a header with the same stamp as RGB image with `camera_info` with a different stamp (device time?).
Since camera info does not change for most devices, it works now by subscribing camera info only once instead of synchronizing every time.
I will create a PR for fix.

k-okada, 2017-10-07:
please report this problem to upstream
1. if device publishes different timestamp between `camea_info` and `image`, then report to driver package
2. `It seems that if a node doesn't read enough data from socket on every spin, OS queue expands up to OS limit.` if this happens only rospy, not roscpp, then report to rospy repository.
3. what happens if we set `queue_size=1` and small `buff_size`
4. can we change OS queue behavior by http://wiki.bit-hive.com/linuxkernelmemo/pg/listen%20backlog%20%A1%DA3.6%A1%DB ?

PR #2206 (2017-10-17): jsk_perception: people_pose_estimation_2d.py: unsynchronize camera info
This pull request disables synchronization of camera info with images and let the node use it only once instead.
Fixes #2165
closed 2017-10-17 by k-okada