jsk-ros-pkg/jsk_recognition #2170: resized_image_transport fails

Horisu, 2017-07-17:
ubuntu 14.04
ros indigo
deb ros-indigo-resized-image-transport   1.2.1-0trusty-20170715- amd64

I got this error.
``` bash
$ roslaunch resized_image_transport resized_image_transport.launch CAMERA_NS:=/stereo/left INPUT_IMAGE:=image_rect_color
... logging to /home/grxuser/.ros/log/12afd736-6aa5-11e7-a728-fcaa14adade6/roslaunch-garlic-18157.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/grxuser/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.101.192:32900/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /stereo/left/image_resizer/resize_scale_x: 0.25
 * /stereo/left/image_resizer/resize_scale_y: 0.25

NODES
  /stereo/left/
    image_resizer (resized_image_transport/image_resizer)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[stereo/left/image_resizer-1]: started with pid [18175]
[ERROR] [1500267780.937170559]: Failed to load nodelet [/stereo/left/image_resizer] of type [resized_image_transport/ImageResizer] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type ImageResizer as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1500267780.937243024]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libresized_image_transport.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib//libresized_image_transport.so: undefined symbol: _ZN15jsk_topic_tools22ConnectionBasedNodelet35warnOnInitPostProcessCalledCallbackERKN3ros14WallTimerEventE)

```

mmurooka, 2017-07-18:
```
(Poco exception = /opt/ros/indigo/lib//libresized_image_transport.so: undefined symbol: _ZN15jsk_topic_tools22ConnectionBasedNodelet35warnOnInitPostProcessCalledCallbackERKN3ros14WallTimerEventE)
```
Maybe jsk_topic_tools is old. So please try `roscd jsk_topic_tools; git pull` (if you install from source) or `apt-get install ros-indigo-jsk-topic-tools` (if you install from apt), and build again.

k-okada, 2017-07-18:
I see, may be current node requries 2.2.4 of jsk_topic_tools https://github.com/jsk-ros-pkg/jsk_common/commit/52785f4d7cb5f58e538b6e593af8be268edd02ad

https://github.com/jsk-ros-pkg/jsk_recognition/pull/2173 will install latest version of jsk_topic_tools when yo apt-get resized_image_transport.

see https://github.com/ros-infrastructure/bloom/issues/414#issuecomment-267227753, why we have to write this manually.

Horisu, 2017-07-18:
I updated jsk_topit_tools to 2.2.5-0trusty-20170715, and it goes well now.
Thank you.

closed 2017-07-19 by k-okada