jsk-ros-pkg/jsk_recognition #2178: jsk_perception/PeoplePoseEstimation2d: Duplicated persons in array

furushchev, 2017-07-25:
Please report detail here @sasabot

sasabot, 2017-07-25:
echo returns something like this:
```
header: 
  seq: 1
  stamp: 
    secs: 1500966560
    nsecs: 108812325
  frame_id: head_rgbd_sensor_rgb_frame
poses: 
  - 
    limb_names: ['Nose', 'Neck', 'RShoulder', 'RElbow', 'RWrist', 'LShoulder', 'LElbow', 'LWrist', 'RHip', 'RKnee', 'RAnkle', 'LHip', 'LKnee', 'LAnkle', 'REye', 'LEye', 'REar', 'LEar', 'Bkg']
    poses: 
      - 
        position: 
          x: -0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -0.0
          y: 0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
[... 3756 lines cut ...]
          z: 0.0
          w: 1.0
      - 
        position: 
          x: -1.58990278661
          y: -0.661412359743
          z: 3.41000008583
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      - 
        position: 
          x: 0.0
          y: -0.0
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    scores: [0.5854012966156006, 0.8191861510276794, 0.8086917400360107, 0.7757990956306458, 0.7833817601203918, 0.8150124549865723, 0.7959932088851929, 0.8866803646087646, 0.8465234637260437, 0.7833183407783508, 0.8701238632202148, 0.8950873613357544, 0.6441076993942261, 0.8017038702964783, 0.2680170238018036]
  - 
```

furushchev, 2017-07-26:
Closed via #2179

closed 2017-07-26 by furushchev