jsk-ros-pkg/jsk_recognition #2221: Test failure of color_histogram.test on Travis

wkentaro, 2017-11-21:
it fails because of jsk_perception/test_results/jsk_perception/rosunit-test_color_histogram.xml: 1 tests, 0 errors, 1 failures, 0 skipped.

```
<?xml version="1.0" encoding="utf-8"?>
<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="30.011">
  <testcase classname="__main__.TestTopicPublished" name="test_published" time="30.0112">
    <failure type="AssertionError">Topic [/color_histogram/intensity_histogram] is not published
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/opt/ros/indigo/share/jsk_tools/src/test_topic_published.py", line 96, in test_published
    checker.topic_name)
  File "/usr/lib/python2.7/unittest/case.py", line 424, in assertTrue
    raise self.failureException(msg)
    </failure>
  </testcase>
  <system-out><![CDATA[]]></system-out>
  <system-err><![CDATA[]]></system-err>
</testsuite>
```

https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/302559010
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/302559014
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/302559015

Ridhwanluthra, 2018-02-27:
I would like to take up this issue is it solved or can i work on it?

knorth55, 2018-02-28:
It is not solved yet, but this is a maintainer problem.
We are working on it, and I strongly recommend you to wait for our update.

closed 2019-02-28 by k-okada