jsk-ros-pkg/jsk_recognition #2228: Build failed in Jenkins: Kdev__jsk_common__ubuntu_xenial_amd64 #10

wkentaro, 2017-11-26:
I got below email:

---

See <http://build.ros.org/job/Kdev__jsk_common__ubuntu_xenial_amd64/10/display/redirect>

```
------------------------------------------
[...truncated 23.51 KB...]

  ENTRYPOINT ["sh", "-c"]
  CMD ["rosdep update && PYTHONPATH=/tmp/ros_buildfarm:$PYTHONPATH python3 -u /tmp/ros_buildfarm/scripts/devel/create_devel_task_generator.py --rosdistro-name kinetic --workspace-root /tmp/catkin_workspace --os-name ubuntu --os-code-name xenial --arch amd64 --distribution-repository-urls http://repositories.ros.org/ubuntu/testing --distribution-repository-key-files /tmp/keys/0.key --dockerfile-dir /tmp/docker_build_and_install && PYTHONPATH=/tmp/ros_buildfarm:$PYTHONPATH python3 -u /tmp/ros_buildfarm/scripts/devel/create_devel_task_generator.py --rosdistro-name kinetic --workspace-root /tmp/catkin_workspace --os-name ubuntu --os-code-name xenial --arch amd64 --distribution-repository-urls http://repositories.ros.org/ubuntu/testing --distribution-repository-key-files /tmp/keys/0.key --dockerfile-dir /tmp/docker_build_and_test --testing"]
+ echo # END SECTION
# END SECTION
+ echo # BEGIN SECTION: Build Dockerfile - generating devel tasks
# BEGIN SECTION: Build Dockerfile - generating devel tasks
+ cd <http://build.ros.org/job/Kdev__jsk_common__ubuntu_xenial_amd64/ws/docker_generating_dockers>
+ python3 -u <http://build.ros.org/job/Kdev__jsk_common__ubuntu_xenial_amd64/ws/ros_buildfarm/scripts/misc/docker_pull_baseimage.py>
Get base image name from Dockerfile 'Dockerfile': ubuntu:xenial
Check docker base image for updates: docker pull ubuntu:xenial
xenial: Pulling from library/ubuntu
Digest: sha256:7c67a2206d3c04703e5c23518707bdd4916c057562dd51c74b99b2ba26af0f79
Status: Image is up to date for ubuntu:xenial
+ docker build --force-rm -t devel_task_generation.kinetic_jsk_common .
Sending build context to Docker daemon  15.87kB
Step 1/25 : FROM ubuntu:xenial
 ---> 20c44cd7596f
Step 2/25 : VOLUME /var/cache/apt/archives
 ---> Using cache
 ---> 27f5c077e10a
Step 3/25 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 049e1b3b90a4
Step 4/25 : RUN for i in 1 2 3; do apt-get update && apt-get install -q -y locales && apt-get clean && break || if [[ $i < 3 ]]; then sleep 5; else false; fi; done
 ---> Using cache
 ---> 64fb4cbc7f9a
Step 5/25 : RUN echo "en_US.UTF-8 UTF-8" >> /etc/locale.gen
 ---> Using cache
 ---> dea0530fd853
Step 6/25 : RUN locale-gen en_US.UTF-8
 ---> Using cache
 ---> 24c5ea514ea8
Step 7/25 : ENV LANG en_US.UTF-8
 ---> Using cache
 ---> 52f938375535
Step 8/25 : ENV TZ PST+08
 ---> Using cache
 ---> ed8b9be2ae9a
Step 9/25 : RUN useradd -u 1001 -m buildfarm
 ---> Using cache
 ---> 95a5942f9362
Step 10/25 : RUN mkdir /tmp/keys
 ---> Using cache
 ---> df32785333fd
Step 11/25 : RUN echo "-----BEGIN PGP PUBLIC KEY BLOCK-----\nVersion: GnuPG v1.4.11 (GNU/Linux)\n\nmQGiBEsy5KkRBADJbDSISoamRM5AA20bfAeBuhhaI+VaiCVcxw90sq9AI5lIc42F\nWzM2acm8yplqWiehAqOLKd+iIrqNGZ+VavZEPTx7o06UZUMRoPBiTFaCwrQ5avKz\nlt7ij8PRMVWNrJ7A2lDYXfFQVV1o3Xo06qVnv0KLLUmiur0LBu4H/oTH3wCgt+/I\nD3LUKaMJsc77KwFBTjHB0EsD/26Z2Ud12f3urSNyN6VMWnP3rz6xsmtY4Qsmkbnr\nJuduxCQBZv6bX1Cr2ulXkv0fFOr+s5OyUv7zyCPbxiJFh3Br7fJGb0b5/M208KPe\ngiITY9hMh/aUbKjXCPoOXPxSL6SWOWV8taR6903EFyLBN0qno/kXIBKnVqBZobgn\njIEPA/0fTnxtZtE7EpirGQMF2caJfv7/LCgXmRs9xAhgbE0/caoa1tnc79uaHmLZ\nFtbGFoAO31YNYM/IUHtmabbGdvZ4oYUwDhjBevVvC7aI+XhuNGK5mU8qCLLSEUOl\nCUr6BJq/0iFmjwjmwk9idZEYhqSNy2OoYJbq45rbHfbdKLEVrbQeUk9TIEJ1aWxk\nZXIgPHJvc2J1aWxkQHJvcy5vcmc+iGAEExECACAFAksy5KkCGwMGCwkIBwMCBBUC\nCAMEFgIDAQIeAQIXgAAKCRBVI7rusB+hFmk7AJ0XsLp05KA8l3YzAumZfjSN04MZ\njQCfQHfp4aQUXdOCUtetVo0QZUX3IuO5Ag0ESzLkrhAIAOCuSC83VXYWf8gOMSzd\nxwpsH/uLV9Wze2LGnajsJLjEOhcsz2BHfxqNXhYaE9aQaodPCpbUAkPq8tLbpXy0\nSWRCx0F5RcplXx5vIWbP6TlfPbRpK70w7IWd6vsNrjwEHjlhOLcNcj42sp5pgx4b\ndceK06k5Ml2hYovPnD9o2TYgjOqg5FHZ2g1J0103n/66bN/hZnpLaZJYQiPWCyq6\nK0565i1k2Y7hgWB/OXqwaqCehqmLTvpyQGzE1UJvKLuYU+T+4hBnSPbT3KIi5fCz\nlIwvxijOMcfbkLhzYQXcU0Rd1VItcd5nmPL4z97jBxzuhkgxXpGR4WGKhvsA2Z9Y\nUtsAAwYH/3Bf44bTpD9bVADUdab3e7zm8iHfh9K/a83mIgDB7mHV6WuemQVTf/1d\neu4mI5WtpbOCoucybGfjGIIAcSxwIx6VfC7HSp4J51bOpHhbdDffUEk6QVsZjwoF\nyn3W9W3ZVeTI+ch/Qoo5a98SnmdjN8eXI/qCuiXOHc6rXDXc2R0iox/1EAS8xGVd\ncYZe7IWBO2CjCknyhLrWxZHoy+i1GCZ9KvPF/Ef2dmLhCydT73ZlumsY8N5vm76Q\nul1G7f8LNbnMgXQafRkPffrAXSVhGY3Z2IiBwFNgxcKTq479l7yedYRGeU1A+SYI\nYmRFWHXt3rTkMlQSpxCsB0fAYfrwEqqISQQYEQIACQUCSzLkrgIbDAAKCRBVI7ru\nsB+hFpryAJ4puo6cMZxa6wITHFAM/k84+aRijwCeItuWpUngP25xDuDGMsKarcNi\nqYE=\n=Vgio\n-----END PGP PUBLIC KEY BLOCK-----\n" > /tmp/keys/0.key && apt-key add /tmp/keys/0.key
 ---> Using cache
 ---> eea397e5b93b
Step 12/25 : RUN echo deb http://repositories.ros.org/ubuntu/testing xenial main | tee -a /etc/apt/sources.list.d/buildfarm.list
 ---> Using cache
 ---> 0e6cd2014ad4
Step 13/25 : RUN mkdir /tmp/wrapper_scripts
 ---> Using cache
 ---> 29d5d5fc1bca
Step 14/25 : RUN echo "#!/usr/bin/env python3\n\n# Copyright 2014-2016 Open Source Robotics Foundation, Inc.\n#\n# Licensed under the Apache License, Version 2.0 (the \"License\");\n# you may not use this file except in compliance with the License.\n# You may obtain a copy of the License at\n#\n#     http://www.apache.org/licenses/LICENSE-2.0\n#\n# Unless required by applicable law or agreed to in writing, software\n# distributed under the License is distributed on an \"AS IS\" BASIS,\n# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.\n# See the License for the specific language governing permissions and\n# limitations under the License.\n\nimport subprocess\nimport sys\nfrom time import sleep\n\n\ndef main(argv=sys.argv[1:]):\n    max_tries = 10\n    known_error_strings = [\n        'Failed to fetch',\n        'Failed to stat',\n        'Hash Sum mismatch',\n        'Unable to locate package',\n        'is not what the server reported',\n    ]\n\n    command = argv[0]\n    if command in ['update', 'source']:\n        rc, _, _ = call_apt_repeatedly(\n            argv, known_error_strings, max_tries)\n        return rc\n    elif command == 'update-install-clean':\n        return call_apt_update_install_clean(\n            argv[1:], known_error_strings, max_tries)\n    else:\n        assert \"Command '%s' not implemented\" % command\n\n\ndef call_apt_update_install_clean(\n        install_argv, known_error_strings, max_tries):\n    tries = 0\n    command = 'update'\n    while tries < max_tries:\n        if command == 'update':\n            rc, _, tries = call_apt_repeatedly(\n                [command], known_error_strings, max_tries - tries,\n                offset=tries)\n            if rc != 0:\n                # abort if update was unsuccessful even after retries\n                break\n            # move on to the install command if update was successful\n            command = 'install'\n\n        if command == 'install':\n            # any call is considered a try\n            tries += 1\n            known_error_strings_redo_update = [\n                'Size mismatch',\n                'maybe run apt update',\n                'The following packages cannot be authenticated!',\n                'Unable to locate package',\n                'has no installation candidate',\n                'corrupted package archive',\n            ]\n            rc, known_error_conditions = \\\\\n                call_apt(\n                    [command] + install_argv,\n                    known_error_strings + known_error_strings_redo_update)\n            if not known_error_conditions:\n                if rc != 0:\n                    # abort if install was unsuccessful\n                    break\n                # move on to the clean command if install was successful\n                command = 'clean'\n                continue\n\n            # known errors are always interpreted as a non-zero rc\n            if rc == 0:\n                rc = 1\n            # check if update needs to be rerun\n            if (\n                set(known_error_conditions) &\n                set(known_error_strings_redo_update)\n            ):\n                command = 'update'\n                print(\"'apt install' failed and likely requires \" +\n                      \"'apt update' to run again\")\n                # retry with update command\n                continue\n\n            print('')\n            print('Invocation failed due to the following known error '\n                  'conditions: ' + ', '.join(known_error_conditions))\n            print('')\n            if tries < max_tries:\n                sleep_time = 5\n            [... cut ...]
[xUnit] [INFO] - [GoogleTest-1.6] - No test report file(s) were found with the pattern 'catkin_workspace/test_results/**/*.xml' relative to '<http://build.ros.org/job/Kdev__jsk_common__ubuntu_xenial_amd64/ws/'> for the testing framework 'GoogleTest-1.6'.  Did you enter a pattern relative to the correct directory?  Did you generate the result report(s) for 'GoogleTest-1.6'?
[xUnit] [WARNING] - No test reports found for the metric 'GoogleTest' with the resolved pattern 'catkin_workspace/test_results/**/*.xml'.
[xUnit] [INFO] - Skipping the metric tool processing.
[xUnit] [INFO] - There are errors when processing test results.
[xUnit] [INFO] - Skipping tests recording.
```

k-okada, 2017-11-26:
> amd64/10/display/redirect

error message from Kdev is ok, it just compile current source tree.
Kbin, Ksrc is used for deb build, so we have to be care about it.

wkentaro, 2017-11-26:
I see. Thank you!

closed 2017-11-26 by wkentaro