jsk-ros-pkg/jsk_recognition #2245: error in building dependencies

Ridhwanluthra, 2018-03-07:
when setting up jsk_recognision i clone the repo and run `rosdep install --from-paths <path> --ignore-src -r -y`.
This causes the error:
```
ERROR: the following rosdeps failed to install
  pip: command [pip install -U fcn] failed
  pip: Failed to detect successful installation of [fcn]
```
when installing fcn manually it says:
```
Downloading/unpacking fcn
  Downloading fcn-6.3.0.tar.gz
  Running setup.py (path:/tmp/pip_build_root/fcn/setup.py) egg_info for package fcn
    warning: no files found matching '*' under directory 'static'
Requirement already satisfied (use --upgrade to upgrade): chainer>=2.0.0 in /usr/local/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): chainercv in /usr/local/lib/python2.7/dist-packages (from fcn)
Downloading/unpacking gdown (from fcn)
  Downloading gdown-3.4.5.tar.gz
  Running setup.py (path:/tmp/pip_build_root/gdown/setup.py) egg_info for package gdown
Downloading/unpacking tqdm (from fcn)
  Downloading tqdm-4.19.6-py2.py3-none-any.whl (52kB): 52kB downloaded
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/local/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): scipy in /usr/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): matplotlib in /usr/lib/pymodules/python2.7 (from fcn)
Downloading/unpacking pandas (from fcn)
  Downloading pandas-0.22.0.tar.gz (11.3MB): 11.3MB downloaded
  Running setup.py (path:/tmp/pip_build_root/pandas/setup.py) egg_info for package pandas
    no previously-included directories found matching 'doc/build'
    warning: no previously-included files matching '*.so' found anywhere in distribution
    warning: no previously-included files matching '*.pyd' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '#*' found anywhere in distribution
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '.DS_Store' found anywhere in distribution
    warning: no previously-included files matching '*.png' found anywhere in distribution
Requirement already satisfied (use --upgrade to upgrade): PyYAML in /usr/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): scikit-learn in /usr/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): scikit-image in /usr/lib/python2.7/dist-packages (from fcn)
Requirement already satisfied (use --upgrade to upgrade): filelock in /usr/local/lib/python2.7/dist-packages (from chainer>=2.0.0->fcn)
Requirement already satisfied (use --upgrade to upgrade): protobuf>=3.0.0 in /usr/local/lib/python2.7/dist-packages (from chainer>=2.0.0->fcn)
Requirement already satisfied (use --upgrade to upgrade): six>=1.9.0 in /usr/local/lib/python2.7/dist-packages (from chainer>=2.0.0->fcn)
Requirement already satisfied (use --upgrade to upgrade): Pillow in /usr/local/lib/python2.7/dist-packages (from chainercv->fcn)
Requirement already satisfied (use --upgrade to upgrade): requests in /usr/lib/python2.7/dist-packages (from gdown->fcn)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/lib/python2.7/dist-packages (from matplotlib->fcn)
Downloading/unpacking tornado (from matplotlib->fcn)
  Downloading tornado-5.0.tar.gz (504kB): 504kB downloaded
  Running setup.py (path:/tmp/pip_build_root/tornado/setup.py) egg_info for package tornado
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_root/tornado/setup.py", line 146, in <module>
        raise ImportError("Tornado requires an up-to-date SSL module. This means "
    ImportError: Tornado requires an up-to-date SSL module. This means Python 2.7.9+ or 3.4+ (although some distributions have backported the necessary changes to older versions).
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/tornado/setup.py", line 146, in <module>

    raise ImportError("Tornado requires an up-to-date SSL module. This means "

ImportError: Tornado requires an up-to-date SSL module. This means Python 2.7.9+ or 3.4+ (although some distributions have backported the necessary changes to older versions).

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/tornado
Storing debug log for failure in /root/.pip/pip.log
```

furushchev, 2018-03-07:
Thanks for reporting.
It should be a problem not related to our software but your environment.
(It looks like you are using python in a custom environment.)
Please make sure that you are using python builtin the OS and follow the setup for ROS installation.

knorth55, 2018-03-20:
@Ridhwanluthra 
Your python version seems lower than `2.7.9`, which causes this error.
Is it solved?

pazeshun, 2018-08-03:
@Affonso-Gui also faces this problem.

After I and @YutoUchimi debugged, we found that this problem is occurred by `matplotlib` installed by apt of Ubuntu14.04.

When you install `matplotlib` by `sudo apt-get install python-matplotlib` in 14.04, version 1.3.1 is installed.
In this version, `matplotlib` explicitly depends on `tornado`.
```
$ pip show matplotlib
---
Metadata-Version: 1.1
Name: matplotlib
Version: 1.3.1
Summary: Python plotting package
Home-page: http://matplotlib.org
Author: John D. Hunter, Michael Droettboom
Author-email: mdroe@stsci.edu
License: BSD
Location: /usr/lib/pymodules/python2.7
Requires: numpy, python-dateutil, tornado, pyparsing, nose
```
However, apt version of `tornado` is not installed by apt-get due to lacking the dependency
```
$ apt-cache show python-matplotlib
Package: python-matplotlib
Priority: optional
Section: universe/python
Installed-Size: 7653
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Python Modules Team <python-modules-team@lists.alioth.debian.org>
Architecture: amd64
Source: matplotlib
Version: 1.3.1-1ubuntu5
Depends: python-dateutil, python-matplotlib-data (>= 1.3.1-1ubuntu5), python-pyparsing, python-tz, python (<< 2.8), python (>= 2.7), python-numpy (>= 1:1.8.0), python-numpy-abi9, python-support (>= 0.90.0), libc6 (>= 2.14), libfreetype6 (>= 2.2.1), libgcc1 (>= 1:4.1.1), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.12.0), libgtk2.0-0 (>= 2.8.0), libpng12-0 (>= 1.2.13-4), libstdc++6 (>= 4.6), libtcl8.6 (>= 8.6.0), libtk8.6 (>= 8.6.0)
```
In this situation, when you execute `pip install fcn`, latest `tornado` is installed by old `matplotlib` because `tornado` exists nowhere.
Latest `tornado` depends on Python 2.7.9 and upper.

pazeshun, 2018-08-03:
When you use Ubuntu16.04, this problem disappears because `python-matplotlib`' version becomes 1.5.1, which doesn't require tornado:
```
$ pip show matplotlib
Name: matplotlib
Version: 1.5.1
Summary: Python plotting package
Home-page: http://matplotlib.org
Author: John D. Hunter, Michael Droettboom
Author-email: matplotlib-users@python.org
License: BSD
Location: /home/pazeshun/.local/lib/python2.7/site-packages
Requires: python-dateutil, numpy, pyparsing, pytz, cycler
```

closed 2019-08-20 by YutoUchimi