jsk-ros-pkg/jsk_recognition #2246: jsk_perception build failure on Kinetic

knorth55, 2018-03-14:
Related to these issues, probably.
https://github.com/ros/catkin/issues/856
https://github.com/PR2/pr2_simulator/issues/138

ROS `kinetic`
`catkin_tools` `0.4.4`
```
__________________________________________________________________________________
Errors     << jsk_perception:cmake /home/shingo/ros/kinetic/logs/jsk_perception/build.cmake.005.log
CMake Error at /home/shingo/ros/kinetic/devel/share/image_view2/cmake/image_view2Config.cmake:148 (message):
  Project 'jsk_perception' tried to find library 'vtkCommon'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'image_view2'? Did you find_package() it before the subdirectory containing
  its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:17 (find_package)

cd /home/shingo/ros/kinetic/build/jsk_perception; catkin build --get-env jsk_perception | catkin env -si  /usr/bin/cmake /home/shingo/ros/kinetic/src/jsk-ros-pkg/jsk_recognition/jsk_perception --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/shingo/ros/kinetic/devel/.private/jsk_perception -DCMAKE_INSTALL_PREFIX=/home/shingo/ros/kinetic/install; cd -
..................................................................................
Failed     << jsk_perception:cmake                  [ Exited with code 1 ]        
Failed    <<< jsk_perception                        [ 0.7 seconds ]               
[build] Summary: 16 of 17 packages succeeded.                                     
[build]   Ignored:   124 packages were skipped or are blacklisted.                
[build]   Warnings:  None.                                                        
[build]   Abandoned: None.                                                        
[build]   Failed:    1 packages failed.                                           
[build] Runtime: 4.3 seconds total.
```

I comment out one line, and it succeeded to build it.
The error message does not tell us what the exact reason, so I could not find out the critical point.
```diff
--- a/jsk_perception/CMakeLists.txt
+++ b/jsk_perception/CMakeLists.txt
@@ -22,7 +22,7 @@ find_package(catkin REQUIRED COMPONENTS
   geometry_msgs
   image_geometry
   image_transport
-  image_view2
+  # image_view2
   jsk_data
   jsk_recognition_msgs
   jsk_recognition_utils
```

knorth55, 2018-03-14:
`vtkCommon` can be found in `image_view2Config.cmake`

k-okada, 2018-03-15:
Humm, I can not reproduce your error ... 
can you check this make log ? https://gist.github.com/c32d3a5b5789e8fdb302471a2cf05910

```
user@ubuntu-16:~/jsk_ws$ grep  version src/jsk_common/jsk_ros_patch/image_view2/package.xml 
  <version>2.2.6</version>
user@ubuntu-16:~/jsk_ws$ grep  version src/jsk_recognition/jsk_perception/package.xml 
  <version>1.2.4</version>
```
catkin version
```
$ apt show python-catkin-tools
Package: python-catkin-tools
Version: 0.4.4-1
```

knorth55, 2018-03-15:
I clean up all the workspace and rebuilt.
This time, this error does not occurs, it is realy weird because the error always occurs yesterday, but not today :(

knorth55, 2018-03-20:
close this issue because this error cannot be reproduced.

closed 2018-03-20 by knorth55