jsk-ros-pkg/jsk_recognition #2254: [jsk_pcl_ros] cannot make a height map on kinetic

yuki-shark, 2018-04-02:
When _heightmap_converter.launch_ in _jsk_pcl_ros_ is launched, some topics are not adequately published. And it fails to make a heightmap.

I am using Ubuntu16.04 and ros kinetic now.
I installed _jsk_recognition_ package from github source and changed branches from _master_ to _enable_kinetic_.
_jsk_common_ package was also installed from source.

I added only _INPUT_ argument when I executed the launch. And other arguments are default.
Although _/current_heightmap/output_ and _/filtered_current_heightmap/output_ were properly published, _/accumulated_heightmap_ was not published.

When DEBUG_VIEW argument is true 4 image viewers out of 6 work, and the other two (_accumulated_map_ and _accumulated_quality_) do not.

Building packages were succeeded.
Is there any package to run the launch successfully?
I tried rebuilding packages, but the situation was same as before.

Messages when _heightmap_converter.launch_ was executed are following.

```
started roslaunch server http://16CORSAIR:36779/

SUMMARY
========

PARAMETERS
 * /accumulated_heightmap/bilateral_filter_size: 4
 * /accumulated_heightmap/bilateral_sigma_color: 0.028
 * /accumulated_heightmap/bilateral_sigma_space: 3
 * /accumulated_heightmap/center_frame_id: BODY
 * /accumulated_heightmap/fixed_frame_id: map
 * /accumulated_heightmap/use_bilateral: False
 * /accumulated_heightmap_pointcloud_static/target_frame_id: map
 * /current_heightmap/center_frame_id: BODY
 * /current_heightmap/fixed_frame_id: map
 * /current_heightmap/max_x: 13.0
 * /current_heightmap/max_y: 8.0
 * /current_heightmap/min_x: -3.0
 * /current_heightmap/min_y: -8.0
 * /current_heightmap/projected_center_frame_id: BODY_on_map
 * /current_heightmap/resolution_x: 1600
 * /current_heightmap/resolution_y: 1600
 * /current_heightmap/use_projected_center: False
 * /filtered_current_heightmap/bilateral_filter_size: 4
 * /filtered_current_heightmap/bilateral_sigma_color: 0.028
 * /filtered_current_heightmap/bilateral_sigma_space: 3
 * /filtered_current_heightmap/mask_size: 3
 * /filtered_current_heightmap/max_variance: 0.0002
 * /filtered_current_heightmap/smooth_method: average_distance
 * /filtered_current_heightmap/use_bilateral: True
 * /heightmap_nodelet/nodelets_0: [{'remappings': [...
 * /heightmap_visualization/accumulated_map/heightmap_view/autosize: True
 * /heightmap_visualization/accumulated_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/accumulated_quality/heightmap_view/autosize: True
 * /heightmap_visualization/filtered_map/heightmap_view/autosize: True
 * /heightmap_visualization/filtered_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/filtered_quality/heightmap_view/autosize: True
 * /heightmap_visualization/latest_map/heightmap_view/autosize: True
 * /heightmap_visualization/latest_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/latest_quality/heightmap_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /heightmap_visualization/latest_quality/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /heightmap_visualization/filtered_quality/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /
    heightmap_nodelet (jsk_topic_tools/standalone_complexed_nodelet)
[... 41 lines cut ...]
[ INFO] [1522695283.120816041]: Succeeded to load nodelet [/accumulated_heightmap_pointcloud -- jsk_pcl/HeightmapToPointCloud]
[ INFO] [1522695283.120853964]: i_nodelet 4
[ INFO] [1522695283.120931022]: remapping: /accumulated_heightmap_pointcloud_static/input => /accumulated_heightmap_pointcloud/output
[ INFO] [1522695283.129877267]: Succeeded to load nodelet [/accumulated_heightmap_pointcloud_static -- jsk_pcl/TfTransformCloud]
[ INFO] [1522695283.188953176]: done reading parmaeters
[ INFO] [1522695283.189011233]: loaded nodelets: 5
[ INFO] [1522695283.189056425]: loaded nodelet: /accumulated_heightmap
[ INFO] [1522695283.189089591]: loaded nodelet: /accumulated_heightmap_pointcloud
[ INFO] [1522695283.189106794]: loaded nodelet: /accumulated_heightmap_pointcloud_static
[ INFO] [1522695283.189119685]: loaded nodelet: /current_heightmap
[ INFO] [1522695283.189136645]: loaded nodelet: /filtered_current_heightmap
[ WARN] [1522695288.056279362]: '/accumulated_heightmap' subscribes topics only with child subscribers.
[ WARN] [1522695288.127333682]: '/accumulated_heightmap_pointcloud_static' subscribes topics only with child subscribers.
[ WARN] [1522695298.191050569]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695358.191939450]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695418.202204129]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695478.209089043]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695538.212653668]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695598.220218094]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695658.229046355]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695718.233025237]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695778.236569538]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695838.237309771]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1522695898.244160842]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
```

k-okada, 2018-04-02:
please install everything from deb and test again

yuki-shark, 2018-04-03:
Although I installed from deb and tested again, the same error occurred.
The workspace I tested was /opt/ros/kinetic and I cleaned build and devel before the test just in case.

k-okada, 2018-04-03:
If you install from deb, you do not need build and devel. And that is not a
‘same’ error, at least you can find different path in your message. Please
check again.

YoheiKakiuchi, 2018-04-03:
I think this is highly suggestive message.
```
[ WARN] [1522695298.191050569]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
```

'map' frame and 'BODY' frame may be required to create accumulated map.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/launch/heightmap_converter.launch#L10-L11

k-okada, 2018-04-03:
please also turn on DEBUG mode for `ros.jsk_pcl_ros.message_notifier` in rosconsole
```
Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
```

yuki-shark, 2018-04-03:
I turned on DEBUG mode, but the DEBUG messages seemed nothing to do with the problem.
I attached messages at the end of this post.

'BODY' frame can be specified as /camera_link, but 'map' frame should be defined somewhere.
I am trying to publish /map frame in /tf topic, but I have no idea about how to publish it.

Transformation between 'BODY' and 'map' is not static, so I think tf_static_publisher cannot be used in this case.

[image]

```
[ros] yuki@16CORSAIR:~$ roslaunch jsk_pcl_ros heightmap_converter.launch INPUT:=/camera/depth_registered/points DEBUG_VIEW:=true
... logging to /home/yuki/.ros/log/f8c4a996-3759-11e8-8cea-fcaa14795e8e/roslaunch-16CORSAIR-10367.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/yuki/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://16CORSAIR:34019/

SUMMARY
========

PARAMETERS
 * /accumulated_heightmap/bilateral_filter_size: 4
 * /accumulated_heightmap/bilateral_sigma_color: 0.028
 * /accumulated_heightmap/bilateral_sigma_space: 3
 * /accumulated_heightmap/center_frame_id: BODY
 * /accumulated_heightmap/fixed_frame_id: map
 * /accumulated_heightmap/use_bilateral: False
 * /accumulated_heightmap_pointcloud_static/target_frame_id: map
 * /current_heightmap/center_frame_id: BODY
 * /current_heightmap/fixed_frame_id: map
 * /current_heightmap/max_x: 13.0
 * /current_heightmap/max_y: 8.0
 * /current_heightmap/min_x: -3.0
 * /current_heightmap/min_y: -8.0
 * /current_heightmap/projected_center_frame_id: BODY_on_map
 * /current_heightmap/resolution_x: 1600
 * /current_heightmap/resolution_y: 1600
 * /current_heightmap/use_projected_center: False
 * /filtered_current_heightmap/bilateral_filter_size: 4
 * /filtered_current_heightmap/bilateral_sigma_color: 0.028
 * /filtered_current_heightmap/bilateral_sigma_space: 3
 * /filtered_current_heightmap/mask_size: 3
 * /filtered_current_heightmap/max_variance: 0.0002
 * /filtered_current_heightmap/smooth_method: average_distance
 * /filtered_current_heightmap/use_bilateral: True
 * /heightmap_nodelet/nodelets_0: [{'remappings': [...
 * /heightmap_visualization/accumulated_map/heightmap_view/autosize: True
 * /heightmap_visualization/accumulated_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/accumulated_quality/heightmap_view/autosize: True
 * /heightmap_visualization/filtered_map/heightmap_view/autosize: True
 * /heightmap_visualization/filtered_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/filtered_quality/heightmap_view/autosize: True
 * /heightmap_visualization/latest_map/heightmap_view/autosize: True
 * /heightmap_visualization/latest_quality/colorize_heightmap/channel: 1
 * /heightmap_visualization/latest_quality/heightmap_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /heightmap_visualization/latest_quality/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /heightmap_visualization/filtered_quality/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /
    heightmap_nodelet (jsk_topic_tools/standalone_complexed_nodelet)
[... 44 lines cut ...]
[ INFO] [1522772055.219418662]: Succeeded to load nodelet [/accumulated_heightmap_pointcloud_static -- jsk_pcl/TfTransformCloud]
[ INFO] [1522772055.311270564]: done reading parmaeters
[ INFO] [1522772055.311336630]: loaded nodelets: 5
[ INFO] [1522772055.311369286]: loaded nodelet: /accumulated_heightmap
[ INFO] [1522772055.311400673]: loaded nodelet: /accumulated_heightmap_pointcloud
[ INFO] [1522772055.311440023]: loaded nodelet: /accumulated_heightmap_pointcloud_static
[ INFO] [1522772055.311489313]: loaded nodelet: /current_heightmap
[ INFO] [1522772055.311538745]: loaded nodelet: /filtered_current_heightmap
[ WARN] [1522772060.158464321]: '/accumulated_heightmap' subscribes topics only with child subscribers.
[ WARN] [1522772060.214996517]: '/accumulated_heightmap_pointcloud_static' subscribes topics only with child subscribers.
[ WARN] [1522772070.321575067]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.jsk_pcl_ros.message_notifier] rosconsole logger to DEBUG for more information.
[DEBUG] [1522772080.206239809]: MessageFilter [target=map ]: Removed oldest message because buffer is full, count now 10 (frame_id=camera_rgb_optical_frame, stamp=1522772075.047244)
[DEBUG] [1522772080.206501016]: MessageFilter [target=map ]: Added message in frame camera_rgb_optical_frame at time 1522772078.985, count now 10
[DEBUG] [1522772080.575296786]: MessageFilter [target=map ]: Removed oldest message because buffer is full, count now 10 (frame_id=camera_rgb_optical_frame, stamp=1522772075.481067)
[DEBUG] [1522772080.575461743]: MessageFilter [target=map ]: Added message in frame camera_rgb_optical_frame at time 1522772079.418, count now 10
[DEBUG] [1522772080.948226340]: MessageFilter [target=map ]: Removed oldest message because buffer is full, count now 10 (frame_id=camera_rgb_optical_frame, stamp=1522772075.814751)
[DEBUG] [1522772080.948325597]: MessageFilter [target=map ]: Added message in frame camera_rgb_optical_frame at time 1522772079.752, count now 10
[... cut ...]
[DEBUG] [1522772082.113070200]: MessageFilter [target=map ]: Removed oldest message because buffer is full, count now 10 (frame_id=camera_rgb_optical_frame, stamp=1522772076.982625)
[DEBUG] [1522772082.113236690]: MessageFilter [target=map ]: Added message in frame camera_rgb_optical_frame at time 1522772080.920, count now 10
[DEBUG] [1522772082.496612710]: MessageFilter [target=map ]: Removed oldest message because buffer is full, count now 10 (frame_id=camera_rgb_optical_frame, stamp=1522772077.316346)
```

yuki-shark, 2018-04-03:
When I added following arguments, the launch worked.
```
[ros] yuki@16CORSAIR:~$ roslaunch jsk_pcl_ros heightmap_converter.launch INPUT:=/camera/depth_registered/points DEBUG_VIEW:=true ROBOT_FRAME:=/camera_link STATIC_FRAME:=/camera_link
```

closed 2018-04-03 by yuki-shark

YoheiKakiuchi, 2018-04-03:
FYI:
heightmap_converter.launch generates heightmap from single point cloud and accumulated heightmaps from multiple point cloud while camera moving.
So, it requires that INPUT should be alined as Z axis is up direction.
And, 'map' frame is requires, it represents 'world origin' of moving camera.