jsk-ros-pkg/jsk_recognition #2271: 最新のros-kinetic-opencv3をインストールするとビルドが通らなくなる

chibi314, 2018-04-25:
ubuntu 16.04 + ROS kineticにおける問題ですが，
jsk_recognitionのビルドが通らず，/opt/ros/kinetic/include/opencv-3.3.1-devがないというエラーが一部で出たので
```$ sudo apt-get install ros-kinetic-opencv3```
としてアップグレードしたところ，/opt/ros/kinetic/include/opencv-3.3.1が/opt/ros/kinetic/include/opencv-3.3.1-devになり，このエラーはなくなり，逆に元のディレクトリはないためさらにエラーが出るようになりました．

そこでもとに戻すためにまずremoveしたのですが，aptを使ってバージョンを元に戻すことができませんでした．
```$ apt-cache show ros-kinetic-opencv3```
としてもVersion: 3.3.1-5xenial-20180315-113535-0800しか出てこず，元のバージョンが指定できませんでした．

そこでhttps://mirror.vtti.vt.edu/packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-opencv3/ からdebを持ってきて
``` $ sudo dpkg -i hogehoge.deb```
したら一応もとに戻ったという次第です．

この後もう一度jsk_recognitionをビルドしたらやはり/opt/ros/kinetic/include/opencv-3.3.1-devを要求してくるものがあったので，それら(cv_bridgeなど)をソースから入れ，全体のビルドを通せました．

まずaptでバージョンを戻せなかったという問題と，ros-kinetic-opencv3の別のバージョンを見ているパッケージがあるという問題があったのですが，どのようにするのが正しいのでしょうか．

k-okada, 2018-04-25:
コマンドの実行結果，あるいはエラーについても全て情報をpasteしましょう．(gist.github.comを使っても良いです．http://wiki.ros.org/Support#Do も参考に）

で，shadow-fixedを使っているのかな？最近opencv3のパッケージに何回か変更があったので，
https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384/5
それに対応したパッケージがリポジトリにまだないタイミングか，install(update)していなかったんじゃないかとおもいます．
（いかんせん，状況がつかめないので全くの空想です）

ということで，これが正しいかを書くにするのにcv_bridgeなどのソースをいっかい外して
`apt-get install ros-kinetic-opencv3`して，jsk_recognitionのエラーが出ないことを確認して，（エラーはgistに貼り付けて．．．）

で，
```
$ sudo apt-get update
$ sudo apt-get dist-upgrade
```
してみてください．このあと`catkin clean ; catkin b`　してみてください．

なぜこれが必要かは https://github.com/ros-infrastructure/bloom/issues/414

chibi314, 2018-04-26:
ビルドが通るようになりました．

dist-upgradeによってopencvと依存関係にあるパッケージもupgradeされることで問題が解消したということでいいのでしょうか．
結局，OpenCVConfig.cmakeを各CMakeLists.txtで読み，OpenCV_INCLUDE_DIRS等が（バージョンによって変わる）設定されてビルドできるから，ソースから入れると問題がない．

debの場合は古いバージョンだと元のOpenCV_INCLUDE_DIRS等を参照しているためアップグレードしてやらないと問題が起きる．
という理解であっているのでしょうか．

k-okada, 2018-04-26:
良かったです．
upgradeしたときのlogがあるともう少し正確なことが言えるとおもいますが，
```
k-okada@p51s:/opt/ros/kinetic/share/cv_bridge/cmake$ grep include/opencv-3.3.1-dev *
cv_bridgeConfig.cmake:if(NOT "include;/opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv " STREQUAL " ")
cv_bridgeConfig.cmake:  set(_include_dirs "include;/opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv")

```
みたいにコンパイルした時の環境をつかって，依存しているパッケージの中の情報にパスが直書きされるところがあるので，
これが残っていると，このissueみたいなことが起こるんだとおもいます．

closed 2019-03-01 by knorth55